KLayout/klayout #488: Is KLayout at FOSDEM?

mithro, 2020-02-01:
Hi!

Sorry, I wasn't sure what the best way to contact you was. I was wondering if anyone from the KLayout project was attending FOSDEM? I would love to chat with anyone involved in the project around efforts I'm doing regarding open source PDKs.

Tim '@mithro' Ansell

klayoutmatthias, 2020-02-02:
Hi Tim,

not FOSDEM, I'm afraid. But I'm going to be at another conference, FSiC, which is taking place in June (4th to 6th) in Zurich (https://wiki.f-si.org/index.php/FSiC2020). Open source PDKs are a very important topic IMHO and I'd love to chat about that.

Best regards,

Matthias

klayoutmatthias, 2020-02-02:
P.S. you could post such topics in the KLayout forum (https://www.klayout.de/forum)

closed 2020-02-19 by klayoutmatthias