KLayout/klayout #494: When writing a python macro, is importing python libraries using Pip supported?

skylarjhdownes, 2020-02-05:
Hello!

I've been writing a macro using Python on Windows, and I'd like to be able to import some external libraries from pypi.  I've tried running in a subprocess, but that uses my locally installed Python, and running 
```
from pip._internal import main
main(['install', '--ignore-installed', '-v', 'matplotlib '])
```
in the macro development window results in this error:
```
Could not install packages due to an EnvironmentError.
Traceback (most recent call last):
  File "C:\Users\MyAccount\KLayout\lib\python3.7\site-packages\pip\_internal\req\req_tracker.py", line 52, in add
    with open(entry_path) as fp:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\MyAccount\\AppData\\Local\\Temp\\pip-req-tracker-l2yuud_f\\456653283d012fbbc7fb74231947e327099b2093cd1ccfaf917f889c'
```

Is there a supported way to use external python libraries with KLayout macros?

klayoutmatthias, 2020-02-10:
Hi,

Windows does not have a package manager, so KLayout comes with it's own Python copy. I'm taking it from the MSYS2 project, so you could basically pull packages from there. But as this is a rolling release, it's likely that it advanced too far now.

I'm sorry, but Windows, KLayout and PyPI simply don't get together. That's why I included a few important modules (such as numpy) in the distribution. 

If you want standalone KLayout DB module, you can go the other way round and use Anaconda and pull the klayout PyPI module. This will give you scripting on database level, but not the UI.

Matthias

skylarjhdownes, 2020-02-13:
Sounds like if I want to work with both external libraries and the QT UI on Windows, I'll have to build any libraries I need in a way that I can access them from the ...\KLayout\python directory.  I've been able to do that for some simple Python-only dependencies, but I'm not as sure of that approach when I need scripts with embedded C code and the like.  

Thanks for the clarification!

closed 2020-02-13 by skylarjhdownes

lukasc-ubc, 2020-02-13:
On Mac OSX and Linux, I’ve been using a similar approach as you describe, namely using pip to install numpy, scipy, requests, etc.  This is included in SiEPIC-Tools (on salt, and GitHub) if you are interested in seeing the code.