KLayout/klayout #495: Using Python scripting API to generate png from GDS file

mithro, 2020-02-07:
I would like to use the klayout scripting API to generate a PNG image from a GDS file.

From what I can see I need the [`LayoutView.get_screenshot` function](https://www.klayout.de/doc-qt5/code/class_LayoutView.html#m_get_screenshot) and I can load a gds file using the following;
```python
import pya

ly = pya.Layout()
ly.read("my.gds")
```

But I'm unclear how to get a LayoutView from a Layout? In fact I'm unclear how I access any of the API described at https://www.klayout.de/doc-qt5/code/module_lay.html ?

mithro, 2020-02-07:
Gotten a little further, looks like I want the following to get the klayout application;
```python
app = pya.Application.instance()
mw = app.main_window()
```

mithro, 2020-02-07:
It seems what I need is;
```python
import pya
app = pya.Application.instance()
mw = app.main_window()
mw.load_layout(IN_FILE, 0)
lv = mw.current_view()
lv.save_image(OUT_FILE, 1000, 1000)
```
Then I can do;
```
/opt/klayout/klayout -z -r image.py
```

It appears the `Application` module doesn't appear when you import `pya` in a script directly.

Just trying to figure out how to give the script arguments now....

mithro, 2020-02-07:
Seems I need `/opt/klayout/klayout -z -r image.py -rd IN_FILE='path-to-my-gds'`

klayoutmatthias, 2020-02-10:
Right :)

Looks good. 

Is this actually a ticket? For questions of this kind, you might want to use the forum: https://www.klayout.de/forum/

Tickets are for tasks and finally get closed. So this information will vanish and is no longer available for others.

Matthias

klayoutmatthias, 2020-02-19:
I'll close this ticket - I don't see what I can do further.

closed 2020-02-19 by klayoutmatthias