KLayout/klayout #497: Problems with recreating a QDialog window from PyQt to KLayout's Qt

rizoschrist, 2020-02-13:
For start, I know that KLayout doesn't use/have PyQt, but provides an interface to the tool's Qt via Python, and there are conflicts to try using both.

I am trying to "port" a QDialog window which I've already created in a PyQt environment, using that interface, unsuccessfully.

From what I understand the problem lies in the nested Q[H/V]BoxLayout I use to set my window, but I am not sure about that.

Is there a way to import PyQt to KLayout or to fix this breakdown?

P.S. I will try to prepare an example that works in PyQt, but not in KLayout, if there isn't any fast/known solution.

klayoutmatthias, 2020-02-16:
Please provide this example.

Actually, PyQt can be used with KLayout. On Linux, KLayout shares the Python installation with that of the system, so it's sufficient to just install PyQt there. But there are restrictions of interoperability because PyQt does not have access to KLayout's binary API and UI classes.

KLayout on Windows is built against MSYS2 packages. You may try to find something there, but as this is a rolling release, there is only a small change you'll find a fitting version. I think this is what you mean by "conflicts".

Matthias

rizoschrist, 2020-02-17:
Hi Matthias,

an example of what I am describing:

	# from PyQt5 import QHBoxLayout, QVBoxLayout, QGridLayout, QGroupBox, QDialog, QTreeView
	from pya import QHBoxLayout, QVBoxLayout, QGridLayout, QGroupBox, QDialog, QTreeView
	
	class MyWindow(QDialog) :

		def __init__(self, parent=None) :

			super(MyWindow, self).__init__(parent)

			mainGrid = QGridLayout(self)

			mainLayout = QVBoxLayout()

			mainGrid.addLayout(mainLayout, 0, 0, 1, 1)

			self.createGroupBox()
			mainGrid.addWidget(self.topGroupBox, 0, 0)

			self.setLayout(mainGrid)
			self.setWindowTitle("MyWindow")
		# __init__

		def createGroupBox(self):
			self.topGroupBox = QGroupBox("Say something here to display the problem:")

			topGroupLayout = QHBoxLayout()

			tree = QTreeView(self)

			topGroupLayout.addWidget(tree)

			self.topGroupBox.setLayout(topGroupLayout)
		# createTopGroupBox
	# MyWindow

So we can use both of the following commands?
import pya
import PyQt5 # or PyQt4

because, I think I read in a discussion in KLayout's forum that, we should not use both simultaneously, otherwise, may occur conflicts in the usage of the Qt modules.

Chris

klayoutmatthias, 2020-02-17:
Hi Chris,

The problem is because "setLayout" does not transfer ownership currently. So in this line:

```
self.topGroupBox.setLayout(topGroupLayout)
```

"topGroupLayout" still belongs to the calling code and this object is deleted when the function returns. I have created a ticket to collect those issues: https://github.com/KLayout/klayout/issues/501

An easy workaround is to do the widget association in the constructor:

```
	def createGroupBox(self):
		self.topGroupBox = QGroupBox("Say something here to display the problem:")

		topGroupLayout = QHBoxLayout(self.topGroupBox)

		tree = QTreeView(self)

		topGroupLayout.addWidget(tree)
```

With this modification it works for me.

Regarding PyQt: as KLayout on Linux is build against Qt4, you *must* use PyQt4. Trying PyQt5 will crash immediately. 

This line works for me:

```
from PyQt4.QtGui import QHBoxLayout, QVBoxLayout, QGridLayout, QGroupBox, QDialog, QTreeView
```

With the above modification the dialogs come up the same way with PyQt4 and KLayout's Qt binding. But note that there are considerable differences when it comes to event handling. The Qt binding of KLayout utilizes generic events rather than signals/slots. This means, instead of 

```
        btn1.clicked.connect(self.buttonClicked)
```

you need to write

```
        btn1.clicked += self.buttonClicked
```

This concept delivers it's full power with Proc's and lambdas in Ruby, but Python provides few of Ruby's metaprogramming concepts. And BTW, Ruby support is the main reason I did not use PyQt.

There are many more tiny differences, you I don't think it's feasible to write code that runs on PyQt and KLayout's Qt binding.

Matthias

rizoschrist, 2020-02-19:
Hi, Matthias,

Thanks for the last info, I also read a lot about this in the forum.

Ok, I thought I read that there are conflicts using PyQt[5 or 4] when importing them in KLayout scripts, that's why I wanted to "port" the code in the first place.

Thank you for your clarifications, I will use plain PyQt.

Chris

closed 2020-02-19 by rizoschrist