KLayout/klayout #506: Support for Ruby2.7?

utkarsh2102, 2020-02-20:
Hi,

Thank you for your awesome work on klayout so far!
However, while having Ruby2.7 transitions in Debian, we saw some failures on different architectures. Can you take a look at https://buildd.debian.org/status/package.php?p=klayout and help us out a bit? :)

klayoutmatthias, 2020-02-21:
Thanks a lot for reporting this ... but frankly, none of these issues I see in the link seem to be related to Ruby 2.7. Could you give me a hint where to look at?

Thanks,

Matthias

klayoutmatthias, 2020-02-22:
Hi,

I just built with Fedora/Rawhide (they have Ruby 2.7.0 already). The build runs smoothly with the build script. So it's not a Ruby 2.7 issue.

Configuration with Rawhide is: gcc 10.0.1, Qt 5.13.2, Python 3.8.2, Ruby 2.7.0.

I tried Debian/Bullseye as well, but they are still on Ruby 2.5.

Matthias

klayoutmatthias, 2020-03-08:
Any feedback?

utkarsh2102, 2020-03-17:
Hi @klayoutmatthias,

Thanks for taking a look. We have Ruby 2.7 on Debian/Sid (unstable).
However, someone recently fixed this in Debian (it was because of some Python issue).

But there's a failure on mipsel architecture, see here:
https://buildd.debian.org/status/fetch.php?pkg=klayout&arch=mipsel&ver=0.26.2-2&stamp=1584077604&raw=0

Any idea what's causing this?

pankajp, 2020-08-18:
> dwz: debian/klayout/usr/lib/klayout/libklayout_QtWidgets.so.0.26.2: **Could not allocate memory: Cannot allocate memory**
> 
> dh_dwz: error: dwz -mdebian/klayout/usr/lib/debug/.dwz/mipsel-linux-gnu/klayout.debug -M/usr/lib/debug/.dwz/mipsel-linux-gnu/klayout.debug -- debian/klayout/usr/lib/klayout/db_plugins/libcif.so.0.26.2 debian/klayout/usr/lib/klayout/db_plugins/libdxf.so.0.26.2 debian/klayout/usr/lib/klayout/db_plugins/libgds2.so.0.26.2 debian/klayout/usr/lib/klayout/db_plugins/liblefdef.so.0.26.2 debian/klayout/usr/lib/klayout/db_plugins/libmag.so.0.26.2 debian/klayout/usr/lib/klayout/db_plugins/libnet_tracer.so.0.26.2 debian/klayout/usr/lib/klayout/db_plugins/liboasis.so.0.26.2 debian/klayout/usr/lib/klayout/db_plugins/libpcb.so.0.26.2 debian/klayout/usr/lib/klayout/klayout debian/klayout/usr/lib/klayout/lay_plugins/libbool_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/libcif_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/libcommon_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/libdiff_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/libdxf_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/libgds2_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/libimport_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/liblefdef_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/libmag_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/libnet_tracer_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/liboasis_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/libpcb_ui.so.0.26.2 debian/klayout/usr/lib/klayout/lay_plugins/libxor_ui.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_QtCore.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_QtDesigner.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_QtGui.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_QtMultimedia.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_QtNetwork.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_QtPrintSupport.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_QtSql.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_QtSvg.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_QtWidgets.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_QtXml.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_QtXmlPatterns.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_ant.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_bd.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_db.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_drc.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_edt.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_gsi.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_img.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_lay.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_laybasic.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_lib.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_lvs.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_lym.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_pya.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_qtbasic.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_rba.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_rdb.so.0.26.2 debian/klayout/usr/lib/klayout/libklayout_tl.so.0.26.2 debian/klayout/usr/lib/klayout/pymod/klayout/QtCore.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/QtDesigner.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/QtGui.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/QtMultimedia.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/QtNetwork.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/QtPrintSupport.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/QtSql.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/QtSvg.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/QtWidgets.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/QtXml.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/QtXmlPatterns.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/dbcore.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/lay.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/libcore.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/rdbcore.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/pymod/klayout/tlcore.cpython-38-mipsel-linux-gnu.so debian/klayout/usr/lib/klayout/strm2cif debian/klayout/usr/lib/klayout/strm2dxf debian/klayout/usr/lib/klayout/strm2gds debian/klayout/usr/lib/klayout/strm2gdstxt debian/klayout/usr/lib/klayout/strm2mag debian/klayout/usr/lib/klayout/strm2oas debian/klayout/usr/lib/klayout/strm2txt debian/klayout/usr/lib/klayout/strmclip debian/klayout/usr/lib/klayout/strmcmp debian/klayout/usr/lib/klayout/strmrun debian/klayout/usr/lib/klayout/strmxor returned exit code 1
> make: *** [debian/rules:24: binary-arch] Error 25
> dpkg-buildpackage: error: fakeroot debian/rules binary-arch subprocess returned exit status 2

The error is due to "out of memory" on the build machine. You can probably get a build by eliminating the dwz command.

utkarsh2102, 2020-08-18:
> The error is due to "out of memory" on the build machine. You can probably get a build by eliminating the dwz command.

Aye, you're right. Thanks!

klayoutmatthias, 2021-08-24:
Ruby 2.7 is standard now. Closing this old issue.

closed 2021-08-24 by klayoutmatthias