KLayout/klayout #511: asyncio, for asynchronous subprocesses

lukasc-ubc, 2020-02-24:
Dear Matthias,

I am using KLayout to launch long device and circuit simulations. Python has a means of launching a subprocess asynchronously.  This means we could have the KLayout design environment continue, while the computer performs simulations in the background.  Then when the process is over, a dialog box could appear to let the user know it is complete.

https://docs.python.org/3/library/asyncio-subprocess.html

I tried to implement this, and ran into an issue:

> asyncio.run() cannot be called from a running event loop

 It looks like you are already using this feature on the same thread:

https://stackoverflow.com/questions/55409641/asyncio-run-cannot-be-called-from-a-running-event-loop/55409674

There appear to be some solutions:

1) do what Jupyter notebook people managed, where users have access to the same event loop

or 

2) add a python module that allows nested event loops:

https://pypi.org/project/nest-asyncio/

Do you think these would be possible?

thank you
Lukas

klayoutmatthias, 2020-03-08:
Hi Lukas,

I'm not sure how asyncio does it, but I think the problem is related with the Qt event loop. Basically every framework provides it's own implementation of the event loop and here they interfere.

Does "nest-asyncio" solve the issue? By "add a python module" you mean to add it to the Windows distribution?

Best regards,

Matthias

lukasc-ubc, 2020-03-09:
Hi Matthias,

I haven't been able to test nest-asyncio.  I can try this and get back to you.