KLayout/klayout #512: DRC reports duplicated width check violations

Mitsuhide-T, 2020-02-25:
Hi Matthias,

I'm testing my simple width check DRC on version 0.25.9 and version 0.26.3.
Please refer to the attached ZIP file [widthcheck.zip](https://github.com/KLayout/klayout/files/4247704/widthcheck.zip) that contains eight files listed in the table below.

The expected number of width check violations is 28=14x2. 
The results using version 0.25.9 is 28 as expected. However, version 0.26.3 reports 38 with 10 duplications.

|Srl.No.|File Name|Contents|
|:------:|:--------:|:--------------:|
1-1|widthcheck_0.25/widthcheck.lys|Session file|
1-2|widthcheck_0.25/widthcheck.oas|Test design in OASIS|
1-3|widthcheck_0.25/widthcheck.lydrc|Simple DRC macro|
1-4|widthcheck_0.25/widthcheck.lyrdb|Report database|
2-1|widthcheck_0.26/widthcheck.lys|Session file|
2-2|widthcheck_0.26/widthcheck.oas|Same as 1-2|
2-3|widthcheck_0.26/widthcheck.lydrc|Same as 1-3|
2-4|widthcheck_0.26/widthcheck.lyrdb|Report database|

The target polygon in the 0/1 layer has 14 notches and each notch has two rectangular bumps (<20um).
[image]
[image]

> My simple DRC.
```
report("Simple DRC")
lay = 0
dty = 1
val = 20.um
inlayer = input( lay.to_i, dty.to_i )
categoryNameWidth = "width check " + val.to_s + "um"
inlayer.width( val, projection ).output( categoryNameWidth )
```
> Results

**version 0.25.9 returns 28 edge pairs.**
[image]

**version 0.26.3 returns 38 edge pairs.**
[image]

My platform. 
Linux Mint 19 is based on Ubuntu18.04 LTS.
[image]

Thanks.
Mitsuhide

Kazzz-S, 2020-02-25:
Here are two more cases tested on Mac.

### Version 0.26.3 on Catalina reports 35 (with duplications) violations. ###
[image]

### Version 0.25.9 on Yosemite reports 28 (correct) violations. ###
[image]

Mitsuhide, my colleague, is supposed to add some Valgrind results later.

Regards,
Kazzz

klayoutmatthias, 2020-02-26:
Thanks a lot for this nice bug report!

I'll take a look asap. I need a little time.

Best regards,

Matthias

klayoutmatthias, 2020-02-26:
Hi kazzz, hi Mitsuhide,

The bug you found is another effect of issue #481 (although your bug report is much better than mine!).

This bug is already fixed in master branch and the fix will be released soon (there is only #482 on my list for 0.26.4). I could confirm your issue is no longer seen on the master branch.

Best regards,

Matthias

Kazzz-S, 2020-02-27:
Hi Matthias,

Thanks for your prompt response.
I have built the master branch f1aca46a on **macOS Catalina** and confirmed that the issue had been fixed.
[image]

Mitsuhide is going to test the same branch on the Linux.

With warm regards,
Kazzz

Mitsuhide-T, 2020-02-28:
Hi Matthias,

I have built the master branch(download the zip on 2020-02-28) on Linux Mint 19.1 and confirmed that the issue had been fixed.
So I'm closing this issue. 
[image]

I'm looking forward to the next version.

Kind regards,
Mitsuhide

closed 2020-02-28 by Mitsuhide-T