KLayout/klayout #513: RuntimeError: Cannot determine format from filename in SaveLayoutOptions.set_format_from_filename

tkoyat, 2020-02-25:
I used the packaging tool pyinstaller to package the code, causing a format problem:

Traceback (most recent call last):
  File "1.py", line 27, in <module>
    value = opt.set_format_from_filename(file_gds)
RuntimeError: Cannot determine format from filename in SaveLayoutOptions.set_format_from_filename
[7023] Failed to execute script 1

klayoutmatthias, 2020-02-26:
What is the value of "file_gds"?

klayoutmatthias, 2020-03-08:
I think the problem is related to the pyinstaller tool and the way it packages the libraries. It does not package the dynamically-linked stream reader plugins. 

Please don't use pyinstaller. Use setup.py/distutils instead.

Matthias

closed 2020-03-08 by klayoutmatthias