KLayout/klayout #516: Ruler can no longer be selected with a key binding

trevorstirling, 2020-02-27:
I recently updated from Klayout 0.23.11 to 0.26.3 on Mac.

In 0.23.11 you could select the ruler by going to Edit>Mode>Ruler, but in 0.26.3 you no longer can (you can go to Edit>Mode>Ruler>Ruler to select the ruler type, but not to select it as the active mode). This breaks key bindings to quickly select Ruler as a shortcut, so it now must be manually selected from the menu bar and can not be assigned a shortcut.

Kazzz-S, 2020-02-28:
Hi,

I've tested Version 0.26.3 on **Catalina** by assigning **Key-[R] to Ruler**.
**Pressing Key-[R]**:
[image]
**Pressing Key-[ESC]**:
[image]
**Pressing  Key-[R] and Key-[ESC] alternately**:
[Issue-516-Movie](https://github.com/KLayout/klayout/files/4264682/Issue-516.zip)

One suggestion is to make **$HOME/.klayout/klayoutrc** afresh (rename it then restart KLayout) because you made a long jump from  0.23.11 to 0.26.3.

Regards,
Kazzz

trevorstirling, 2020-02-28:
Hi Kazzz,

Thanks for the quick reply. I am also running on Catalina, and have assigned "Ctrl+R" to edit_menu.mode_menu.ruler. I tried renaming klayoutrc and letting Klayout generate a new one, but this did not fix my issue.

When I press Command+R (the Mac equivalent of Ctrl+R) the edit menu at the top flashes to indicate that it has selected the menu item Edit>Mode>Ruler, however the current mode stays on "Select".

See attached video of me both pressing Command+R and trying to manually select the ruler from the edit menu.

Thanks,
Trevor
[Issue-516-reply.zip](https://github.com/KLayout/klayout/files/4264720/Issue-516-reply.zip)

Kazzz-S, 2020-02-29:
Hi Trevor,

Thanks for uploading the video. Your problem is clear.
Regarding the first half part of the video (shortcut key), I could reproduce your problem.
It looks **Ctrl+R** is strongly bonded to **file_menu.reload** internally.
This issue does not reproduce on Windows (maybe Linux, neither). So, I think this is macOS-specific.
Perhaps Qt-related but not sure.
**Ctrl+R** does not work as we expected, but **Alt+R** does. Can you change your shortcut preference?

As for the second half part (Edit > Mode > Ruler), the video captures the regular behavior;
it is the same as on Linux and Windows.
If the present mode is **Ruler** in the main menu bar, then **Edit > Mode > Cross** takes effect as expected.

By the way, a set of customized rulers are used in my team.
The next comment might be useful for you (to be continued).

Regards,
Kazzz

Kazzz-S, 2020-02-29:
Hi Trevor,

In my company, KLayout users need some customized rulers to check their designs.
For them, minimizing key and mouse operations is crucial to gain work speed.
The green-color ruler icons in the image below are such tools where Key-[R] is the shortcut to **Ruler**,
Key-[C] to **Cross**, and Key-[M] to **Measure**.
[image]
Please refer to the [demo video](https://github.com/KLayout/klayout/files/4269755/CustomizedRulers.m4v.zip), too.

If you are interested in this, refer to the [attachment](https://github.com/KLayout/klayout/files/4269756/Issue-516-20200229.zip), which was prepared by my colleague @Mitsuhide-T.

To use the attachment,
1. copy files and directory in **group1** to **$HOME/.klayout/ruby/**
[image]
2. copy **MyKLayout.app** application (Bash) bundle in **group2** to **/Applications**.

Regards,
Kazzz

trevorstirling, 2020-02-29:
Hi Kazzz,

Thank you, your solution worked well. Strangely if I change the shortcut from "R" to "Ctrl+R" for the new menu command *Ruler (in the CustomizedRulers.rb file), this seems to be enough to override the strong binding to file_menu.reload and I am then able to use Command+R to select the ruler as I initially desired. This is similarly the case if I add the new command as an option in the tools menu rather than in the toolbar.

Thanks again,
Trevor

closed 2020-02-29 by trevorstirling