KLayout/klayout #519: MacOS build for 0.26.4

Kazzz-S, 2020-03-07:
## Mac DMGs for 0.26.4 ##

Hi Matthias (@klayoutmatthias) ,

Different DMG packages of **KLayout 0.26.4 for Mac** are available as below ( [0.26.4 Dropbox Link](https://www.dropbox.com/sh/54i3ymnc42cwh93/AAC_SBjnkUuUgSSjq6oI-ptta?dl=0))
**Please note that _pymod_ is not included in those packages.**

|Srl.No.|Target OS (1)|DMG file name (2) |Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
1|ElCapitan (10.11)|ST-klayout-0.26.4-MacOSX-ElCapitan-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled| (3) |
2|Sierra (10.12)|ST-klayout-0.26.4-macOS-Sierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|
3|HighSierra ( 10.13)|ST-klayout-0.26.4-macOS-HighSierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|
4|Mojave (10.14)|ST-klayout-0.26.4-macOS-Mojave-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|
5|Catalina (10.15)|ST-klayout-0.26.4-macOS-Catalina-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|

(1) Yosemite (10.10) has been dropped due to the compiler issue reported in #360.
(2) DMG file naming convention has been changed from earlier (< 0.26.x) releases, as shown below.
**0.26.4 for ElCapitan** is taken as an example here.
[image]

where
(a) Type of package:  ST-: standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt5 type:  qt5MP: MacPorts; qt5Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:    sys: OS-bundled; hb27: Ruby2.7 from Homebrew; ana3: Ruby2.5 from Anaconda3
(f) **P**ython type: sys: OS-bundled; hb37: Python 3.7 from Homebrew; ana3: Python3.7 from Anaconda3

(3) Only Qt's frameworks are included in the DMG package, which is the standard (ST-).

----
The three DMG packages below are experimental and available only for Catalina (10.15) but may be useful for some users.

|Srl.No.|Target OS|DMG file name (2) |Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
6|Catalina (10.15)|HW-klayout-0.26.4-macOS-Catalina-1-qt5Brew-RsysPhb37.dmg|Homebrew|OS-bundled|Homebrew|(4)|
7|Catalina (10.15) with Homebrew dev. env.|LW-klayout-0.26.4-macOS-Catalina-1-qt5Brew-Rhb27Phb37.dmg|Homebrew|Homebrew|Homebrew| (5) |
8|Catalina (10.15) with Anaconda3 dev. env.|LW-klayout-0.26.4-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(6)|

(4) Both Qt5 and Python3.7 frameworks are included, hence, heavyweight. As for Ruby, the OS-bundled one is used.  [The known issue](https://github.com/KLayout/klayout/issues/460#issuecomment-587968841) is not yet fixed.

(5) None of Qt, Ruby, and Python frameworks are included in the DMG, hence, lightweight. This package is for those who have a Homebrew development environment.

(6) None of Qt, Ruby, and Python frameworks are included in the DMG, hence, lightweight. This package is for those who have an Anaconda3 development environment.
* I'm mainly using this package with a Bash script bundle in `script-bundles.zip`.

With warm regards,
Kazzz

klayoutmatthias, 2020-03-08:
Hi kazzz,

thanks a lot for providing these builds. I have uploaded them to the server.

If it's OK I'd close this ticket then.

Best regards,

Matthias

Kazzz-S, 2020-03-08:
Hi Matthias,

Thanks for publishing the builds.
Yes, please close this ticket.

Warm regards,
Kazzz

closed 2020-03-15 by klayoutmatthias