KLayout/klayout #523: Submenus dissapearing on Wayland

jagrieve, 2020-03-19:
Hi there,

I'm running KLayout 0.26 (from the repos) on Manjaro with Gnome DE, and the Wayland compositor. The default installed state would fail to load (core dump), but I was able to get it up and running by unsetting `XDG_SESSION_TYPE` and adding `QT_QPA_PLATFORM=wayland` to the environment variables. So far, it's been rock solid -- no crashes, few interface quirks (other than the one I'm about to describe).

Occasionally, a submenu will fail to draw. Usually it is `Edit->Selection`, and the only solution is to save all files and restart KLayout.

There is nothing much sent to stdout or stderr when this happens, though there is the message `Warning: Wayland does not support QWindow::requestActivate()` periodically. I haven't been able to isolate which user action triggers that message, and it doesn't appear to be (directly) connected to the missing submenu.

The only lead I have at this point is that sometimes, submenus are not being drawn where you would expect them -- for example:
[image]

So perhaps the submenu is attempting to draw somewhere off-screen? Happy to gather any other data if it would be helpful.

klayoutmatthias, 2020-03-24:
Thanks for reporting this.

Basically the whole UI is rendered through Qt and I don't use any specialities - at least not for the main menu. I'm using Gnome myself although not on Wayland but on X11 without seeing any issues similar to the ones you describe.

I wonder what packages you're using. Are they provided by Manjaro? Have you tried building from sources?

Matthias

jagrieve, 2020-03-26:
Thanks for your reply! The packages are from the Manjaro Community repo, which I gather is maintained by the Manjaro team themselves. I'd be happy to try building from source, though I'm in the middle of a project right now so it will have to wait a week or two.

A quick search for similar problems does throw up examples of [Wayland and Qt misbehaving](https://bugreports.qt.io/browse/QTBUG-51640), so I wonder if my hunch is correct. I ran KLayout on my previous machine (also Gnome, with X11) and didn't see this at all.