KLayout/klayout #529: Slow access to attributes in Python

amccaugh, 2020-03-30:
Hi Matthias, I'm doing some work using your standalone Python module which involves a lot of manipulation of the Trans objects.  It's surprisingly slow, and I think I've narrowed down why exactly.  If I run this example code in ipython:

```
import klayout.db as kdb
x = kdb.DCplxTrans()
%timeit x.angle
```
I get the following result:

`198 ns ± 0.616 ns per loop (mean ± std. dev. of 7 runs, 10000000 loops each)`

Which seems like an absurdly slow access time for a single value.  Any suggestions as to whether this is a bug or something intrinsic to the module?

Note this seems to be the case both on my mac and a windows PC (Python 3.7, klayout 0.26.0.dev11 in both cases)

amccaugh, 2020-03-31:
Another example, which is a little closer to my actual use case (and probably not as swamped by Python's ~100-ns-for-everything overhead), showing accessing an instance's DCplxTrans takes ~600ns and accessing+changing the `angle` attribute takes about a microsecond:

```
import klayout.db as kdb
layout = kdb.Layout()
points = [[1,2],[3,4],[9,9]]
polygon = kdb.DSimplePolygon([kdb.DPoint(x, y) for x, y in points]) # x and y must be floats
kl_cell1 = layout.create_cell('toplevel')
kl_cell2 = layout.create_cell('under')
kl_DCellInstArray = kdb.DCellInstArray(
    kl_cell2.cell_index(),
    kdb.DCplxTrans(),
    )
kl_instance = kl_cell1.insert(kl_DCellInstArray)
trans = kl_instance.dcplx_trans

%timeit -n50 -r50 kl_instance.dcplx_trans
# This took 636 ns ± 67.7 ns per loop (mean ± std. dev. of 50 runs, 50 loops each)
%timeit -n50 -r50 kl_instance.dcplx_trans.angle = 50
# This took 961 ns ± 41.8 ns per loop (mean ± std. dev. of 50 runs, 50 loops each)
```

klayoutmatthias, 2020-04-05:
Any benchmarks? Is there a comparable case which justifies a biased judgement like "absurdly slow"?

After all you're coding in Python. Don't expect performance of a compiled language.

klayoutmatthias, 2020-04-05:
Running timeit inside the application gives me on a i7-4800MQ CPU @ 2.70GHz on Ubuntu 18:

```
> timeit.timeit(setup='import pya; tr=pya.DTrans()', stmt='tr.angle', number=10000000)
0.8428268432617188
> timeit.timeit(setup='import pya; tr=pya.DTrans()', stmt='tr.angle*2', number=10000000)
1.2762889862060547
> timeit.timeit(setup='import pya; tr=pya.DTrans()', stmt='abs(tr.angle*2)', number=10000000)
1.6883089542388916
```

So the plain access is 85ns. Which isn't bad given there is an elaborate marshalling layer below. Even a simple multiplication or "abs" call in Python will add more than 40ns. I surely can bring down the time somewhat, but at the expense of killing all the assertions and sanity checks.

The instance example gives me:

> timeit.timeit(setup='import pya; ly=pya.Layout(); tc=ly.create_cell("TOP"); inst=tc.insert(pya.CellInstArray(tc.cell_index(), pya.DTrans()))', stmt='inst.dcplx_trans', number=10000000)
2.4667999744415283

So 247ns. Which also isn't bad given a object creation (DCplxTrans) and marshalling is involved.

And finally 

> timeit.timeit(setup='import pya; ly=pya.Layout(); tc=ly.create_cell("TOP"); inst=tc.insert(pya.CellInstArray(tc.cell_index(), pya.DTrans()))', stmt='inst.dcplx_trans.angle=50', number=10000000)
3.7029948234558105

(which does not make sense BTW).

And last but not least a comparable case maybe:

> timeit.timeit(setup='import numpy', stmt='x=numpy.array([])', number=10000000)
4.098313093185425

These 410ns are consistent with the numbers before.

So in general the numbers are smaller than yours and reasonable in the context of a dynamic language. If you want nanosecond performance switch to C++.

klayoutmatthias, 2020-04-05:
And finally some timing drill-down: of the roughly 80ns for the "angle attribute" access

* 40ns go to Python. This is what an empty method call cost you.
* 20ns go to Python too for the production of the result object (PyLong_FromLong for this case)
* 20ns go on my account

amccaugh, 2020-04-05:
Hi @klayoutmatthias , thanks for taking a look at this.  I'm not terribly familiar with the overhead between Python<->C interaction (much less C-object interaction times) I was just a bit confused at the numbers I was seeing, so sorry if I came across as judgemental, that was not my intention! (For the record, you've done really fantastic work with KLayout!)

I'm running this on a system with specs not too different from yours, so I'm still not clear why my values are twice as large but I'll try on a couple other systems and see if I can get something that matches your values a little closer.  Thanks again!

closed 2020-04-05 by amccaugh