KLayout/klayout #541: KLayout does not Respect X-Server dpi Settings

metroid120, 2020-04-15:
Hello all,

I have an Issue with KLayout under Arch Linux and a 4k monitor. Most other applications either respect the X-Server's dpi settings or other QT environment variables to adjust the GUI scaling. However for Klayout I have not been able to get the GUI to scale, so it is very tiny.

Thanks for you awesome work!

klayoutmatthias, 2020-04-15:
What is tiny? The icons? In case it's a Qt issue I can't fix. 

I don't own a 4k monitor either so I can't debug the issue.

Matthias

metroid120, 2020-04-16:
Hi Matthias, after a restart it worked. Sorry for reporting the Heisenbug... 
Everything was very tiny. Here a screenshot:
[image]

Qt Applications do respect the Dpi Settings of X-Server or the environment variables, see [arch wiki](https://wiki.archlinux.org/index.php/HiDPI#Qt_5) or [qt website](https://www.qt.io/blog/2016/01/26/high-dpi-support-in-qt-5-6). Works in Klayout after restart.

closed 2020-04-16 by metroid120

klayoutmatthias, 2020-04-17:
Very good :) Thanks for letting us know.

Regards,

Matthias