KLayout/klayout #552: MacOS build for 0.26.5

Kazzz-S, 2020-05-04:
Started working ...
Place holder.

Kazzz-S, 2020-05-05:
Hi @klayoutmatthias,

Please close this ticket after publishing the files.

Kazzz

klayoutmatthias, 2020-05-06:
Thanks a lot! Nicely done as always! :)

Files are on the server.

Best regards,

Matthias

closed 2020-05-06 by klayoutmatthias

lukasc-ubc, 2020-05-11:
Hi @Kazzz-S 

I get the following error when running LW-klayout-0.26.5-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg on my mac 10.15.3 with Anaconda installed:

```
(base) host58-104:MacOS lukasc$ conda -V
conda 4.8.2
(base) host58-104:MacOS lukasc$ ./klayout 
dyld: Library not loaded: @rpath/libpython3.7m.dylib
  Referenced from: /Applications/klayout 0.26.5 LW-klayout-0.26.5-macOS-Catalina-1-qt5Ana3-Rana3Pana3.app/Contents/MacOS/./klayout
  Reason: image not found
Abort trap: 6

```

lukasc-ubc, 2020-05-11:
Dear @Kazzz-S 

I am trying to use pip within KLayout to install missing packages (numpy, etc). 

There are two ways to do this:

1) the recommended way, which requires that the user has access to the pip binary: https://pip.pypa.io/en/latest/user_guide/#using-pip-from-your-program

2) using "import pip", which is not recommended.

I have been using the 2nd option successfully for a few years.

I tried this with HW-klayout-0.26.5-macOS-Catalina-1-qt5Brew-RsysPhb37.dmg, and obtain the following error:

```
No module named 'numpy'
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/numpy/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/numpy/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/numpy/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/numpy/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/numpy/
Could not fetch URL https://pypi.org/simple/numpy/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/numpy/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
ERROR: Could not find a version that satisfies the requirement numpy (from versions: none)
ERROR: No matching distribution found for numpy

```

My code looks like follows:

```
import pip
from pip import main
main(['install', 'numpy'])

```

So it looks like we are missing the SSL package in the HW distribution.  Could you please add this, as well as numpy, pyparsing, scipy (the packages I am using)?

thank you

lukasc-ubc, 2020-05-11:
I got it working after doing a brew update.  

Is there a reason why you are distributing Python2 with the regular distribution?  Could you switch it to Python3?  Otherwise, I need to ask my users to use your experimental distributions:

https://github.com/lukasc-ubc/SiEPIC-Tools/wiki/Installation

thanks

Kazzz-S, 2020-05-11:
Hi @lukasc-ubc,

Regarding your two comments above, https://github.com/KLayout/klayout/issues/552#issuecomment-626467499 and https://github.com/KLayout/klayout/issues/552#issuecomment-626477730, please refer to my previous comments https://github.com/KLayout/klayout/issues/460#issuecomment-589331601.

Due to time constraints, I **have not made any progress in Homebrew and Python3.x issue since then.** Sorry about that. Can anyone around you contribute to these issues?

Thank you,
Kazzz

Kazzz-S, 2020-05-11:
Hi @lukasc-ubc,

> Hi @Kazzz-S
>
> I get the following error when running LW-klayout-0.26.5-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg on my mac 10.15.3 with Anaconda installed:
>
> ```
> (base) host58-104:MacOS lukasc$ conda -V [...]

My environment looks slightly different from yours:
```
(base) MacBookPro2{Kazzz-S}(32)$ conda -V
conda 4.8.3
```
The following ZIP file contains a Bash script bundle **KLayoutAna3.app**.
https://www.dropbox.com/s/l7ssy0b6l809j3e/script-bundles.zip?dl=0
The contents are:
```
#!/bin/bash
#-------------------------------------------------------------------------------------
# Bundle: KLayoutAna3.app
#
# Descriptions:
#  This is to invoke "klayout-ana3.app" using Anaconda3 distributed as
#  a binary package for Mac in "editor" mode.
#-------------------------------------------------------------------------------------

#---------------------------------------------------------
# Export 'PYTHONHOME' environment variable
#---------------------------------------------------------
export PYTHONHOME=$HOME/opt/anaconda3

#---------------------------------------------------------
# With "-n" option, multiple instances can be invoked
#---------------------------------------------------------
myconfig=$HOME/.klayout/klayoutrc-ana3

open -n -a /Applications/klayout-ana3.app --args -e -c $myconfig -style=fusion $@
```

To use this script bundle, you need to rename the application as below.
```
  /Applications/klayout 0.26.5 LW-klayout-0.26.5-macOS-Catalina-1-qt5Ana3-Rana3Pana3.app
    |
    |
    V
  /Applications/klayout-ana3.app/
```

Can you try this?
Thank you,
Kazzz

Kazzz-S, 2020-05-24:
Dear @klayoutmatthias,

Regarding [this issue](https://github.com/KLayout/klayout/issues/557), I have rebuilt two DMGs with some updates, which are available under [this Dropbox directory](https://www.dropbox.com/sh/eprqsyv59asc8pr/AAA_Fao7GhMCzfX6B_ohlE7Xa?dl=0).
Please replace the currently published ones (build-1) with these new ones (build-3).

|DMG|Target OS|Shared Development Environment|
|:------:|:--------:|:--------:|
|LW-klayout-0.26.5-macOS-Catalina-3-qt5Ana3-Rana3Pana3.dmg|Catalina|Anaconda3|
|LW-klayout-0.26.5-macOS-Catalina-3-qt5Brew-Rhb27Phb37.dmg|Catalina|Homebrew|

These packages have an additional folder containing the support documents for developers.
[image]
[image]

Regards,
Kazzz-S