KLayout/klayout #557: KLayout 0.26.5 for macOS Catalina with Anaconda3

Kazzz-S, 2020-05-16:
This ticket is to address the issue raised in https://github.com/KLayout/klayout/issues/552#issuecomment-626460971.

I have rebuilt another DMG modifying the build-environment for macOS assuming the following conditions:
1. Anaconda3 is installed under $HOME/opt/anaconda3
2. There is a symbolic link: /Applications/anaconda3 ---> $HOME/opt/anaconda3/
3. **ruby** has been installed by ```conda install ruby```
4. **PYTHONHOME** environment variable is set to **/Applications/anaconda3**

Please refer to the images below.

[image]

[image]

[image]

[image]

[image]

[image]

Kazzz-S, 2020-05-16:
Dear  @lukasc-ubc,

I have uploaded the second DMG (LW-klayout-0.26.5-macOS-Catalina-2-qt5Ana3-Rana3Pana3.dmg) under [0.26.5-Tests](https://www.dropbox.com/sh/zy0pdmdg08uyiw4/AAAJvIpa6q9A9hTMCIKFFTXOa?dl=0).

Please try this one if you have time.

Regards,
Kazzz-S

siepic, 2020-05-16:
I have downloaded

> Dear @lukasc-ubc,
> 
> I have uploaded the second DMG (LW-klayout-0.26.5-macOS-Catalina-2-qt5Ana3-Rana3Pana3.dmg) under [0.26.5-Tests](https://www.dropbox.com/sh/zy0pdmdg08uyiw4/AAAJvIpa6q9A9hTMCIKFFTXOa?dl=0).
> 
> Please try this one if you have time.
> 
> Regards,
> Kazzz-S

Dear @Kazzz-S 

That works!  The way to put an environment variable and make it available system-wide, is

`launchctl setenv PYTHONHOME "/Applications/anaconda3"`

You could add that to your instructions (4) above, since it took me some searching to figure that out.

Kazzz-S, 2020-05-16:
> Dear @Kazzz-S
> 
> That works! The way to put an environment variable and make it available system-wide, is
> 
> `launchctl setenv PYTHONHOME "/Applications/anaconda3"`
> 
> You could add that to your instructions (4) above, since it took me some searching to figure that out.

Hi,

Thank you for testing the DMG package and sharing the idea to set the environment variable.

----
The Dropbox directory mentioned above includes a ZIP file `script-bundles.zip` that contains a sample Bash script bundle `KLayoutAna3.app` generated by using the OS tool **Autometer.app**. The contents are shown below.

To use this script bundle, you need to rename `/Applications/klayout.app` to `/Applications/klayout-ana3.app` after installing the DMG package.
**KLayoutAna3.app** uses its own configuration file (klayoutrc-ana3) to coexist with the
standard package that uses the OS-bundled Ruby and Python (2.7).

```
#!/bin/bash
#-------------------------------------------------------------------------------------
# Bundle: KLayoutAna3.app
#
# Descriptions:
#  This is to invoke "klayout-ana3.app" using Anaconda3 distributed as
#  a binary package for Mac in "editor" mode.
#-------------------------------------------------------------------------------------

#---------------------------------------------------------
# Export 'PYTHONHOME' environment variable
#---------------------------------------------------------
export PYTHONHOME=/Applications/anaconda3

#---------------------------------------------------------
# With "-n" option, multiple instances can be invoked
#---------------------------------------------------------
myconfig=$HOME/.klayout/klayoutrc-ana3

open -n -a /Applications/klayout-ana3.app --args -e -c $myconfig -style=fusion $@
```

I should have added this.
Regards,
Kazzz-S

Kazzz-S, 2020-05-24:
I have rebuilt two DMGs with some updates, which are available under [this Dropbox directory](https://www.dropbox.com/sh/zy0pdmdg08uyiw4/AAAJvIpa6q9A9hTMCIKFFTXOa?dl=0).

|DMG|Target OS|Shared Development Environment|
|:------:|:--------:|:--------:|
|LW-klayout-0.26.5-macOS-Catalina-3-qt5Ana3-Rana3Pana3.dmg|Catalina|Anaconda3|
|LW-klayout-0.26.5-macOS-Catalina-3-qt5Brew-Rhb27Phb37.dmg|Catalina|Homebrew|

These packages have an additional folder containing the support documents.

[image]
[image]

Regards,
Kazzz-S

lukasc-ubc, 2020-05-28:
@Kazzz-S 
this works beautifully!
thanks for the bundle. I like the script bundle and icon.

Kazzz-S, 2020-05-28:
@lukasc-ubc 

Thank you for testing the new package.
Glad to hear that you liked it.
I close this ticket with this comment.

Kazzz-S

closed 2020-05-28 by Kazzz-S