KLayout/klayout #559: Recompile layout, check layout changes by default

lukasc-ubc, 2020-05-20:
Hi @klayoutmatthias 

When we open an layout containing PCells,

1) it can take a long time to re-calculate PCells, and
2) there can be differences in the Library, so different users will see different things.

It would be useful if KLayout did the following:

- asks the user if they want the PCells re-calculated, or just loaded from the GDS file; this would
  - allow a faster load
  - allow a foundry to read the layout as the designer intended (similar to Save As without "context info", in this case disabling it on load rather than save).
- if re-calculated, compare the re-calculated geometry versus the geometry as saved in GDS; this would be useful for noticing PCell changes in a team of designers

Perhaps these could be a new menu, under File | Load with Options.

klayoutmatthias, 2020-05-23:
Hi Lukas,

I see the point.

I'm not sure, but I think recalculation only happens in editor mode. In viewer, recalculation should not happen. For editor mode, the recomputation is somewhat crucial. If recomputation only happens upon a change, this would cause seemingly random updates in the layout, specifically in places where a change is not expected (if a PCell is used in other places too). To avoid this, my reasoning was that the best thing to do is to restore a consistent state before.

I'd like to propose versioning for the PCell implementation. I imagine the PCell class should communicate a version number which is changed when the PCell code changes substantially. This way, the PCell reader could identify the need for updates on a need basis rather then unconditionally. If there is no need to update, this would not happen.

Still an explicit recomputation makes sense. I though I had a function for this already, but maybe this was only available in script API.

Best regards,

Matthias

lukasc-ubc, 2020-06-02:
Hi Matthias,

Versioning is a good idea.  That would speed up loading layouts where the PCell version is the same.

The reader open is interesting, but doesn't address my use case:
- loading people's designs (without touching them)
- merging them

Presently I added a menu item to SiEPIC-Tools called "Export for fabrication"

```
  # Save the layout, without PCell info, for fabrication
  save_options = pya.SaveLayoutOptions()
  save_options.write_context_info=False  # remove $$$CONTEXT_INFO$$$
  save_options.format='GDS' # standard format
  save_options.format='OASIS' # smaller file size
  save_options.oasis_compression_level=10
  file_out = os.path.join(os.path.dirname(layout_filename), os.path.splitext(os.path.basename(layout_filename))[0]+'_exported.'+save_options.format[0:3].lower())
  ly.write(file_out,save_options)

```

EmaOPTIUS, 2020-08-27:
Hi Matthias and Lukas,

As mentioned, when opening a design, updated PCells will be modified.
How difficult would it be to automatically identify every updated instance? A very user-friendly solution could be to show a dialog window listing all instance which were updated in the current design. This would it make very easy to verify the integration of all those instance in the current design, and fix errors if necessary.