KLayout/klayout #560: Library associated with multiple Technologies

lukasc-ubc, 2020-05-22:
Presently, a Technology can have multiple libraries.

What about a Library that can function for multiple technologies?

I know there are obvious reasons not to do this.

But it is possible to make Library components that can be used in different technologies, especially when the differences between the processes are subtle.  

Any chance we could have the following?

`      self.technology=[tech_name1, tech_name2, ...]`

Then in our PCell code, the code can check which technology it is, and adjust accordingly.

An example of this would be an optical waveguide.  Similar to the "Path" function, which applies to all technologies that have metal, we have a "Waveguide" in all integrated photonic technologies.  Hence, it would be nice be able to make PCells that work for many technologies. 

Thank you
Lukas

klayoutmatthias, 2020-05-23:
Hi Lukas,

In general that should be possible. 

I just wonder how to keep these files (macros, libraries ...). I was reasoning that they are conveniently kept inside a tech folder, but with common libs, they need to be stored outside in a common place. 

Do you have a particular directory layout in mind?

Matthias

lukasc-ubc, 2020-06-02:
Hi Matthias,
I don't have a specific implementation / folder structure in mind.

closed 2020-06-05 by klayoutmatthias