KLayout/klayout #561: KLayout compilation issue with MSVC 2019

alessiocacciatori, 2020-05-22:
Dear,
  I am trying to compile Klayout under Win10 64bit with MSVC 2019 (Community) .
I don't need the entire UI, just the "building" libraries to manage GDSII and the database.
I converted the Klayout.pro and with qmake.
When building subprojects, _klayout_tl_ and _klayout_gsi_ go smooth.
When compiling _klayout_db_ , I got several C2679 errors related to std::list ( "_Assign_cast")
Errors seem to arise from dbDeepEdges.cc and dbDeepRegion.cc.
Any clue on that?
Thank you in advance for your help.
Best

klayoutmatthias, 2020-05-22:
Please provide more details like line numbers, file names and actual error messages. This reports is too vague to be considered by me.

Please note that MSVC 2019 isn't a supported build environment and I cannot give support for customized build schemes. For the supported build modes please refer to https://www.klayout.de/build.html (section "Building KLayout on Windows").

alessiocacciatori, 2020-05-23:
Dear Matthias,
  Thank you very much for your prompt reply.
I know that MSVC2019 is not a supported IDE but I had to give it a try.
I also know that this a out-of-standard compilation scheme so I am just asking at least if you can figure out what _could be_ the root cause. Not asking to fix it.

Well, about the error: it is a weird one.
When compiling the above-mentioned files (e.g. _dbDeepEdges.cc_) the error I have is actually inside **MSVC/14.26.28801/list** @ line 1285

C2679 binary '='  no operator found which takes a right-hand operator of type 'const std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<T>>,std::set<TI,std::less<TI>,std::allocator<db::edge<db::Coord>>>>,db::local_processor_cell_context<TS,TI,TR>>'

C2679 binary '='  no operator found which takes a right-hand operator of type 'const std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<T>>,std::set<TI,std::less<TI>,std::allocator<db::edge<db::Coord>>>>,db::local_processor_cell_context<TS,TI,TR>>'

C2679 binary '='  no operator found which takes a right-hand operator of type 'const std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<T>>,std::set<TI,std::less<TI>,std::allocator<db::polygon_ref<db::Polygon,db::Disp>>>>,db::local_processor_cell_context<TS,TI,TR>>'

What is puzzling me is the fact that the error lies in this snippet of code of the Windows file "list"

    template <class _Target_ref, class _UIter>
    void _Assign_cast(_UIter _UFirst, const _UIter _ULast) {
        // assign [_UFirst, _ULast), casting existing nodes to _Target_ref
        const auto _Myend = _Mypair._Myval2._Myhead;
        auto _Old         = _Myend->_Next;
        for (;;) { // attempt to reuse a node
            if (_Old == _Myend) { // no more nodes to reuse, append the rest
                _List_node_insert_op2<_Alnode> _Op(_Getal());
                _Op._Append_range_unchecked(_UFirst, _ULast);
                _Op._Attach_at_end(_Mypair._Myval2);
                return;
            }

            if (_UFirst == _ULast) {
                // input sequence was shorter than existing list, destroy and deallocate what's left
                _Unchecked_erase(_Old, _Myend);
                return;
            }

            // reuse the node
            **reinterpret_cast<_Target_ref>(_Old->_Myval) = *_UFirst;**
            _Old                                        = _Old->_Next;
            ++_UFirst;
        }
    }

(Actual error line in bold)

Unfortunately, there's no indication of the line code inside dbDeepEdges that is causing the error.
So my feeling is that there some _list_ inside dbDeepEdges  calling operator = with elements that do not have an overloaded = operator.

Again I am not asking for anyone to fix it, but since you write the code, probably you can give me at least the right direction to look at.

Thank you in advance, 
 Best

alessiocacciatori, 2020-05-23:
sorry: the acutal line that is an error is
reinterpret_cast<_Target_ref>(_Old->_Myval) = *_UFirst;

klayoutmatthias, 2020-05-23:
Hi,

So then, why do you file a bug report when you don't expect a fix?

These compilation problems are caused be the instantiation of a template. In this case it's probably the assignment of a std::list to another. 

But there has to be an indication of the original code line - the one where the instantiation happens. Usually there are additional messages after the error, saying what is instantiated and from where. I you want me to help, please provide the full log. Pieces are not useful.

Matthias

alessiocacciatori, 2020-05-24:
Sorry about the bug report. I am new to github. If you want me to close this issue, that's ok.
As said, I was just trying to see if you have some quick thought about this.
Anyway, this is the full line that I got

1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include\list(1285,1): error C2679: binary '=' no operator found which takes a right-hand operator of type 'const std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<T>>,std::set<TI,std::less<TI>,std::allocator<db::edge<db::Coord>>>>,db::local_processor_cell_context<TS,TI,TR>>'. 
1>        with
1>        [
1>            T=db::array<db::CellInst,db::Trans>,
1>            TI=db::Edge,
1>            TS=db::Edge,
1>            TR=db::EdgePair
1>        ]
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include\utility(233,11): message : it could be 'std::pair<_Kty,_Ty> &std::pair<_Kty,_Ty>::operator =(volatile const std::pair<_Kty,_Ty> &)'
1>        with
1>        [
1>            _Kty=std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<db::array<db::CellInst,db::Trans>>>,std::set<db::Edge,std::less<db::Edge>,std::allocator<db::edge<db::Coord>>>>,
1>            _Ty=db::local_processor_cell_context<db::Edge,db::Edge,db::EdgePair>
1>        ]
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include\list(1285,1): message :  while trying to match the argument list '(std::pair<_Kty,_Ty>, const std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<T>>,std::set<TI,std::less<TI>,std::allocator<db::edge<db::Coord>>>>,db::local_processor_cell_context<TS,TI,TR>>)'
1>        with
1>        [
1>            _Kty=std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<db::array<db::CellInst,db::Trans>>>,std::set<db::Edge,std::less<db::Edge>,std::allocator<db::edge<db::Coord>>>>,
1>            _Ty=db::local_processor_cell_context<db::Edge,db::Edge,db::EdgePair>
1>        ]
1>        and
1>        [
1>            T=db::array<db::CellInst,db::Trans>,
1>            TI=db::Edge,
1>            TS=db::Edge,
1>            TR=db::EdgePair
1>        ]
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include\xhash(876): message : see reference to 'void std::list<std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<T>>,std::set<TI,std::less<TI>,std::allocator<db::edge<db::Coord>>>>,db::local_processor_cell_context<TS,TI,TR>>,std::allocator<std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<T>>,std::set<TI,std::less<TI>,std::allocator<db::edge<db::Coord>>>>,db::local_processor_cell_context<TS,TI,TR>>>>::_Assign_cast<std::pair<_Kty,_Ty>&,std::_List_unchecked_const_iterator<std::_List_val<std::_List_simple_types<std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<T>>,std::set<TI,std::less<TI>,std::allocator<db::edge<db::Coord>>>>,db::local_processor_cell_context<TS,TI,TR>>>>,std::_Iterator_base0>>(_UIter,const _UIter)' of the model function
1>        with
1>        [
1>            T=db::array<db::CellInst,db::Trans>,
1>            TI=db::Edge,
1>            TS=db::Edge,
1>            TR=db::EdgePair,
1>            _Kty=std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<db::array<db::CellInst,db::Trans>>>,std::set<db::Edge,std::less<db::Edge>,std::allocator<db::edge<db::Coord>>>>,
1>            _Ty=db::local_processor_cell_context<db::Edge,db::Edge,db::EdgePair>,
1>            _UIter=std::_List_unchecked_const_iterator<std::_List_val<std::_List_simple_types<std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<db::array<db::CellInst,db::Trans>>>,std::set<db::Edge,std::less<db::Edge>,std::allocator<db::edge<db::Coord>>>>,db::local_processor_cell_context<db::Edge,db::Edge,db::EdgePair>>>>,std::_Iterator_base0>
1>        ]
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\include\xhash(875): message : vedere il riferimento all'istanza 'void std::list<std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<T>>,std::set<TI,std::less<TI>,std::allocator<db::edge<db::Coord>>>>,db::local_processor_cell_context<TS,TI,TR>>,std::allocator<std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<T>>,std::set<TI,std::less<TI>,std::allocator<db::edge<db::Coord>>>>,db::local_processor_cell_context<TS,TI,TR>>>>::_Assign_cast<std::pair<_Kty,_Ty>&,std::_List_unchecked_const_iterator<std::_List_val<std::_List_simple_types<std::pair<const std::pair<std::set<db::CellInstArray,std::less<db::CellInstArray>,std::allocator<T>>,std::set<TI,std::less<TI>,std::allocator<db::edge<db::Coord>>>>,db::local_processor_cell_context<TS,TI,TR>>>>,std::_Iterator_base0>>(_UIter,const _UIter)' della funzione modello di cui è in corso la compilazione
1>        with
1>        [
1>            T=db::array<db::CellInst,db::Trans>,
1>            TI=db::Edge,
1>            TS=db::Edge,
1>            TR=db::EdgePair,
[... cut ...]
1>C:\projects\hpdg\klayout\src\db\db\dbHierProcessor.h(444): message : vedere il riferimento all'istanza 'db::local_processor_cell_contexts<TS,TI,TR>' di modello classe di cui è in corso la compilazione
1>        with
1>        [
1>            TS=db::Edge,
1>            TI=db::Edge,
1>            TR=db::EdgePair
1>        ]
1>C:\projects\hpdg\klayout\src\db\db\dbDeepEdges.cc(1628): message : vedere il riferimento all'istanza 'db::local_processor<db::Edge,db::Edge,db::EdgePair>' di modello classe di cui è in corso la compilazione

klayoutmatthias, 2020-05-24:
Funny how the messages blend from English to Italian ... :)

Basically I think something is wrong with Microsoft's implementation of ```std::unordered_map```. Internally it seems to utilize a ```std::list<std::pair<const Key, Value> >``` as the basic container, but then it tries to assign a ```std::pair<Key, Value>``` (without const for the Key). Without insight into the STL sources it's hard to say if this is compiler issue or a STL implementation bug.

KLayout utilizes a complex variant of that map, but all compiler combinations I am using are able to digest it. This includes many versions of gcc and clang along with a variety of STL implementations plus MSVC 2017 with Microsoft's STL.

I think there is little one can do from KLayout's code base. In the worst case one could utilize a third-party implementation of ```std::unordered_map```. But as the problem appears to be on Microsoft's side and MSVC2017 is available, I don't want to support this path.

Matthias

alessiocacciatori, 2020-05-25:
Ah! Sorry about mixed languages, I left some sentence in Italian by mistake.
Still, I see you grasped the meaning.
And thank you for your feedback, this is exactly what I was looking for. From my view, this closes the issue.
Have a nice day!

closed 2020-05-25 by alessiocacciatori