KLayout/klayout #566: MacOS: no file formats in open/save (can't open/save GDS, OASIS etc.)

demisjohn, 2020-05-24:
Can't open/save into GDS format (or any other format), on the MacOS 10.13 package of v0.26.5:
https://www.klayout.org/downloads/MacOS/ST-klayout-0.26.5-macOS-HighSierra-1-qt5MP-RsysPsys.dmg

"All layout files" dropdown in open/save dialogues shows an empty list of formats:
[image]

Attached Screenshot of empty "format" dropdown list in the Save dialog for a brand new layout.
[image]

v0.26.4, the previous version, also appears to have this issue.  Renaming ~/.klayout or the klayout.de preference file doesn't fix.

This problem doesn't exist in v0.25.9, on either MacOS 10.13 or 10.14 packages.

lukasc-ubc, 2020-05-24:
It works for me on:

klayout 0.26.5 HW-klayout-0.26.5-macOS-Catalina-1-qt5Brew-RsysPhb37
and
klayout 0.26.5-macOS-Catalina-1-qt5MP-RsysPsys

I wonder if this might be due to permissions.  I notice that KLayout is often asking for permission to access various folders.  It seems like Mac OSX has increased security preventing applications from reading files.  Maybe it can’t read a file that lists the formats?

klayoutmatthias, 2020-05-24:
That's definitely not an issue of the release.

Please check your installation: the plugins for the stream formats are kept in a folder called "db_plugins" next to a file called (I think) "libklayout_db.dylib" in the installation folder. In this folder you should find the stream format drivers like "libgds2.dylib" etc.

When you run KLayout with high debug level (```klayout -d 41```) you should see the paths being scanned. On my Linux box this is:

```
Scanning for db plugins: /home/matthias/.klayout/db_plugins
Scanning for db plugins: /usr/bin/db_plugins
Scanning for db plugins: /usr/bin/x86_64-linux-gcc/db_plugins
Scanning for db plugins: /usr/lib/klayout/db_plugins
Loading plugin: /usr/lib/klayout/db_plugins/libcif.so
Registered object 'CIF' with priority 2100
Loading plugin: /usr/lib/klayout/db_plugins/libdxf.so
Registered object 'DXF' with priority 2000
Loading plugin: /usr/lib/klayout/db_plugins/libgds2.so
Registered object 'GDS2Text' with priority 1
Registered object 'GDS2' with priority 0
Loading plugin: /usr/lib/klayout/db_plugins/liblefdef.so
Registered object 'LEFDEF' with priority 500
Loading plugin: /usr/lib/klayout/db_plugins/libmag.so
Registered object 'MAG' with priority 2200
Loading plugin: /usr/lib/klayout/db_plugins/libnet_tracer.so
Registered object 'NetTracerPlugin' with priority 13000
Loading plugin: /usr/lib/klayout/db_plugins/liboasis.so
Registered object 'OASIS' with priority 10
Loading plugin: /usr/lib/klayout/db_plugins/libpcb.so
Registered object 'GerberPCB' with priority 1000
...
```

Note that on Linux, the suffix for shared objects is ".so" while it is ".dylib" on MacOS.

Matthias

demisjohn, 2020-05-25:
Running Debug level 41:
On my 2017 MacBook pro w/ Mac OS 10.14.2, I see these lines that are most similar to what you posted:

```
Initializing plugins:
  SaltController [100]
  TechnologyController [110]
  ...
  edt::PartialService [4030]
  CommonReader [10000]
  OASISReader [10000]
  OASISWriter [10000]
  GDS2Writer [10000]
  GDS2Reader [10000]
  DXFWriter [10000]
  DXFReader [10000]
  CIFWriter [10000]
  CIFReader [10000]
  BrowseShapesPlugin [10000]
  LEFDEFReader [10001]
  GDS2TextWriter [10001]
  BrowseInstancesPlugin [11000]
  MarkerBrowserPlugin [12000]
  NetTracerPlugin [13000]
  SearchReplacePlugin [20000]
  FillDialogPlugin [20000]
  ClipDialogPlugin [20000]
```

Full text output to CLI is attached.
[2020-05-24_1935_debug-41 - 2017 MacbookPro macos 10.14.2.txt](https://github.com/KLayout/klayout/files/4674993/2020-05-24_1935_debug-41.-.2017.MacbookPro.macos.10.14.2.txt)

demisjohn, 2020-05-25:
Launching with debug level 41:
On my 2017 MacBook Pro w/ MacOS 10.14.2:

full output attached, here are some interesting lines:

```
Warning: Class Instance: no Python mapping for method []=
Warning: Class Edges: no Python mapping for method (reserved word) in
Warning: Class Region: no Python mapping for method (reserved word) in
Warning: Class QJsonArray_Iterator: no Python mapping for method ->
Warning: Class QJsonArray_Const_iterator: no Python mapping for method ->
Warning: Class QJsonObject_Iterator: no Python mapping for method ->
Warning: Class QJsonObject_Const_iterator: no Python mapping for method ->
Warning: Class QJsonValuePtr: no Python mapping for method ->
Warning: Class QJsonValueRefPtr: no Python mapping for method ->
Warning: Class QMargins: no Python mapping for method =
Warning: Class QMarginsF: no Python mapping for method =
Warning: Class QMatrix: no Python mapping for method =
Warning: Class QMatrix4x4: no Python mapping for method =
Warning: Class QPageSetupDialog_Native: no Python mapping for method (reserved word) exec
Warning: Class QPoint: no Python mapping for method =
Warning: Class QPointF: no Python mapping for method =
Warning: Class QQuaternion: no Python mapping for method =
Warning: Class QSize: no Python mapping for method =
Warning: Class QSizeF: no Python mapping for method =
Warning: Class QSqlDatabase: no Python mapping for method (reserved word) exec
Warning: Class QSqlResult: no Python mapping for method (reserved word) exec
Warning: Class QVector2D: no Python mapping for method =
Warning: Class QVector3D: no Python mapping for method =
Warning: Class QVector4D: no Python mapping for method =
Warning: Class QCoreApplication_Native: no Python mapping for method (reserved word) exec
Warning: Class QDrag_Native: no Python mapping for method (reserved word) exec
Warning: Class QEventLoop_Native: no Python mapping for method (reserved word) exec
Warning: Class QGuiApplication_Native: no Python mapping for method (reserved word) exec
Warning: Class QPixmap_Native: no Python mapping for method (reserved word) not
Warning: Class QSqlQuery: no Python mapping for method (reserved word) exec
Warning: Class QTextDocument_Native: no Python mapping for method (reserved word) print
Warning: Class QThread: no Python mapping for method (reserved word) exec
Warning: Class QTransform: no Python mapping for method =
Warning: Class QApplication_Native: no Python mapping for method (reserved word) exec
Warning: Class QDialog_Native: no Python mapping for method (reserved word) exec
Warning: Class QDialog: no Python mapping for method (reserved word) exec
Warning: Class QErrorMessage: no Python mapping for method (reserved word) exec
Warning: Class QFileDialog: no Python mapping for method (reserved word) exec
Warning: Class QFontDialog: no Python mapping for method (reserved word) exec
Warning: Class QInputDialog: no Python mapping for method (reserved word) exec
Warning: Class QLineEdit_Native: no Python mapping for method (reserved word) del
Warning: Class QMessageBox: no Python mapping for method (reserved word) exec
Warning: Class QPrintPreviewDialog: no Python mapping for method (reserved word) exec
Warning: Class QPrintPreviewWidget_Native: no Python mapping for method (reserved word) print
Warning: Class QProgressDialog: no Python mapping for method (reserved word) exec
Warning: Class QWizard: no Python mapping for method (reserved word) exec
Warning: Class BrowserDialog: no Python mapping for method (reserved word) exec
Warning: Class Application: no Python mapping for method (reserved word) exec
Warning: Class QAbstractPrintDialog_Native: no Python mapping for method (reserved word) exec
Warning: Class QAbstractPrintDialog: no Python mapping for method (reserved word) exec
Warning: Class QColorDialog: no Python mapping for method (reserved word) exec
Warning: Class QPlainTextEdit_Native: no Python mapping for method (reserved word) print
Warning: Class QPrintDialog_Native: no Python mapping for method (reserved word) exec
Warning: Class QPrintDialog: no Python mapping for method (reserved word) exec
Warning: Class QTextEdit_Native: no Python mapping for method (reserved word) print
Initializing Python: 0.35 (user) 0.04 (sys) 0.407 (wall) 0.00M (mem)
```

And the most similar lines to what you posted are:

```
Initializing plugins:
  SaltController [100]
  TechnologyController [110]
  MacroController [120]
  LibraryController [150]
  FontController [160]
  MainPlugin [1000]
  ext::GerberImportPlugin [1200]
  ext::StreamImportPlugin [1300]
  ext::LEFDEFImportPlugin [1400]
  MacroEditor [1500]
  LayoutViewConfig [2000]
  GridNetPlugin [2010]
  ext::XORPlugin [3000]
  ant::Plugin [3000]
  ext::DiffPlugin [3001]
  ext::BooleanOperationsPlugin [3010]
  edt::MainService [4000]
  img::Plugin [4000]
  edt::Service(Polygons) [4010]
  edt::Service(Boxes) [4011]
  edt::Service(Texts) [4012]
  edt::Service(Paths) [4013]
  edt::Service(CellInstances) [4020]
  edt::PartialService [4030]
  CommonReader [10000]
  OASISReader [10000]
  OASISWriter [10000]
  GDS2Writer [10000]
  GDS2Reader [10000]
  DXFWriter [10000]
  DXFReader [10000]
  CIFWriter [10000]
  CIFReader [10000]
  BrowseShapesPlugin [10000]
  LEFDEFReader [10001]
  GDS2TextWriter [10001]
  BrowseInstancesPlugin [11000]
  MarkerBrowserPlugin [12000]
  NetTracerPlugin [13000]
  SearchReplacePlugin [20000]
  FillDialogPlugin [20000]
  ClipDialogPlugin [20000]
```

demisjohn, 2020-05-25:
On my 2017 MacBookPro w/ MacOS 10.14.2, after running in `-d` mode (and accidentally launching the old 0.25.9 version a few times), the issue magically resolved itself - v0.26.5 can open all layout formats and the dropdown menus show the typical formats.

On my 2010 MacBook Pro w/ 10.13.6, however, the issue is stranger:
When I run Layout in debug mode 41 from the CLI, all the formats are available.  I was able to open a DXF drawing as normal.

However if I run **not** in debug mode, just by double-clicking the executable file, I get the empty format dropdown, and attempting to open the same DXF, I get this error reported in the OP:
[image]

Any other info I can provide to debug?
I suppose I can download the 10.14 Klayout version and try running that on 10.13 and see if the issue occurs (each laptop has the different Layout ditribution on it, matching the different OS versions).

Kazzz-S, 2020-05-25:
Hi @demisjohn,

I have built the **High Sierra** package on my VMware virtual machine (the host is MacBook Pro Mid. 2012 w/ Yosemite 10.10.5), and the issue does not reproduce, as shown below. It's weird.

## OS and Package ##
[image]
[image]

## Open ##
[image]

## Save as ##
[image]
[image]

## -d 41 ##
[klayout-0.26.5-HighSierra-d41.log.zip](https://github.com/KLayout/klayout/files/4676569/klayout-0.26.5-HighSierra-d41.log.zip)

```
Scanning for db plugins: /Users/sekigawa/.klayout/db_plugins
Scanning for db plugins: /Users/sekigawa/.klayout/x86_64-linux-clang/db_plugins
Scanning for db plugins: /Applications/klayout.app/Contents/MacOS/db_plugins
Loading plugin: /Applications/klayout.app/Contents/MacOS/db_plugins/libcif.0.dylib
Registered object 'CIF' with priority 2100
Loading plugin: /Applications/klayout.app/Contents/MacOS/db_plugins/libdxf.0.dylib
Registered object 'DXF' with priority 2000
Loading plugin: /Applications/klayout.app/Contents/MacOS/db_plugins/libgds2.0.dylib
Registered object 'GDS2Text' with priority 1
Registered object 'GDS2' with priority 0
Loading plugin: /Applications/klayout.app/Contents/MacOS/db_plugins/liblefdef.0.dylib
Registered object 'LEFDEF' with priority 500
Loading plugin: /Applications/klayout.app/Contents/MacOS/db_plugins/libmag.0.dylib
Registered object 'MAG' with priority 2200
Loading plugin: /Applications/klayout.app/Contents/MacOS/db_plugins/libnet_tracer.0.dylib
Registered object 'NetTracerPlugin' with priority 13000
Loading plugin: /Applications/klayout.app/Contents/MacOS/db_plugins/liboasis.0.dylib
Registered object 'OASIS' with priority 10
Loading plugin: /Applications/klayout.app/Contents/MacOS/db_plugins/libpcb.0.dylib
Registered object 'GerberPCB' with priority 1000
:
:
```
No idea about additional info to debug. Sorry about that.
Regards,
Kazzz-S

P.S. This may not help, but what would happen if you remove a white space character (maybe) between **DeepSpace** and **Lubin** in the directory name?

demisjohn, 2020-05-26:
On the 2010, 10.13 MacBook Pro:
db_plugins folder is present, with appropriate items:

[image]

Looks like the `-d 41` debug output shows that it is loading l-plugins, AND the Open dialogue shows the expected file formats:
[image]

---
Again, simply double-click the Layout icon results in NO file formats available (how can I get debugging output in this state?)
[image]

------

If I launch layout from the command-line with NO debug, just `./klayout` I can now see all the file formats in the dropdown, and it acts as normal!
Able to load a GDS layout like normal.

SO: On Mac OS 10.13, **what is different about double-clicking the icon versus launching via CLI?**
And on Mac OS 10.14, what happened such that the OS magically fixed the difference?

klayoutmatthias, 2020-05-26:
Hi,

I think the CLI version works because of the current directory.

I'm not a Mac user myself but I recall KLayout is usually called through some script on MacOS. Maybe running that directly isn't supported at all?

Technically the reason might be the dynamic linker load path (LD_LIBRARY_PATH on Linux, DYLD_LIBRARY_PATH on MacOS). Maybe setting this environment variable to the installation path of KLayout will give you the stream formats.

On Linux the installation path is predefined and built into the binary (RPATH). Hence this isn't a problem. On Windows, DLL search includes relative to the binary, so that's not a problem either. On Mac I can't tell.

Matthias

demisjohn, 2020-05-27:
Could it be picking up a different python installation (both latops have _macports_ python27 & python35 installed from macports), which messed up the _cwd_?

That is perhaps one thing I have on both laptops that could be uncommon to many users.

Kazzz-S, 2020-05-27:
> Could it be picking up a different python installation (both latops have _macports_ python27 & python35 installed from macports), which messed up the _cwd_?
> 
> That is perhaps one thing I have on both laptops that could be uncommon to many users.

I think you have installed **ST-klayout-0.26.5-macOS-HighSierra-1-qt5MP-RsysPsys.dmg**.
This package should attempt to link the OS-bundled Ruby and Python; not from MacPorts.
But Qt5 Frameworks from MacPorts are embedded.
Have you globally set **PYTHONHOME** environment variable to the Python from MacPorts under /opt/local/?

I'll post the next comment shortly.

Kazzz-S

Kazzz-S, 2020-05-27:
Hi,

If possible, please try the following steps.
**1. OS Update (released today!)**
[image]

**2. Completely uninstall klayout.app (including old versions if any)** 
[image]

**3. Also move $HOME/.klayout directory**

**4. Reinstall klayout.app from the DMG installer**
[image]

**5. Open klayout.app from Finder**
* I could normally start the tool. OPEN is also as expected. The issue does not reproduce.

**6. Using script bundles: this is optional**
* Visit [Dropbox](https://www.dropbox.com/sh/eprqsyv59asc8pr/AAA_Fao7GhMCzfX6B_ohlE7Xa?dl=0)
* Download ``script-bundles.zip`` and unzip it. Then, you will find:
[image]
* Copy (install) ``KLayoutEditor.app`` and ``KLayoutViewer.app`` to ``/Applications``
[image]
* These **script bundles** invokes ``klayout.app`` with some arguments.
* With "-e" option, klayout.app starts in the editor mode; "-ne" option starts it in the viewer mode.
* The core is nothing but a simple Bash script that you can edit using **Automator.app** tool.
* So can pass the debug option if required.
* This is **the script** that Matthias mentioned above.
[image]

Kazzz-S

Kazzz-S, 2020-05-27:
Hi @demisjohn,

Knowing that you are a MacPorts user, I have built a special package for HighSierra.
Please give a shot at it if you like.

----

Visit [Dropbox](https://www.dropbox.com/sh/zy0pdmdg08uyiw4/AAAJvIpa6q9A9hTMCIKFFTXOa?dl=0) and download ``LW-klayout-0.26.5-macOS-HighSierra-4-qt5MP-Rmp26Pmp37.dmg``.

This package is special because it:
* contains only KLyout related binaries
* shares the MacPorts development environment regarding Qt5, Ruby, and Python.
```
HighSierra:MacOS sekigawa$ otool -L klayout | grep qt5
	/opt/local/libexec/qt5/lib/QtDesigner.framework/Versions/5/QtDesigner (compatibility version 5.14.0, current version 5.14.2)
	/opt/local/libexec/qt5/lib/QtPrintSupport.framework/Versions/5/QtPrintSupport (compatibility version 5.14.0, current version 5.14.2)
	/opt/local/libexec/qt5/lib/QtMultimediaWidgets.framework/Versions/5/QtMultimediaWidgets (compatibility version 5.14.0, current version 5.14.2)
	/opt/local/libexec/qt5/lib/QtSvg.framework/Versions/5/QtSvg (compatibility version 5.14.0, current version 5.14.2)
	/opt/local/libexec/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets (compatibility version 5.14.0, current version 5.14.2)
	/opt/local/libexec/qt5/lib/QtMultimedia.framework/Versions/5/QtMultimedia (compatibility version 5.14.0, current version 5.14.2)
	/opt/local/libexec/qt5/lib/QtGui.framework/Versions/5/QtGui (compatibility version 5.14.0, current version 5.14.2)
	/opt/local/libexec/qt5/lib/QtXmlPatterns.framework/Versions/5/QtXmlPatterns (compatibility version 5.14.0, current version 5.14.2)
	/opt/local/libexec/qt5/lib/QtNetwork.framework/Versions/5/QtNetwork (compatibility version 5.14.0, current version 5.14.2)
	/opt/local/libexec/qt5/lib/QtXml.framework/Versions/5/QtXml (compatibility version 5.14.0, current version 5.14.2)
	/opt/local/libexec/qt5/lib/QtSql.framework/Versions/5/QtSql (compatibility version 5.14.0, current version 5.14.2)
	/opt/local/libexec/qt5/lib/QtCore.framework/Versions/5/QtCore (compatibility version 5.14.0, current version 5.14.2)

HighSierra:MacOS sekigawa$ otool -L klayout | grep ruby
	/opt/local/lib/libruby.2.6.dylib (compatibility version 2.6.0, current version 2.6.6)

HighSierra:MacOS sekigawa$ otool -L klayout | grep Python
	/opt/local/Library/Frameworks/Python.framework/Versions/3.7/Python (compatibility version 3.7.0, current version 3.7.0)

```
As you can see above, you need to install (1) qt5, (2) ruby26, (3) python37 by ``sudo port install``.

[image]
[image]

Kazzz-S

demisjohn, 2020-05-28:
(on the 2010 MacOS 10.13.6): 

OK I figured it out!

I usually keep Klayout in a subfolder `/Applications/KLayout ƒ` (I keep old KLayout versions in case of issues like this).  Changing the directory to `/Applications/KLayout f` (regular 'F') fixed the issue.  So having an unusual character in the path to the executable breaks the file formats for some reason.  

While Launching from CLI (even with the strange directory name) does not encounter this problem!  (For example, I can paste the following into the CLI and get a fully functional Klayout: `/Applications/KLayout\ ƒ/klayout.app/Contents/MacOS/klayout 
`)

(I've been using that folder name for years without issue, go figure...)
--------------

If I do `env` I see no variable named `PYTHONHOME` or anything related (nothing at all to do with Python it seems).

My available python commands (tab-complete options):
```
<<demis@TheLionofJudah:MacOS [20:10:31]> python  <<Tab,Tab>>
python             python3            python3.6m-config  pythontex
python-config      python3-config     python3.8          pythonw
python.app         python3.6          python3.8-config   pythonw2.7
python2.7          python3.6-config   python3m           
python2.7-config   python3.6m         python3m-config    
<<demis@TheLionofJudah:MacOS [20:10:31]> which python
/opt/local/bin/python
```

FYI, I tried the suggested fixes - using AppCleaner to remove KLayout, removing `~/.klayout`, re-downloading and right-click>Open (while holding ⇧ to enable the "open from unidentified developer") - same behavior.  Still no file-formats in the open/save dialog, but from CLI running ` /Applications/KLayout\ ƒ/klayout.app/Contents/MacOS/klayout` _does_ show all the expected file formats.

closed 2020-05-28 by demisjohn

demisjohn, 2020-05-28:
I closed this, but I assume you don't necessarily want KLayout to be sensitive to OS-permitted but non-ASCII characters.  However, I assume this is merely an edge-case, so ok to leave unfixed.

I hope it doesn't point to other possible breakages in the future, as something certainly changed from 0.25.x to 0.26.x to cause this to break.  I leave that to you both to decide.

Thanks for helping me try things, sorry for not noticing that before.