KLayout/klayout #573: Help to install deb package on Ubuntu 20.04

leoheck, 2020-05-29:
Hi, I am trying to install the `klayout_0.26.5-1_amd64.deb` on  `Ubuntu 20.04`
Do you have any idea how to work around these issues?

I tried to find these packages but they are not available on `Ubuntu 20.04` with apt
Do you see as a good idea to install .deb from previews Ubuntu versions to solve this?

[image]

klayoutmatthias, 2020-05-30:
Hi,

There is no package for Ubuntu 20.04LTS yet. You can build from sources if you need so.

Please don't use the bug tracker for these support requests.

Matthias

closed 2020-05-30 by klayoutmatthias

leoheck, 2020-05-31:
Sure! This is sad about the GitHub, this is a collaborative place, the issues-place should be used for this reason. People came and go from different places. 

Thanks for the info tho.

BUT... do you mind in teach me where I can ask this kind of issue?