KLayout/klayout #577: MainWindow.MarkerBrowser

lukasc-ubc, 2020-06-02:
Dear @klayoutmatthias 

I am using the Marker database, and in scripts, call the show function:

`        lv.show_rdb(rdb_i, cv.cell_index)`

However, if the window has been closed by the user, this function doesn't re-open the window, the user needs to go to the menu to find it.

Would it be possible to have a function call that opens the the Marker Browser window and brings it to the front?

Similar to the "show_macro_editor" function:
https://www.klayout.de/doc-qt5/code/class_MainWindow.html#method157

Thanks
Lukas

klayoutmatthias, 2020-06-02:
I'd even say that's a bug ... I think "show_rdb" should open the marker browser window when it was closed manually.  At least, the "show" indicates so :)

In contrast to the macro editor, the marker browser is something that is sticky to a layout view. The layout view keeps a state of "I need to show a marker browser". So when you switch tabs, the marker browser will close or open again. I think it's just a matter of managing this state properly in the "show_rdb" function.

klayoutmatthias, 2020-06-04:
Hi,

I'm having issues reproducing the problem.

Here is what I do:
* Load a layout
* Open the Marker Database Browser and load a marker database
* I close the Marker Database Browser Window - either with the window's close button on with "Close"
* In The Macro IDE's console I use (Ruby): ```RBA::LayoutView::current.show_rdb(0, 0)``` which is the first RDB loaded into the first layout in the current view

As expected calling "show_rdb" opens the Marker Database Browser (on Linux).

So maybe that is an OS issue?

Matthias