KLayout/klayout #585: KLayout 0.26.6 crashes on macOS Catalina 10.15.5

Kazzz-S, 2020-06-06:
Dear @klayoutmatthias,

Thank you for making the maintenance release of 0.26.6.
I have started building it as usual for macOS (Catalina to begin with) and got a crash when opening a design file. [Dropbox with intermediate works](https://www.dropbox.com/sh/5we05ajmxts5qs7/AAAy-b0r9-MzUf1Ndfsx1J-ia?dl=0)

- Qt: 5.14.2 from MacPorts
- Ruby: 2.6.3 OS-bundled
- Python: 2.7.16 OS-bundled

[image]

**Full-Text**
```
Signal number: 11
Address: 0x120
Program Version: KLayout 0.26.6 (2020-06-06 r3f4c2a7e)

Backtrace:
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x204fda _ZN3lay25enable_signal_handler_guiEb
/usr/lib/system/libsystem_platform.dylib +0x35fd _sigtramp
/usr/lib/system/libsystem_platform.dylib +0xffff8001a34224dc _sigtramp
/Applications/klayout.app/Contents/Frameworks/libklayout_laybasic.0.dylib +0x25942b _ZN3lay10LayoutView16select_cellviewsERKNSt3__14listINS_8CellViewENS1_9allocatorIS3_EEEE
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x13ff1e _ZThn16_N3lay15HelpAboutDialogD0Ev
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x142817 _ZThn16_N3lay15HelpAboutDialogD0Ev
/Applications/klayout.app/Contents/Frameworks/libklayout_tl.0.dylib +0x66ace _ZN2tl23DeferredMethodScheduler10do_executeEv
/Applications/klayout.app/Contents/Frameworks/libklayout_tl.0.dylib +0x73cdf _ZN2tl25DeferredMethodSchedulerQt5timerEv
/Applications/klayout.app/Contents/Frameworks/libklayout_tl.0.dylib +0x73ca4 _ZN2tl25DeferredMethodSchedulerQt5eventEP6QEvent
/Applications/klayout.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0xc32e _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent
/Applications/klayout.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets +0xd1e8 _ZN12QApplication6notifyEP7QObjectP6QEvent
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x595bb _ZN3lay14GuiApplication6notifyEP7QObjectP6QEvent
/Applications/klayout.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x133218 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent
/Applications/klayout.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x133c51 _ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThreadData
/Applications/klayout.app/Contents/PlugIns/platforms/libqcocoa.dylib +0x2dc0c qt_plugin_instance
/Applications/klayout.app/Contents/PlugIns/platforms/libqcocoa.dylib +0x2e1b6 qt_plugin_instance
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x83de2 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x83d81 __CFRunLoopDoSource0
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x83b9b __CFRunLoopDoSources0
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x828ca __CFRunLoopRun
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x81ece CFRunLoopRunSpecific
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x2fabd RunCurrentEventLoopInMode
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x2f6f4 ReceiveNextEventCommon
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x2f579 _BlockUntilNextEventMatchingListInModeWithFilter
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x41829 _DPSNextEvent
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x40070 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x31d7e -[NSApplication run]
/Applications/klayout.app/Contents/PlugIns/platforms/libqcocoa.dylib +0x2d4af qt_plugin_instance
/Applications/klayout.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x130f47 _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE
/Applications/klayout.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x133679 _ZN16QCoreApplication4execEv
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x59f83 _ZN3lay14GuiApplication4execEv
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x58d4e _ZN3lay15ApplicationBase3runEv
/Applications/klayout.app/Contents/MacOS/klayout +0xf1cc
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1b98f _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x17390a rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x1623fe rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x170265 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5dbe8 ruby_exec_node
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5dae9 ruby_run_node
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1b925 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/Applications/klayout.app/Contents/MacOS/klayout +0xeba5
/usr/lib/system/libdyld.dylib +0x1acc9 start
/usr/lib/system/libdyld.dylib +0xffff80008df80003 start

```

- Unfortunately, **Valgrind** does not support Catalina as of today.

- If I forcibly continue the operation, the navigator sub-window remains blank.
[image]

**cf. in 0.26.5**
[image]

**Starting from the command line and open a design**
There is an apparent difference as below.
```
== 0.26.6: ST-klayout-0.26.6-macOS-Catalina-1-qt5MP-RsysPsys.dmg ==
    :
    Warning: Populating font family aliases took 285 ms. Replace uses of missing font family "Monospace" with one that exists to avoid this cost.
    Loading file: /Users/sekigawa/KLayout/Abacus.gds with technology:
!!! sh: addr2line: command not found
    ERROR: Exception caught: Operation cancelled

== 0.26.5: ST-klayout-0.26.5-macOS-Catalina-1-qt5MP-RsysPsys.dmg ==
    :
    Warning: Populating font family aliases took 287 ms. Replace uses of missing font family "Monospace" with one that exists to avoid this cost.
    Loading file: /Users/sekigawa/KLayout/Abacus.gds with technology:
```
----
So far, I have confirmed that this issue reproduces in:
1. Catalina lightweight package (LW-) sharing Anaconda3 environment
2. ElCapitan standard package (ST-)

Regards,
Kazzz-S

Kazzz-S, 2020-06-06:
## Linux Mint 19 ##

I have observed a similar (the same?) issue in the source code build on Linux Mint 19.
**Valgrind**'s report is attached.  [valgrind-Mint19.zip](https://github.com/KLayout/klayout/files/4739466/valgrind-Mint19.zip)

```
Kazzz-S@LM19:~/GitWork/klayout/bin-release$ uname -a
Linux LM19 5.3.0-53-generic #47~18.04.1-Ubuntu SMP Thu May 7 13:10:50 UTC 2020 \
                            x86_64 x86_64 x86_64 GNU/Linux

Kazzz-S@LM19:~/GitWork/klayout/bin-release$ g++ --version
g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Kazzz-S@LM19:~/GitWork/klayout/bin-release$ ruby -v
ruby 2.5.1p57 (2018-03-29 revision 63029) [x86_64-linux-gnu]

Kazzz-S@LM19:~/GitWork/klayout/bin-release$ python3 -V
Python 3.6.9

Kazzz-S@LM19:~/GitWork/klayout/bin-release$ qmake -v
QMake version 3.1
Using Qt version 5.9.5 in /usr/lib/x86_64-linux-gnu
```

[image]

Kazzz-S

Kazzz-S, 2020-06-07:
## Additional Observations ##

1. Delete (move) $HOME/.klayout/klayoutrc
2. Start KLayout afresh --> Can open a design normally
3. As soon as the **Navigator** sub-window is enabled, the crash occurs both in Catalina and Linux Mint 19
4. In the previous **klayoutrc** settings, the **Navigator** was enabled by default

Kazzz-S

klayoutmatthias, 2020-06-07:
Hi kazzz,

thanks for this information. I could reproduce the issue and fix it. It's somewhat embarrassing that this escaped me, but apparently I'm myself not a user of the navigator window :(

I have patched the 0.26.6 release now - I'm usually not doing this, but it has been a short time between the release and your notice. The Python module on PyPI is not affected.

I have
* moved the release tag to revision e71c0dcb
* updated the sources and binaries on www.klayout.de/downloads

I think with this patch, the problem should be solved.

Thanks for reporting this,

Matthias

Kazzz-S, 2020-06-07:
Hi @klayoutmatthias,

Thank you for applying the patch so quickly.
Yes, the problem has been solved as below.
[image]

I close this ticket.

Kazzz-S

closed 2020-06-07 by Kazzz-S