KLayout/klayout #586: Detection of duplicated cell names in PYA/RBA

Kazzz-S, 2020-06-08:
Dear @klayoutmatthias,

In the manual operation, it is not possible to create a new cell with a name that already exists. 
This is, of course, fine.
## Manual Operation ##
[image]
[image]

----

In my PYA similar to below, I want to create some static cells from a PCell template.
## PYA: dupCells-AAA.py ##
```
#
# Why can I have duplicated cells?
#
import pya

MainWindow  = pya.MainWindow.instance()
CurrentView = MainWindow.current_view()
Layout      = CurrentView.active_cellview().layout()

def AddMyDonut(newName):
  global Layout

  paraL  = "layer"
  paraR1 = "actual_radius1"
  paraR2 = "actual_radius2"
  paraN  = "npoints"
  params         = dict()
  params[paraL]  = pya.LayerInfo(1, 0)
  params[paraR1] = 50
  params[paraR2] = 100
  params[paraN]  = 128
  sbcell1        = Layout.create_cell( "DONUT", "Basic", params ).cell_index()
  sbcell2        = Layout.convert_cell_to_static( sbcell1 )
  print( "Cell ID: (PCell, Static)=(%d, %d)" % (sbcell1, sbcell2) )
  # Expect an Exception if the cell name is duplicated
  Layout.rename_cell( sbcell2, newName )
  Layout.delete_cell( sbcell1 )

# Misuse
MyDonuts = [ "A", "A", "A" ]

# Correct use
#MyDonuts = [ "A", "B", "C" ]

for donut in MyDonuts:
  AddMyDonut(donut)

print( " I'm going to SREF the generated donuts into the TOP cell." )
print( "   I want AddMyDonut() function to detect the duplication of cell names." )
print( "     Happy to have many grade-A donuts, though..." )

```
If I execute this PYA, I can get duplicated **A** cells.
[image]
[image]

----

## PYA: dupCells-ABC.py ##
```
#
# Why can I have duplicated cells?
#
import pya

MainWindow  = pya.MainWindow.instance()
CurrentView = MainWindow.current_view()
Layout      = CurrentView.active_cellview().layout()

def AddMyDonut(newName):
  global Layout

  paraL  = "layer"
  paraR1 = "actual_radius1"
  paraR2 = "actual_radius2"
  paraN  = "npoints"
  params         = dict()
  params[paraL]  = pya.LayerInfo(1, 0)
  params[paraR1] = 50
  params[paraR2] = 100
  params[paraN]  = 128
  sbcell1        = Layout.create_cell( "DONUT", "Basic", params ).cell_index()
  sbcell2        = Layout.convert_cell_to_static( sbcell1 )
  print( "Cell ID: (PCell, Static)=(%d, %d)" % (sbcell1, sbcell2) )
  # Expect an Exception if the cell name is duplicated
  Layout.rename_cell( sbcell2, newName )
  Layout.delete_cell( sbcell1 )

# Misuse
#MyDonuts = [ "A", "A", "A" ]

# Correct use
MyDonuts = [ "A", "B", "C" ]

for donut in MyDonuts:
  AddMyDonut(donut)

print( " I'm going to SREF the generated donuts into the TOP cell." )
print( "   I want AddMyDonut() function to detect the duplication of cell names." )
print( "     Happy to have many grade-A donuts, though..." )
```
If I run this PYA, I can get what I want.
[image]
[image]
However, if I run the script twice, I get duplicated cells again.
[image]

It will be helpful if there is a mechanism to detect duplicated cell names in PYA/RBA, too.
This is a kind of fool-safe mechanism rather than an enhancement.

Regards,
Kazzz-S

klayoutmatthias, 2020-07-04:
Hi @kazzz,

I'm a bit reluctant to just issue an exception on rename_cell for these reasons:

1. Within KLayout, duplicate cell names aren't an issue, because KLayout identifies cells through their index, not their names. 

2. Without throwing an exception, implementing a swap of cell names for example is as simple as this:

```
(cell_b.name, cell_a.name) = [cell_a.name, cell_b.name]
```

With an exception, you'd first need to assign a temporary name to b before assigning a the original name of b.

I think the problem exists at a different point in time: when writing to a stream. This will really cause an issue as streams usually identify cells by name. Duplicate names would mess up the files. I wonder if a useful solution was to throw an exception when trying to write a layout with duplicate names to a file.

Best regards,

Matthias

Kazzz-S, 2020-07-04:
Dear @klayoutmatthias,

Thanks a lot for looking into my idea. I understood all the points.
I noticed this issue because of a bug in my application PYA program, which is already fixed.
The application also writes the result to a stream file (OASIS).
So the "writer-guardian" is an appropriate idea.

Regards,
Kazzz-S

klayoutmatthias, 2020-07-07:
This feature now is released with 0.26.7 :)

closed 2020-07-07 by klayoutmatthias