KLayout/klayout #591: crash: Macro Development, Add Location

lukasc-ubc, 2020-06-20:
Hi @klayoutmatthias,

I would like to report that I am getting repeated crashes when trying to Add Location to the Macro Development, in the Python tab.  I am trying to add Project folders. I can add them, but as soon as I click on the folder, I get a Signal 11 crash.

This is on Mac OSX.  I've tried several versions I have installed (same as #589), including 0.26.5 and 0.26.6.  It wasn't a problem in earlier versions (I have many Project folders that work). It is only new folders that seem to cause the crash.

After about an hour, the backtrace is not complete... screenshot:
[image]

klayoutmatthias, 2020-06-27:
Hi @lukasc-ubc,

I'm beginning to understand which question to ask first on MacOS :) 

Which is the build variant you're using?

Thanks,

Matthias

klayoutmatthias, 2020-06-27:
Ah ... I can see this problem on Linux now too. 

The problem in somewhere inside the QTreeViews and their interaction with my tree models.

lukasc-ubc, 2020-06-27:
Wonderful.  It is always nice when you can reproduce the same problems, especially when they are intermittent.

Another one that is intermittent is copy and paste (Command-C, -V) in the Macro Dev window.

The first time you try copy, cut and paste, the action takes place in the GDS layout, rather than the code view.  Particularly I find it hard to copy and paste from the code window to the Console.

Also, the Select Macro Template windows keeps coming up, each time I start the Macro Development.  

It would be nice if there was a Configuration - Application option, “show Select Macro Template window at start-up”, so I can uncheck that.  Or if the window had a “don’t show automatically again”.

klayoutmatthias, 2020-06-29:
I could fix the crash (signal ordering problem in Qt) and was able to provide a way to dismiss the template selection dialog (is tied to the tip window now - if this is dismissed, the template selection won't be shown again).

The copy & paste issue however seems to be MacOS specific and a Qt related issue.

closed 2020-07-03 by klayoutmatthias

klayoutmatthias, 2020-07-05:
A can confirm the copy / paste issue, but it seems to be related to the way Qt treats the keys. If I assume VirtualBox maps my "Windows" key to Mac's "Command" key, then Copy/Paste of *Text* seems to work with Command+C, Command+V from the edit widgets while Copy/Paste of geometry, cells and others seems to work with Control+C, Control+V.

commit 6365ddfe (2020-07-03): Fixed #591 (two topics for macro editor) (#595)

* Fixed #591 (crash on Macro/Add Location)

The main reason was that the QSortFilterProxyModel is very sensitive
to the order of signals it receives from the proxy model.

In this case, dataChanged() must not be send between layoutAboutToBeChanged()
and layoutChanged(). This happened implicitly during load() of a macro while
scanning the freshly added folder.

* Fixed another part of #591: ability to disable template selection pop-up in macro editor. Tied to the tip window now - if this is dismissed, no template selection dialog will be shown.
commit 0507d72d (2020-07-03): Fixed #591 (two topics for macro editor) (#595)

* Fixed #591 (crash on Macro/Add Location)

The main reason was that the QSortFilterProxyModel is very sensitive
to the order of signals it receives from the proxy model.

In this case, dataChanged() must not be send between layoutAboutToBeChanged()
and layoutChanged(). This happened implicitly during load() of a macro while
scanning the freshly added folder.

* Fixed another part of #591: ability to disable template selection pop-up in macro editor. Tied to the tip window now - if this is dismissed, no template selection dialog will be shown.