KLayout/klayout #599: Linker failure building unit tests

Partmedia, 2020-06-30:
KLayout e744eb32 fails to build on FreeBSD:

clang++ -pthread -Wl,-rpath,/usr/local/lib -Wl,-rpath,/usr/local/lib/qt5 -Wl,-rpath-link,/usr/local/lib/qt5 -o ../../build-debug/ut_runner unit_test_main.o utTestConsole.o   -L/usr/local/lib -lz /usr/local/lib/libpython3.7m.so /usr/local/lib/libruby26.so.26 -L/home/kevinz/workspace/eda/klayout/build-debug/unit_tests/.. -lklayout_gsi -lklayout_db -lklayout_rdb -lklayout_lib -lklayout_lym -lklayout_laybasic -lklayout_lay -lklayout_ant -lklayout_img -lklayout_edt -lklayout_qtbasic -lklayout_QtGui -lklayout_QtCore -lklayout_rba -lklayout_pya -lklayout_drc -lklayout_lvs -ldl -lklayout_gsi_test -lklayout_tl -lklayout_QtXml -lklayout_QtWidgets /usr/local/lib/qt5/libQt5Designer.so /usr/local/lib/qt5/libQt5PrintSupport.so /usr/local/lib/qt5/libQt5MultimediaWidgets.so /usr/local/lib/qt5/libQt5Svg.so /usr/local/lib/qt5/libQt5Widgets.so /usr/local/lib/qt5/libQt5Multimedia.so /usr/local/lib/qt5/libQt5Gui.so /usr/local/lib/qt5/libQt5XmlPatterns.so /usr/local/lib/qt5/libQt5Network.so /usr/local/lib/qt5/libQt5Xml.so /usr/local/lib/qt5/libQt5Sql.so /usr/local/lib/qt5/libQt5Core.so -lGL   
ld: error: undefined symbol: tl::get_inst_path()
>>> referenced by unit_test_main.cc:454 (../../src/unit_tests/unit_test_main.cc:454)
>>>               unit_test_main.o:(main_cont(int&, char**))

This happens even after I add "LIBS += -lklayout_tl" to `src/unit_tests/unit_tests.pro`.

Any idea what's going on?

klayoutmatthias, 2020-06-30:
Sorry, but I don't support FreeBSD for now and I don't plan to do so.

Adding -klayout_tl won't help as it is already there. You'll need to look if klayout_tl.so defines this symbol (e.g. with nm) and if not, figure out why this isn't the case. Hint: it's supposed to be defined in tlFileUtils.cc.

Matthias

Partmedia, 2020-06-30:
Hi Matthias,

Thanks for the response; and I understand completely that you can't support FreeBSD per-se.

I did find that `tl::get_inst_path()` is defined in *tlFileUtils.cc.*, but as far as I can tell, it is included in *layout_tl*, because the qmake file that builds the library includes *tlFileUtils.cc*. Should that be true?

I'll keep poking.

klayoutmatthias, 2020-07-03:
Yes, it's defined in tlFileUtils.cc, but if the linker complains, it means that it's not defined in klayout_tl.so and this is probably because it's not defined in "tlFileUtils.o". The reasons are not necessarily obvious. If there are many more errors like this, the compile visibility macros (TL_PUBLIC etc.) may not be defined properly. It should expand to "__attribute__ ((visibility ("default")))" if you compile tlFileUtils.cc. You can check by reading the C preprocessor output (clang -E ...). Otherwise, you need to check the symbols exported by tlFileUtils.o and how the linker exports them from "klayout_tl.so".

If this is just the first error, be prepared to come across many more of this kind.

Matthias

Partmedia, 2020-07-03:
Thanks for the suggestions. Indeed, this is the vein where I'm looking. Among other things, I've tried:

- Reordering the TL_PUBLIC declaration
- Entirely removing -fvisibility=hidden
- Re-compiling with GCC 9

None of these seem to shake out the issues. My linker is LLD 8.0.1, so perhaps the next thing to do is to try the GNU binutils ld. I'll keep you posted; and thanks again.

Partmedia, 2020-08-17:
It looks like the build system was preferring the dynamic libraries from the older, system-wide installation of KLayout. Uninstalling the system-wide KLayout allowed compilation to proceed. Is there any way to ask the build system to prefer build libraries over system KLayout libraries?