KLayout/klayout #602: Cell creation in python Database API

MichalM1234, 2020-07-03:
Hello,
I have few  things I would like to ask:
1.
I have encountered a problem while creating a separate cell that could be used in my GDS structure.
I am trying to run this code:

```
import pya

layout = pya.Layout()
layout.read("TOP.gds")
cell = pya.Cell()
cell_name = cell.name="new_cell"
```

and it reports problem:

> ERROR: Object cannot be created here in Cell.name
command: 

> klayout  -z -r script.py
Is there any solution how I can create cell as standalone and assign in to place in GDS structure I want?
2. 
Same happens if I want to create Instance object and then assign it to specfic cell using .insert() method. It fails as well.
How can i assign Instances to cells?
3.
While creating my structure I have GDS file where I would like to add Cell structure with instances ( see Picture)
[image]
I am trying to add new cell with instances into top level subcell.
Could you give me a hint how could I do that since I have tried many times and always it end up on beeing an error of some kind. I am sorry if I do not see an obvious solution but I started just recently with klayout Python API and it works little bit different then i am used for Python.

Thank you for your time!
Best regards,
Michal

klayoutmatthias, 2020-07-03:
Hi Michal,

please take support questions to the Forum: https://www.klayout.de/forum/discussions. This is a bug tracker. I'll close this as it's not a bug or enhancement request.

Apparently you have not read the documentation. This explains the basic procedure: https://www.klayout.de/doc-qt5/programming/database_api.html#h2-18

Bottom line is you don't create a cell by instantiating a Cell object but by using "layout.create_cell(...)".

closed 2020-07-03 by klayoutmatthias

Kazzz-S, 2020-07-04:
Hi @MichalM1234,

Coincidentally, I was creating a tutorial similar to your problem for my colleagues.
[T-602.zip](https://github.com/KLayout/klayout/files/4872066/T-602.zip) may help you get started with PYA.
As Matthias commented, reading the document set is the first step.

Regards,
Kazzz-S

MichalM1234, 2020-07-13:
Hi,
I am sorry for any inconvenience. Thanks for your reply @klayoutmatthias. @Kazzz-S , your tutorial really helped. Problem was I was trying to create standalone cell without any layout(class) connections. 

Regards,
Michal

Kazzz-S, 2020-07-13:
Hi,

Glad to hear that :-)

Regards,
Kazzz-S