KLayout/klayout #607: Follow FHS for installing files

smunaut, 2020-07-05:
See https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard

When I set a prefix, I expect things to get installed in: `$prefix/bin` for binaries `$prefix/lib` for libs, etc ...
Currently it just dumps everything in a single directory :/

klayoutmatthias, 2020-07-05:
What's the problem?

smunaut, 2020-07-05:
Well as I stated above, all files are just dumped into the $prefix without respecting the FHS standard which is best practice for software installs. To be accepted / packaged in distributions official repos, files need to be installed where they belong.

Looking at the repo someone obviously made some parallel build system to respect the OSX standard and distribute files in the right place for OSX, but the same doesn't exist for class unix / linux distributions installs.

klayoutmatthias, 2020-07-05:
So then why isn't this someone contribution his build script to this project?

Support for another distribution is much more than setting up the build system. There is only one maintainer here and that's me. I'm not interested in supporting another platform and receiving error reports for systems I don't have in stock.

I also think you're underestimating the fun of multi-system deployment. For Windows for example, dumping everything is the only solution as Windows will look up DLLs relative to the prime binary.

smunaut, 2020-07-05:
Sorry I asked.

closed 2020-07-05 by smunaut

klayoutmatthias, 2020-07-05:
No need to be sorry, but that topic is not on my priority list.

I'd welcome contributions if there were some. MacOS support is a positive example - this platform is supported by kazzz. Everything else by me.