KLayout/klayout #611: Mac DMGs for 0.26.7

Kazzz-S, 2020-07-11:
Dear @klayoutmatthias,

Thank you very much for making the maintenance release with extraordinary efforts to support Mac.
As usual, different DMG packages of **KLayout 0.26.7 for Mac** are available as below ([0.26.7 Dropbox Link](https://www.dropbox.com/sh/d3e37x3ub5oent2/AAATh2AdknFACtvcuf6Jjdopa?dl=0)).
**Please note that _pymod_ is not included in those packages.**

## DMG file naming convention ##
**0.26.7 for Catalina** is taken as an example here.
[image]

where
(a) Type of package:  ST-: standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt5 type:  qt5MP: MacPorts; qt5Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:   sys: OS-bundled; mp26: Ruby2.6 from MacPorts; hb27: Ruby2.7 from Homebrew; ana3: Ruby2.5 from Anaconda3
(f) **P**ython type: sys: OS-bundled; mp37: Python 3.7 from MacPorts; hb37: Python 3.7 from Homebrew; ana3: Python3.7 from Anaconda3

## For the latest macOS Catalina (10.15) ##
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
1|Catalina (10.15)|ST-klayout-0.26.7-macOS-Catalina-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1)|
**2**|**Catalina (10.15) with MacPorts dev. env.**|LW-klayout-0.26.7-macOS-Catalina-1-qt5MP-Rmp26Pmp37.dmg|MacPorts|MacPorts|MacPorts|(2)|
3|Catalina (10.15) with Homebrew dev. env.|LW-klayout-0.26.7-macOS-Catalina-1-qt5Brew-Rhb27Phb37.dmg|Homebrew|Homebrew|Homebrew|(3)|
4|Catalina (10.15) with Anaconda3 dev. env.|LW-klayout-0.26.7-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(4)|
5|Catalina (10.15)|HW-klayout-0.26.7-macOS-Catalina-1-qt5Brew-RsysPhb37.dmg|Homebrew|OS-bundled|Homebrew|(5)|

(1) shares OS-bundled Ruby2.6 and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).
(2) **new! experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(3) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(4) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(5) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.7 from Homebrew are embedded, hence, heavy; [The known issue](https://github.com/KLayout/klayout/issues/460#issuecomment-587968841)

## For previous OS ##
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
6|Mojave (10.14)|ST-klayout-0.26.7-macOS-Mojave-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1)|
7|HighSierra ( 10.13)|ST-klayout-0.26.7-macOS-HighSierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1)|
8|Sierra (10.12)|ST-klayout-0.26.7-macOS-Sierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1)|
9|ElCapitan (10.11)|ST-klayout-0.26.7-MacOSX-ElCapitan-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled| (1) |
x|Yosemite (10.10)|has been dropped due to the compiler issue reported in #360.|

Please close this ticket after publishing the packages.
With warm regards,
Kazzz-S

klayoutmatthias, 2020-07-29:
Hi kazzz, 

as this is done, can I close this issue?

Thanks,

Matthias

Kazzz-S, 2020-07-29:
Hi Matthias,

Yes. Thanks for merging the pull request #612, too.

Regards,
Kazzz-S

closed 2020-08-02 by klayoutmatthias