KLayout/klayout #614: How to replace circle polygon with square polygon?

zydanielw, 2020-07-18:
Dear klayout admin,
I read in a flatten GDS. There are so many circuit polygons with 16 segments, but I would like to replace all of them with square polygon. How to do that in klayout?

Best Regards,
Daniel

Kazzz-S, 2020-07-18:
Hi Daniel @zydanielw,

Since this site is for bug tracking, I think the KLayout Forum https://www.klayout.de/forum/discussions is the right place to post this kind of question.
Nevertheless, here is my idea [T614.zip](https://github.com/KLayout/klayout/files/4941596/T614.zip) in PYA.

[image]

Just for your information.
Best regards,
Kazzz-S

zydanielw, 2020-07-18:
Hi Kazzz-S

Sorry, I am totally new with Github and klayout. For some reason, I registered a new account in  KLayout Forum yesterday, but that system doesn't allow me to post anything there. Anyway, thanks for the solution. I really appreciate.

Best Regards,
Daniel

Kazzz-S, 2020-07-18:
Hi Daniel,

No problem. I was also a novice in the beginning (maybe even now. KLayout is technically so deep!).

>, but that system doesn't allow me to post anything there. 

I think your problem with the Forum is due to this. Please wait.
[image]

Best regards,
Kazzz-S

klayoutmatthias, 2020-07-28:
Hi,

I approved your membership yesterday. As I am responsible for content posted on my side, I'm a bit picky and select the users myself.

Please go ahead and post your questions there.

@kazzz Many thanks for your elaborate answer! Your script is awesome - in style, documentation and the choice of the solution! Thanks a lot!

Matthias

closed 2023-08-09 by klayoutmatthias