KLayout/klayout #615: KLayout Salt, KLAYOUT_SALT_MINE env variable

lukasc-ubc, 2020-07-22:
Dear Matthias,

I've had a couple of students who get an authentication error when accessing a private GitHub repo via KLayout.  They can successfully access the repo via the web, but not via KLayout. Previously this was solved by having them change to a simpler password.  Do you know of any password restrictions in place in your implementation?  Length, special characters (specifically French), etc?

[image]

lukasc-ubc, 2020-07-22:
[image]

klayoutmatthias, 2020-07-28:
Hi Lukas,

thanks for reporting this - I'm not aware of password restrictions. Maybe it's a matter of encoding. Did their password include some characters outside the ISO range? 

I'm not using a private GitHub repo myself, but HTTPS access is pretty common for me. However, I still use passwords from the ASCII character range.

Matthias

lukasc-ubc, 2020-07-28:
I don’t know, and didn’t ask for their passwords.. ;)
Next time it comes up I’ll ask them to test it with french characters, or whatever else they may be using.

klayoutmatthias, 2020-07-28:
I browsed the internet a little. It looks like the server can suggest a certain encoding (https://en.wikipedia.org/wiki/Basic_access_authentication). Maybe GitHub either does not provide this encoding or Qt doesn't accept it.

I just learned that GitHub provides free private repos now, so I'll try that.

Matthias

klayoutmatthias, 2020-07-29:
Good news is, I can reproduce it with a private repo on GitHub. I used the "ä" I find on my keyboard - seems to be as good as the french special characters :) 

Good news too is I can debug the issue: apparently, GitHub wants UTF-8 encoding (found out with curl + traffic trace). 

Bad thing is: Qt sends the password Latin1 encoded (ISO 8859-1):

```
//  qauthenticate.cpp: 471
    case QAuthenticatorPrivate::Basic:
        methodString = "Basic ";
        response = user.toLatin1() + ':' + password.toLatin1();
        response = response.toBase64();
        phase = Done;
        break;
```

So the question is how to solve this. I don't know if Qt will change it (latest version is still on ISO8859-1). Building KLayout against curl is possible and probably works, but that's a new setup for MacOS builds. Plus the curl integration is still somewhat experimental.

Another freaky option is to cast the UTF8-Password to ISO8859-1 and recode as UTF8 ... I need to try this.

Matthias

klayoutmatthias, 2020-07-29:
My hack seems to work ... nevertheless I don't know which risk I am taking. That needs some testing.

Matthias

klayoutmatthias, 2020-09-19:
Latest release 0.26.8 fixes this issue.

closed 2020-09-19 by klayoutmatthias