KLayout/klayout #624: KLayout High Sierra QT5 wPython 3 required

vanderpol2, 2020-08-05:
[image]
Can you please create an installer w Python 3 version for High Sierra. Required for SIEPIC-TOOLS . maybe Brew?
Thanks Lou.

lukasc-ubc, 2020-08-05:
The vast majority of SiEPIC tools runs in the legacy python 2. So you could try it, and ignore the warning message for now. If you find some functionality that doesn’t work, let us know as we are developing everything for python 3 these days. 

Lukas Chrostowski
Professor, Electrical and Computer Engineering
University of British Columbia
Quantum Matter Institute
Member, College of the Royal Society of Canada
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca
https://qmi.ubc.ca/about/directory/lukas-chrostowski

vanderpol2, 2020-08-05:
Hi Lukas, 
Vey nice to hear from you.  I am specifically tryin to get your software operating. I am stuck at INTERCONNECT. 
I says failed to load NumPy and also it fails to find  👍this .cml file even when I navigate to it, it shows geyed out in hte menu box.
[image]
[image]
I was pretty excitedio get this far along!:
[image]
[image]
[image]

lukasc-ubc, 2020-08-05:
Hi @vanderpol2 

To learn how to use KLayout for silicon photonics, there are two online edX courses you can consider:

- https://www.edx.org/course/photonic-integrated-circuits-1  (Aug 11)
- https://www.edx.org/course/silicon-photonics-design-fabrication-and-data-ana (September)

The screenshots above show components from a layout using the technology "EBeam", while the library being installed is from technology "GSiP".

GSiP is an example library that presently isn't complete.  I do see it is missing models, so it isn't surprising you are getting the error about missing models.   Here is a screenshot of the models that are presently included.  Note that there is no grating couplers, but there is an edge coupler.

Lastly, do you have Lumerical INTERCONNECT installed, including the Netlist import license feature?  Without it, you won't be able to do circuit simulations.

regards,
Lukas

p.s. since this is not a KLayout issue, it would be preferable to report it here: https://github.com/SiEPIC/SiEPIC-Tools/issues

[image]

vanderpol2, 2020-08-06:
Ok thank you so much, moving to SiEPC-Tools/ssues/ 
[image]
[image]
[image]
[image]

vanderpol2, 2020-08-06:
[image]

Kazzz-S, 2020-08-12:
Hi @vanderpol2,

> Can you please create an installer w Python 3 version for High Sierra. Required for SIEPIC-TOOLS . maybe Brew?
> Thanks Lou.

I've built the second set of installers for HighSierra for testing.
Please visit https://www.dropbox.com/sh/nurelwgvu6kat2t/AACTHYczC4v3BNUTpsIGOgima?dl=0 

Maybe, ```HW-klayout-0.26.7-macOS-HighSierra-2-qt5Brew-RsysPhb37.dmg```  is the one you need.

Regards,
Kazzz-S

vanderpol2, 2020-08-12:
Installer worked!, asked for Numpy install permission at first CTA, no
python 3 requirement error message,
Awesome so far. Thanks!!
Lou
[image: Screen Shot 2020-08-11 at 9.33.20 PM.png]

klayoutmatthias, 2020-10-28:
Very good :)

Can I close this issue?

Thanks,

Matthias

vanderpol2, 2020-10-28:
Yes Thank You.
Best
Lou

thomaslima, 2020-11-27:
I am closing this since this issue was solved.

closed 2020-11-27 by thomaslima