KLayout/klayout #625: Wrong layer used when showing off a 'ROUTED' net from DEF that starts with a VIA

smunaut, 2020-08-08:
In the DEF you find :

```
NETS 1 ;
    - testNet ( Mux1 A0 ) ( Mux1 S0 ) + USE SIGNAL
      + ROUTED li1 ( 209862 226950 ) L1M1_PR ( 212255 * ) L1M1_PR;
```

And this gets displayed as a horizontal wire in the `li1` layer instead of the `met1` layer.

From DEF specs :

> viaName
>
> Specifies a via to place at the last point. If
> you specify a via, layerName for the next
> routing coordinates (if any) is implicitly
> changed to the other routing layer for the via.
> For example, if the current layer is metal1, a
> via12 changes the layer to metal2 for the
> next routing coordinates.

So here the route starts on `li1` (but doesn't actually have anything there) and directly goes through a via `L1M1_PR` (via from `li1` to `met1`) and so the layer should switch to `met1`, go some distance on the X axis, then come down to `li1` again via another via.

magic shows the trace on the `met1` layer, but KLayout shows it on the `li1` layer.

Files ( example DEF + tech LEF and lib LEF ) to reproduce :

[kl.tar.gz](https://github.com/KLayout/klayout/files/5045951/kl.tar.gz)

klayoutmatthias, 2020-08-19:
@smunaut Thanks for the testcase and this bug report.

LEF/DEF is under heavy development currently. One reason is support for MASK separation and a variety of other topics. It turned out that LEF/DEF reading - which started rather naive - is actually far more difficult than expected. With a lot of other issues your bug is already fixed there: if you want to build KLayout yourself, you can try the "lefdef_enhancments" branch (sorry for the missing "e").

Unfortunately the changes are too big to be backported to 0.26, so the fix will only come in 0.27.

I have attached the DEF I have translated with this version. It shows the wire in the correct layer now. 

Please note that KLayout will automatically read all LEF files next to your DEF, but it will look for "*.lef", so it does not recognize "*.tlef" as a LEF file. If you rename it to "..._tech.lef", KLayout will read it together with your DEF without need to specify it explicitly.

Best regards,

Matthias

[translated.zip](https://github.com/KLayout/klayout/files/5099629/translated.zip)

stefanottili, 2024-02-12:
this can be closed, right ?