KLayout/klayout #626: Klayout crashing

huantee, 2020-08-14:
Hi,

   Recently (today) I got this problem.  I uninstalled version 0.26.3 and installed the current version available 0.26.7 and I still have the same problem.  I uninstalled it again and re-installed and I still see the same problem.

Below is a copy of the detailed list.  Any suggestion?

Thanks

Exception code: 0xc0000005
Program Version: KLayout 0.26.7 (2020-07-07 r28bf5259) AMD64

Backtrace:
0x6a61f297 - (klayout_lay.dll) +914071
0x6a620425 - (klayout_lay.dll) +918565
0x3bc4cad - (klayout_laybasic.dll) +1133741
0x6ee02f82 - (klayout_tl.dll) +274306
0x6ee0e489 - (klayout_tl.dll) +320649
0x2539199 - (Qt5Core.dll) +2003353
0x2543efa - (Qt5Core.dll) +2047738
0x2539e5b - (Qt5Core.dll) +2006619
0x66207ab1 - (Qt5Widgets.dll) +31409
0x6620ea04 - (Qt5Widgets.dll) +59908
0x6a54c8db - (klayout_lay.dll) +51419
0x2511e88 - (Qt5Core.dll) +1842824
0x25659ac - (Qt5Core.dll) +2185644
0x66207ab1 - (Qt5Widgets.dll) +31409
0x6620ea04 - (Qt5Widgets.dll) +59908
0x6a54c8db - (klayout_lay.dll) +51419
0x25117dd - (Qt5Core.dll) +1841117
0x2518461 - (Qt5Core.dll) +1868897
0x6dd3811e - (qwindows.dll) +491806
0x25655bf - (Qt5Core.dll) +2184639
0x7ffd6ce05c7d - (USER32.dll) +89213
0x7ffd6ce05672 - (USER32.dll) +87666
0x256492b - (Qt5Core.dll) +2181419
0x6dd38105 - (qwindows.dll) +491781
0x25104f5 - (Qt5Core.dll) +1836277
0x663cc38e - (Qt5Widgets.dll) +1885070
0x663f7abd - (Qt5Widgets.dll) +2063037
0x663f7c01 - (Qt5Widgets.dll) +2063361
0x3abb7b0 - (klayout_laybasic.dll) +47024
0x3d76439 - (klayout_laybasic.dll) +2909241
0x6f5d4bed - (klayout_pya.dll) +84973
0x6dbe159 - (libpython3.7m.dll) +123225
0x6dbe22b - (libpython3.7m.dll) +123435
0x6f0df04 - (libpython3.7m.dll) +1498884
0x6f06443 - (libpython3.7m.dll) +1467459
0x6f0dcc1 - (libpython3.7m.dll) +1498305
0x6e6e416 - (libpython3.7m.dll) +844822
0x6e6e6c1 - (libpython3.7m.dll) +845505
0x6f5c1f81 - (klayout_pya.dll) +8065
0xd87cdc - (klayout_lym.dll) +31964
0xd8d3ad - (klayout_lym.dll) +54189
0xd8d360 - (klayout_lym.dll) +54112
0x6a54db07 - (klayout_lay.dll) +56071
0x401df6 - (klayout_app.exe) +7670
0xd11602 - (klayout_rba.dll) +5634
0x29ddce6 - (x64-msvcrt-ruby260.dll) +1694950
0x29e6603 - (x64-msvcrt-ruby260.dll) +1730051
0x29ecb2b - (x64-msvcrt-ruby260.dll) +1755947
0x29e4fdf - (x64-msvcrt-ruby260.dll) +1724383
0x28b1b8c - (x64-msvcrt-ruby260.dll) +465804
0x28b5949 - (x64-msvcrt-ruby260.dll) +481609
0xd16d62 - (klayout_rba.dll) +28002
0x40197c - (klayout_app.exe) +6524
0x4013a5 - (klayout_app.exe) +5029
0x4014db - (klayout_app.exe) +5339
0x7ffd6df27bd4 - (KERNEL32.DLL) +97236
0x7ffd6e40ce51 - (ntdll.dll) +446033

huantee, 2020-08-14:
In Addition an get this message
[image]

huantee, 2020-08-14:
I renamed file 'klayoutrc' and re-installed klayout an now is working.  File code below:
 
<?xml version="1.0" encoding="utf-8"?>
<config>
 <absolute-units>false</absolute-units>
 <abstract-mode-enabled>false</abstract-mode-enabled>
 <abstract-mode-width>10</abstract-mode-width>
 <apply-text-trans>true</apply-text-trans>
 <background-color>#000000</background-color>
 <bitmap-caching>true</bitmap-caching>
 <bitmap-oversampling>1</bitmap-oversampling>
 <cell-list-sorting>by-name</cell-list-sorting>
 <child-context-color>auto</child-context-color>
 <child-context-dimming>50</child-context-dimming>
 <child-context-enabled>false</child-context-enabled>
 <child-context-hollow>false</child-context-hollow>
 <cib-context-cell/>
 <cib-context-mode>any-top</cib-context-mode>
 <cib-max-inst-count>1000</cib-max-inst-count>
 <cib-window-dim>1</cib-window-dim>
 <cib-window-mode>fit-marker</cib-window-mode>
 <cib-window-state/>
 <circle-points>32</circle-points>
 <clear-ruler-new-cell>false</clear-ruler-new-cell>
 <color-palette>255,157,157[0] 255,128,168[1] 192,128,255[2] 149,128,255[3] 128,134,255[4] 128,168,255[5] 255,0,0[6] 255,0,128[7] 255,0,255[8] 128,0,255[9] 0,0,255[10] 0,128,255[11] 128,0,0[12] 128,0,87[13] 128,0,128[14] 80,0,128[15] 0,0,128[16] 0,64,128[17] 128,255,251[18] 128,255,141[19] 175,255,128[20] 243,255,128[21] 255,194,128[22] 255,160,128[23] 0,255,255[24] 1,255,107[25] 145,255,0[26] 221,255,0[27] 255,174,0[28] 255,128,0[29] 0,128,128[30] 0,128,80[31] 0,128,0[32] 80,128,0[33] 128,128,0[34] 128,80,0[35] 255,255,255 192,192,192 128,128,128 96,96,96 64,64,64 0,0,0</color-palette>
 <combine-mode>add</combine-mode>
 <context-color>auto</context-color>
 <context-dimming>50</context-dimming>
 <context-hollow>false</context-hollow>
 <current-lib-view/>
 <current-ruler-template>0</current-ruler-template>
 <custom-macro-paths/>
 <dbu-units>false</dbu-units>
 <def-import-spec/>
 <default-add-other-layers>true</default-add-other-layers>
 <default-font-size>0</default-font-size>
 <default-grids>0.01,0.005,0.001</default-grids>
 <default-layer-properties>C:/Users/Edgar/ownCloud/PIC Domain/Ranovus PIC working folder/GF PDK/Layout/GF45spclo_Lassen.lyp</default-layer-properties>
 <default-text-size>0.1</default-text-size>
 <diff-detailed>false</diff-detailed>
 <diff-exact>false</diff-exact>
 <diff-expand-cell-arrays>false</diff-expand-cell-arrays>
 <diff-run-xor>false</diff-run-xor>
 <diff-summarize>false</diff-summarize>
 <digits-dbu>2</digits-dbu>
 <digits-micron>5</digits-micron>
 <draw-array-border-instances>false</draw-array-border-instances>
 <drawing-workers>1</drawing-workers>
 <drop-small-cells>false</drop-small-cells>
 <drop-small-cells-condition>0</drop-small-cells-condition>
 <drop-small-cells-value>10</drop-small-cells-value>
 <edit-connect-angle-mode>ortho</edit-connect-angle-mode>
 <edit-grid>global</edit-grid>
 <edit-hier-copy-mode>-1</edit-hier-copy-mode>
 <edit-inst-angle>0</edit-inst-angle>
 <edit-inst-array>false</edit-inst-array>
 <edit-inst-cell-name>TEXT</edit-inst-cell-name>
 <edit-inst-column_x>0</edit-inst-column_x>
 <edit-inst-column_y>0</edit-inst-column_y>
 <edit-inst-columns>1</edit-inst-columns>
 <edit-inst-lib-name>Basic</edit-inst-lib-name>
 <edit-inst-mirror>false</edit-inst-mirror>
 <edit-inst-pcell-parameters>!bias:##0;cspacing:##0;eff_ch:##0.8;eff_cw:##0.6;eff_dr:##0.05;eff_lw:##0.1;font:#0;font_name:'std_font';inverse:false;layer:[layer:];lspacing:##0;mag:##1;text:'';</edit-inst-pcell-parameters>
 <edit-inst-place-origin>false</edit-inst-place-origin>
 <edit-inst-row_x>0</edit-inst-row_x>
 <edit-inst-row_y>0</edit-inst-row_y>
 <edit-inst-rows>1</edit-inst-rows>
 <edit-inst-scale>1</edit-inst-scale>
 <edit-max-shapes-of-instances>1000</edit-max-shapes-of-instances>
 <edit-mode>true</edit-mode>
[... 405 lines cut ...]
</technology-data>
 <test-shapes-in-view>false</test-shapes-in-view>
 <text-color>auto</text-color>
 <text-font>3</text-font>
 <text-lazy-rendering>true</text-lazy-rendering>
 <text-visible>true</text-visible>
 <tip-window-hidden>editor-mode=4,editor-mode=0,hide-empty-layers=4,hide-empty-layers=0</tip-window-hidden>
 <window-geometry>AdnQywADAAAAAAP////7wAAAC4AAAAAAAAAE9v///AAAAAi1AAAAIAAAAAICAAAAB4AAAAQA///73wAAC3//////</window-geometry>
 <window-state>AAAA/wAAAAD9AAAAAgAAAAAAAADcAAADuPwCAAAAA/sAAAAqAG4AYQB2AGkAZwBhAHQAbwByAF8AZABvAGMAawBfAHcAaQBkAGcAZQB0AAAAAAD/////AAAAjQD////7AAAAHABoAHAAXwBkAG8AYwBrAF8AdwBpAGQAZwBlAHQBAAAAVQAAAbUAAAAWAP////sAAAAgAGwAaQBiAHMAXwBkAG8AYwBrAF8AdwBpAGQAZwBlAHQBAAACDgAAAf8AAAAWAP///wAAAAEAAADsAAADuPwCAAAAA/sAAAAqAGIAbwBvAGsAbQBhAHIAawBzAF8AZABvAGMAawBfAHcAaQBkAGcAZQB0AAAAAAD/////AAAAFgD////7AAAAHABsAHAAXwBkAG8AYwBrAF8AdwBpAGQAZwBlAHQBAAAAVQAAA0QAAAAWAP////sAAAAcAGwAdABfAGQAbwBjAGsAXwB3AGkAZABnAGUAdAEAAAOdAAAAcAAAAHAAAABwAAAFsAAAA7gAAAAEAAAABAAAAAgAAAAI/AAAAAEAAAACAAAAAQAAAA4AdABvAG8AbABiAGEAcgEAAAAA/////wAAAAAAAAAA</window-state>
 <xor-anotb>false</xor-anotb>
 <xor-axorb>true</xor-axorb>
 <xor-bnota>false</xor-bnota>
 <xor-input-mode>all</xor-input-mode>
 <xor-layer-offset/>
 <xor-num-workers>1</xor-num-workers>
 <xor-output-mode>rdb</xor-output-mode>
 <xor-region-mode>all</xor-region-mode>
 <xor-summarize>false</xor-summarize>
 <xor-tiling/>
 <xor-tolerances/>
</config>
marize>
 <xor-tiling/>
 <xor-tolerances/>
</config>

huantee, 2020-08-14:
I renamed the file 'klayoutrc' and reinstalled Klayout and now the application is working.  Line 495 is below (is the one missing "<" before marzise)

<xor-tiling/>
 <xor-tolerances/>
</config>
marize>
 <xor-tiling/>
 <xor-tolerances/>
</config>

klayoutmatthias, 2020-08-16:
Hard to tell reason without the full configuration file. Usually, upgrading KLayout should not provide such issues as configuration files should be backward compatible to some degree (-1 major version, minor versions always).

As the problem is solved, can I close this issue?

Matthias

huantee, 2020-08-17:
Hi Matthias,

    Thank you for looking into this problem.  My issue was that I didn't
change anything in Klayout when I started to have this problem.  I was
working fine, however, I noted that the size of the window for Klayout was
slightly off (in full screen mode)  such that I couldn't see the
coordinates of the cursor located at the bottom right corner.  I checked my
screen setting and they were correctly configured, then, I decided to
restart  the computer and that is when I started to have this problem.

   I updated the software from v0.23.3 to v0.23.7, hoping to solve this
problem.  However, it wasn't solved.

  I opened the file "klayoutrc" and checked line 495, column 1. It seems to
miss a '<' symbol at the beginning of the row.  I was going to send you the
full file, however, I realized there is some information in it that I
cannot share with you (NDA).

 By renaming the file (same as deleting it) solved the problem.

  Yes, you can close this issue.

Thank you,
Edgar

klayoutmatthias, 2020-08-20:
Hi Edgar, 

there is a tiny issue which may cause corrupted configuration files: if you close two KLayout instances at once (which may happen if your computer shuts down with two KLayout instances open), both will try to write the config file. This may lead to corrupt files. 

However, this should not make the application crash. I'd like to debug the problem, but I understand that it's not possible to send the file. I'll keep an eye on this nevertheless. Maybe it's also worth locking the configuration file during write to prevent this problem in general.

Thanks and kind regards,

Matthias

closed 2020-08-20 by klayoutmatthias

huantee, 2020-08-21:
Hi Matthias,

   Thank you for continuing looking into this issue.  Now that you mention
it, I was keeping a couple (or more) of KLayout windows open (maybe with
unsaved work) when I closed my laptop (it went to sleep).

Thanks,
Edgar