KLayout/klayout #628: Session not restored on 0.26.7

olcc, 2020-08-17:
Hello,
I have a LYS file that can be resorted properly on KLayout 0.25.6 (Linux), 0.25.9 (Windows), 0.26.2 (Linux), but not on 0.26.7 (Windows).
That session file contains relative paths like `<file-path>Découpe/dicing.gds</file-path>` and `<file-path>panel.oas</file-path>`.
Olivier

klayoutmatthias, 2020-08-17:
Hi Olivier,

There was a change in the way paths inside sessions files are read: https://github.com/KLayout/klayout/issues/353

But this was version 0.26.1. Today all relative file paths should be resolved relative to the location of the session file. 

What is your observation?

Matthias

olcc, 2020-08-17:
I updated my report: in the initial report, I was saying that the LYS file in question could be read with 0.26.2 (Linux), but in fact it couldn't.
Do you mean that LYS files cannot be shared between Klayout versions ≤ 0.25.9 and version ≥ 0.26.1 ?

olcc, 2020-08-18:
In my case, the file paths mentioned above are relative to the location of the LYS file.

klayoutmatthias, 2020-09-13:
LYS files with relative paths cannot be shared because relative paths are treated differently. But LYS files written by KLayout should only have absolute paths. 

Is this answering your question?

Matthias

closed 2023-08-09 by klayoutmatthias