KLayout/klayout #634: building pya python in windows10

HansMalo, 2020-09-03:
Hello, 
 tried to build the klayout python package via pip install klayout, but an error was raised. I compared the downloaded file to other issues and https://github.com/KLayout/klayout/issues/472 uses the same downloaded version. install log is attached. Am I missing build packages, or is there something wrong with the build package?

Best Hans

[pip install log.txt](https://github.com/KLayout/klayout/files/5168668/pip.install.log.txt)

klayoutmatthias, 2020-09-08:
@HansMalo,

KLayout's Python module does not build out of the box - it needs some prerequisites on Windows. In you case, the curl library is not found (and more will follow).

Some details can be found here: https://www.klayout.de/build.html ("Building KLayout on Windows (Version 0.26)"). 

The solution is to install the "bits4klayout" package (https://github.com/klayoutmatthias/klayout_bits4msvc2017/releases) and set "KLAYOUT_BITS" to the path where you installed these (e.g. "c:\klayout-bits\msvc2017\x64").

Matthias

HansMalo, 2020-09-23:
@klayoutmatthias problem solved. 
there seems to be a package file for python 3.8 now that installs well with pip which was not existing before.
My intention was not to build klayout's python module but to install the precompiled python module, which was missing until now.  

```
(klayoutbuild) PS C:\Users\lsturm> python -V
Python 3.8.5
(klayoutbuild) PS C:\Users\lsturm> pip install klayout
Requirement already satisfied: klayout in c:\users\lsturm\miniconda3\envs\klayoutbuild\lib\site-packages (0.26.8)
```

closed 2020-09-23 by HansMalo