KLayout/klayout #658: $menu-path `location` does not work correctly: macOS-specific

Kazzz-S, 2020-10-17:
[PYA-OpenRecent.zip](https://github.com/KLayout/klayout/files/5394817/PYA-OpenRecent.zip)

Attached are simple PYA macros that manipulate the recently opened files.
They are functioning as intended. However, I cannot insert the menus to the right locations.
I've tested the macros with KLayout-0.26.8 on different platforms. It is a purely macOS specific issue.

[image]
[image]

Kazzz-S

klayoutmatthias, 2020-10-26:
@kazzz Thanks for reporting this. Any yes, you're right - it seems to be MacOS specific. I'll need to debug this problem there.

Maybe this will also fix the related issues from #657. There seems to be something wrong with the menu path deduction.

Best regards,

Matthias

klayoutmatthias, 2020-10-27:
I have provided PR #665 for the **0.26 branch** which hopefully fixes the problem. @kazzz maybe you can give that one a try - I'd need some test support on MacOS.

The proper ordering of the menu items was actually disabled because Qt segfaulted sometimes when menu items got deleted and re-inserted. The patch provides a somewhat more elaborate implementation of the menu sync function which - under normal circumstances - should maintain the menu item order.

Kazzz-S, 2020-10-27:
@klayoutmatthias,

Thank you for delivering PR #665. 
I want to test it and get back to you ASAP. However, I'm busy this week.
I hope I can report back around November 3rd.

Kazzz-S

Kazzz-S, 2020-11-02:
@klayoutmatthias,

I've tested PR #665, which works fine, as shown below. 
Different DMGs for **Catalina** are available [here](https://www.dropbox.com/sh/8827e8gsxnjlv6l/AAD49H1nDgK21m5C-DZBSnjUa?dl=0).

Perhaps OK on the older operating systems, too.
I'm going to check one by one.
Thanks a lot!

[image][image]

Kazzz-S

klayoutmatthias, 2020-11-28:
Fix is released in 0.26.9

closed 2020-11-28 by klayoutmatthias