KLayout/klayout #659: Python LYM with a "less than <" character in the <text> ~~~ </text> block

Kazzz-S, 2020-10-17:
[LessThanChar.zip](https://github.com/KLayout/klayout/files/5394950/LessThanChar.zip)

The attached ZIP contains two very simple LYMs.
They differ in an `if statement`, as shown below.
**Count_1** does not show up in the menu, but **Count_2** does.
I've confirmed this issue with KLyout-0.26.8 on Mac and Linux.

```
MacBookPro2{kazzz-S}(3)$ diff count1.lym count2.lym
3c3
<  <description>Count_1</description>>
---
>  <description>Count_2</description>>
22c22
<   if count < 10:
---
>   if count &lt; 10:
```

Kazzz-S

klayoutmatthias, 2020-10-26:
@Kazzz-S Thanks for reporting this issue!

I just wonder how this could happen. If I write a LYM and save it, there should not be an "<" character. Instead it should be replaced by ```&lt;```.

With the "<", the file isn't valid XML and can't be read. So the only solution is to prevent such files can be created.

Matthias

Kazzz-S, 2020-10-27:
As you mentioned [here](https://github.com/KLayout/klayout/issues/657#issuecomment-716858792),
> And I still wonder how the "<" character gets into the .lym files - maybe they were edited manually?

 I used an external text editor and inserted my Python code to <text>~~~</text> block manually.

Kazzz-S