KLayout/klayout #661: region select_overlapping miss properties

T-bag-A, 2020-10-24:
I have some shapes with properties , with layout.begin_shapes to form a region A.
then use  select_overlapping with A and other region B, 
C=A.select_overlapping(B)
I found the shapes in C missing the properties.
**expect result:**
the shapes in C keep the properties from A
I  try to find some clues, but does not know how to do.
###

 OriginalLayerRegionIterator  using shape.polygon to convert  shape to  polygon.
>   void set ()
    {
      while (! m_rec_iter.at_end () && ! (m_rec_iter.shape ().is_polygon () || m_rec_iter.shape ().is_path () || m_rec_iter.shape ().is_box ())) {
        ++m_rec_iter;
      }
      if (! m_rec_iter.at_end ()) {
        m_rec_iter.**shape().polygon**  (m_polygon);
        m_polygon.transform (m_iter_trans * m_rec_iter.trans (), false);
      }
    }

> bool Shape::polygon (Shape::polygon_type &p) const
{
  if (m_type == Polygon) {
    p = polygon ();
    return true;
  } else if (m_type == PolygonRef || m_type == PolygonPtrArrayMember) {
    polygon_ref ().instantiate (p);
    return true;
  } else if (m_type == SimplePolygon) {
    p.clear ();
    const simple_polygon_type &sp = simple_polygon ();
    p.assign_hull (sp.hull ());
    return true;
  } else if (m_type == SimplePolygonRef || m_type == SimplePolygonPtrArrayMember) {
    p.clear ();
    const simple_polygon_ref_type &sp = simple_polygon_ref ();
    p.assign_hull (sp.obj ().begin_hull (), sp.obj ().end_hull (), sp.trans (), false /*no additional compression*/);
    return true;
  } else if (m_type == Path) {
    p = path ().polygon ();
    return true;
  } else if (m_type == PathRef || m_type == PathPtrArrayMember) {
    p = path_ref ().obj ().polygon ();
    p.transform (path_ref ().trans (), false /*no additional compression*/);
    return true;
  } else if (is_box ()) {
    p = Shape::polygon_type (box ());
    return true;
  } else {
    return false;
  }
}

-----------------------
> const polygon_type *basic_ptr (polygon_type::tag) const
  {
    tl_assert (m_type == Polygon);
    if (m_stable) {
      return m_with_props ? &**(((ppolygon_iter_type *) m_generic.iter)) : &**(((polygon_iter_type *) m_generic.iter));
    } else {
      return m_with_props ? m_generic.ppolygon : m_generic.polygon;
    }
  }

klayoutmatthias, 2020-10-28:
I'm sorry, but select_overlapping is a pure polygon (or rather: merged polygon) operation. No properties supported. This is the way, DRC functions act. None of them preserves properties.

T-bag-A, 2021-03-10:
Maybe I should tell the story firstly. I have lots of defect from KLA , then i convert each defect to a box then add some defect meta info (defect id) into box properties. then after selectoverlapping with specific layer (eg M1) , i can identify which defect is true defect and get the defect id from box properties.
my soution
I have changed the OriginalLayerRegionIterator to delivery the Polygon or PolygonWithProperies. 
and change AsIfFlatRegion::selected_interacting_generic  by which polygon is not new generated, but selected by a index property_type from original input region:
  for (RegionIterator p (begin_merged ()); ! p.at_end (); ++p, ++n) {
    if ((selected.find (n) == selected.end ()) == inverse) {
    	db::Polygon * p1= const_cast <db::Polygon *>(&*p);
    	const db::PolygonWithProperties * pp= dynamic_cast< db::PolygonWithProperties *> (p1);
    	if (pp) {
    		output->raw_polygons ().insert (*pp);
    	}else {
    		output->raw_polygons ().insert (*p);
    	}
    }
  }
(current working，but should be review,may i issue a PR ?)

closed 2022-12-06 by klayoutmatthias