KLayout/klayout #662: pypi source missing version.sh

thomaslima, 2020-10-26:
A colleague tried to install with `pip install klayout`, but there was no wheel available for her system. I think it is Catalina with python 3.7. The setup failed as such:
```
 Using cached klayout-0.26.8.zip (2.0 MB)
    ERROR: Command errored out with exit status 1:
     command: /Users/chaoranhuang/opt/anaconda3/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/0t/8hpsjg7d4pbfcbwfhb8slhrr0000gn/T/pip-install-qg8hfmwu/klayout/setup.py'"'"'; __file__='"'"'/private/var/folders/0t/8hpsjg7d4pbfcbwfhb8slhrr0000gn/T/pip-install-qg8hfmwu/klayout/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/0t/8hpsjg7d4pbfcbwfhb8slhrr0000gn/T/pip-pip-egg-info-fdk9dulg
         cwd: /private/var/folders/0t/8hpsjg7d4pbfcbwfhb8slhrr0000gn/T/pip-install-qg8hfmwu/klayout/
    Complete output (7 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/0t/8hpsjg7d4pbfcbwfhb8slhrr0000gn/T/pip-install-qg8hfmwu/klayout/setup.py", line 535, in <module>
        version=config.version(),
      File "/private/var/folders/0t/8hpsjg7d4pbfcbwfhb8slhrr0000gn/T/pip-install-qg8hfmwu/klayout/setup.py", line 334, in version
        with open(version_file, "r") as file:
    FileNotFoundError: [Errno 2] No such file or directory: '/private/var/folders/0t/8hpsjg7d4pbfcbwfhb8slhrr0000gn/T/pip-install-qg8hfmwu/klayout/version.sh'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

I verified that the tarball klayout-0.26.8.zip does not contain the version.sh file required to compute the version. We need to include that in the `MANIFEST.in` file so the `sdist` includes it in the tarball. This has to go in the next version 0.26.9 or 0.27 because we can't replace what's there in pypi.

Thomas

thomaslima, 2020-10-26:
I have asked her to try with `pip install git+https://github.com/KLayout/klayout.git`, which downloads the "master" branch version from github. The g++ build failed here:
```
src/tl/tl/tlEnv.cc:52:15: error: use of undeclared identifier 'getenv'; did you mean 'get_env'?
    char *env = getenv (name.c_str ());
                ^~~~~~
                get_env
  src/tl/tl/tlEnv.cc:41:13: note: 'get_env' declared here
  std::string get_env (const std::string &name, const std::string &def_value)
              ^
  src/tl/tl/tlEnv.cc:52:9: error: no viable conversion from 'std::string' (aka 'basic_string<char, char_traits<char>, allocator<char> >') to 'char *'
    char *env = getenv (name.c_str ());
          ^     ~~~~~~~~~~~~~~~~~~~~~~
  src/tl/tl/tlEnv.cc:68:15: error: use of undeclared identifier 'getenv'; did you mean 'get_env'?
    char *env = getenv (name.c_str ());
                ^~~~~~
                get_env
  src/tl/tl/tlEnv.cc:41:13: note: 'get_env' declared here
  std::string get_env (const std::string &name, const std::string &def_value)
              ^
  src/tl/tl/tlEnv.cc:68:9: error: no viable conversion from 'std::string' (aka 'basic_string<char, char_traits<char>, allocator<char> >') to 'char *'
    char *env = getenv (name.c_str ());
          ^     ~~~~~~~~~~~~~~~~~~~~~~
```
@klayoutmatthias ideas? It works on mine, so this must be a compilation flag issue. I am attaching the full output: [output.txt](https://github.com/KLayout/klayout/files/5439485/output.txt)

klayoutmatthias, 2020-10-26:
Hi Thomas,

thanks for pointing this out. I'll add version.sh to the 0.26 branch as well.

Actually PyPI now allows deleting files! I could delete the .zip file - but after this and trying to upload a new .zip file, twine tells me the file is already is use! :( ... I'll try further to get it there.

Regarding the build error: apparently the compiler does not find the "getenv" function. So the header seems to be missing. This is what used to work so far:

```
...
#elif __APPLE__
#  include <libproc.h>
#  include <unistd.h>
...
```

For some reason the GitHub sources don't trigger the right include :(

Matthias

klayoutmatthias, 2020-10-26:
Stupid ... you can delete a version now, but still you can't re-upload a file that has been deleted :(

Forgot that ... so there is no source package now :(

I'll release 0.26.9 some time in the future anyway. Sorry for this mess.

Matthias

thomaslima, 2020-10-26:
No worries about pypi. It’s probably for the best anyway.

Just to clarify that the compilation works on my computer but not on my colleague’s. We use the same macOS but perhaps not the same SDK. Have you reproduced the error?

Thomas

thomaslima, 2020-10-26:
Ok so when running `g++ --version`, mine says:
```
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/c++/4.2.1
Apple clang version 12.0.0 (clang-1200.0.31.1)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```
My colleague's says:
```
$ g++ --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/c++/4.2.1
Apple LLVM version 7.3.0 (clang-703.0.31)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

so we are clearly using different compilers.

klayoutmatthias, 2020-10-28:
Thanks for clarifying this.

I wonder if "__APPLE__" is the right way to differentiate systems. Maybe it's only given on some compilers.

Matthias

thomaslima, 2020-11-12:
I am unaware of a combination of steps that leads me to a different `g++ --version`. I wonder if one of @Kazzz-S 's virtual machines has this compiler. It appears to be the default compiler of Sierra, which is over 4 years old now. Shall we ignore this?

thomaslima, 2020-11-12:
I asked her for the pre defined macro list of her compiler with the command `g++ -dM -E - < /dev/null`. The result (attached below) includes `__APPLE__`. In my IDE, VSCode identifies `getenv` as defined in `stdlib.h` in my toolkit. Hope it helps.

```
#define OBJC_NEW_PROPERTIES 1
#define _LP64 1
#define __APPLE_CC__ 6000
#define __APPLE__ 1
#define __ATOMIC_ACQUIRE 2
#define __ATOMIC_ACQ_REL 4
#define __ATOMIC_CONSUME 1
#define __ATOMIC_RELAXED 0
#define __ATOMIC_RELEASE 3
#define __ATOMIC_SEQ_CST 5
#define __BIGGEST_ALIGNMENT__ 16
#define __BLOCKS__ 1
#define __BYTE_ORDER__ __ORDER_LITTLE_ENDIAN__
#define __CHAR16_TYPE__ unsigned short
#define __CHAR32_TYPE__ unsigned int
#define __CHAR_BIT__ 8
#define __CONSTANT_CFSTRINGS__ 1
#define __DBL_DECIMAL_DIG__ 17
#define __DBL_DENORM_MIN__ 4.9406564584124654e-324
#define __DBL_DIG__ 15
#define __DBL_EPSILON__ 2.2204460492503131e-16
#define __DBL_HAS_DENORM__ 1
#define __DBL_HAS_INFINITY__ 1
#define __DBL_HAS_QUIET_NAN__ 1
#define __DBL_MANT_DIG__ 53
#define __DBL_MAX_10_EXP__ 308
#define __DBL_MAX_EXP__ 1024
#define __DBL_MAX__ 1.7976931348623157e+308
#define __DBL_MIN_10_EXP__ (-307)
#define __DBL_MIN_EXP__ (-1021)
#define __DBL_MIN__ 2.2250738585072014e-308
#define __DECIMAL_DIG__ __LDBL_DECIMAL_DIG__
#define __DYNAMIC__ 1
#define __ENVIRONMENT_MAC_OS_X_VERSION_MIN_REQUIRED__ 101100
#define __FINITE_MATH_ONLY__ 0
#define __FLT_DECIMAL_DIG__ 9
#define __FLT_DENORM_MIN__ 1.40129846e-45F
#define __FLT_DIG__ 6
#define __FLT_EPSILON__ 1.19209290e-7F
#define __FLT_EVAL_METHOD__ 0
#define __FLT_HAS_DENORM__ 1
#define __FLT_HAS_INFINITY__ 1
#define __FLT_HAS_QUIET_NAN__ 1
#define __FLT_MANT_DIG__ 24
#define __FLT_MAX_10_EXP__ 38
#define __FLT_MAX_EXP__ 128
#define __FLT_MAX__ 3.40282347e+38F
#define __FLT_MIN_10_EXP__ (-37)
#define __FLT_MIN_EXP__ (-125)
#define __FLT_MIN__ 1.17549435e-38F
#define __FLT_RADIX__ 2
#define __FXSR__ 1
#define __GCC_ATOMIC_BOOL_LOCK_FREE 2
#define __GCC_ATOMIC_CHAR16_T_LOCK_FREE 2
#define __GCC_ATOMIC_CHAR32_T_LOCK_FREE 2
#define __GCC_ATOMIC_CHAR_LOCK_FREE 2
#define __GCC_ATOMIC_INT_LOCK_FREE 2
#define __GCC_ATOMIC_LLONG_LOCK_FREE 2
#define __GCC_ATOMIC_LONG_LOCK_FREE 2
#define __GCC_ATOMIC_POINTER_LOCK_FREE 2
#define __GCC_ATOMIC_SHORT_LOCK_FREE 2
#define __GCC_ATOMIC_TEST_AND_SET_TRUEVAL 1
#define __GCC_ATOMIC_WCHAR_T_LOCK_FREE 2
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_1 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_16 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 1
[... 237 lines cut ...]
#define __WINT_TYPE__ int
#define __WINT_WIDTH__ 32
#define __amd64 1
#define __amd64__ 1
#define __apple_build_version__ 7030031
#define __block __attribute__((__blocks__(byref)))
#define __clang__ 1
#define __clang_major__ 7
#define __clang_minor__ 3
#define __clang_patchlevel__ 0
#define __clang_version__ "7.3.0 (clang-703.0.31)"
#define __core2 1
#define __core2__ 1
#define __llvm__ 1
#define __nonnull _Nonnull
#define __null_unspecified _Null_unspecified
#define __nullable _Nullable
#define __pic__ 2
#define __strong 
#define __tune_core2__ 1
#define __unsafe_unretained 
#define __weak __attribute__((objc_gc(weak)))
#define __x86_64 1
#define __x86_64__ 1
```

thomaslima, 2020-11-12:
And here is mine:

```
#define _LP64 1
#define __APPLE_CC__ 6000
#define __APPLE__ 1
#define __ATOMIC_ACQUIRE 2
#define __ATOMIC_ACQ_REL 4
#define __ATOMIC_CONSUME 1
#define __ATOMIC_RELAXED 0
#define __ATOMIC_RELEASE 3
#define __ATOMIC_SEQ_CST 5
#define __BIGGEST_ALIGNMENT__ 16
#define __BLOCKS__ 1
#define __BYTE_ORDER__ __ORDER_LITTLE_ENDIAN__
#define __CHAR16_TYPE__ unsigned short
#define __CHAR32_TYPE__ unsigned int
#define __CHAR_BIT__ 8
#define __CLANG_ATOMIC_BOOL_LOCK_FREE 2
#define __CLANG_ATOMIC_CHAR16_T_LOCK_FREE 2
#define __CLANG_ATOMIC_CHAR32_T_LOCK_FREE 2
#define __CLANG_ATOMIC_CHAR_LOCK_FREE 2
#define __CLANG_ATOMIC_INT_LOCK_FREE 2
#define __CLANG_ATOMIC_LLONG_LOCK_FREE 2
#define __CLANG_ATOMIC_LONG_LOCK_FREE 2
#define __CLANG_ATOMIC_POINTER_LOCK_FREE 2
#define __CLANG_ATOMIC_SHORT_LOCK_FREE 2
#define __CLANG_ATOMIC_WCHAR_T_LOCK_FREE 2
#define __CONSTANT_CFSTRINGS__ 1
#define __DBL_DECIMAL_DIG__ 17
#define __DBL_DENORM_MIN__ 4.9406564584124654e-324
#define __DBL_DIG__ 15
#define __DBL_EPSILON__ 2.2204460492503131e-16
#define __DBL_HAS_DENORM__ 1
#define __DBL_HAS_INFINITY__ 1
#define __DBL_HAS_QUIET_NAN__ 1
#define __DBL_MANT_DIG__ 53
#define __DBL_MAX_10_EXP__ 308
#define __DBL_MAX_EXP__ 1024
#define __DBL_MAX__ 1.7976931348623157e+308
#define __DBL_MIN_10_EXP__ (-307)
#define __DBL_MIN_EXP__ (-1021)
#define __DBL_MIN__ 2.2250738585072014e-308
#define __DECIMAL_DIG__ __LDBL_DECIMAL_DIG__
#define __DYNAMIC__ 1
#define __ENVIRONMENT_MAC_OS_X_VERSION_MIN_REQUIRED__ 101500
#define __FINITE_MATH_ONLY__ 0
#define __FLT16_DECIMAL_DIG__ 5
#define __FLT16_DENORM_MIN__ 5.9604644775390625e-8F16
#define __FLT16_DIG__ 3
#define __FLT16_EPSILON__ 9.765625e-4F16
#define __FLT16_HAS_DENORM__ 1
#define __FLT16_HAS_INFINITY__ 1
#define __FLT16_HAS_QUIET_NAN__ 1
#define __FLT16_MANT_DIG__ 11
#define __FLT16_MAX_10_EXP__ 4
#define __FLT16_MAX_EXP__ 16
#define __FLT16_MAX__ 6.5504e+4F16
#define __FLT16_MIN_10_EXP__ (-4)
#define __FLT16_MIN_EXP__ (-13)
#define __FLT16_MIN__ 6.103515625e-5F16
#define __FLT_DECIMAL_DIG__ 9
#define __FLT_DENORM_MIN__ 1.40129846e-45F
#define __FLT_DIG__ 6
#define __FLT_EPSILON__ 1.19209290e-7F
#define __FLT_EVAL_METHOD__ 0
#define __FLT_HAS_DENORM__ 1
#define __FLT_HAS_INFINITY__ 1
#define __FLT_HAS_QUIET_NAN__ 1
#define __FLT_MANT_DIG__ 24
[... 274 lines cut ...]
#define __amd64__ 1
#define __apple_build_version__ 12000031
#define __block __attribute__((__blocks__(byref)))
#define __clang__ 1
#define __clang_major__ 12
#define __clang_minor__ 0
#define __clang_patchlevel__ 0
#define __clang_version__ "12.0.0 (clang-1200.0.31.1)"
#define __code_model_small_ 1
#define __core2 1
#define __core2__ 1
#define __llvm__ 1
#define __nonnull _Nonnull
#define __null_unspecified _Null_unspecified
#define __nullable _Nullable
#define __pic__ 2
#define __seg_fs __attribute__((address_space(257)))
#define __seg_gs __attribute__((address_space(256)))
#define __strong
#define __tune_core2__ 1
#define __unsafe_unretained
#define __weak __attribute__((objc_gc(weak)))
#define __x86_64 1
#define __x86_64__ 1
```

klayoutmatthias, 2022-12-06:
I think this if fixed.

closed 2022-12-06 by klayoutmatthias