KLayout/klayout #669: Segmentation Fault related to RecursiveShapeIterator

thomaslima, 2020-11-05:
Hello Matthias,

klayout python module

Unfortunately I can't share a reproducible example on this one, first because it happens occasionally and second because I can't share the source code openly.

But essentially I have a fairly large hierarchical layout (4MB-worth) and have the following lines of naive python code:

```python
shapes_iter = layout.begin_shapes(TOP, layer_idx)
while not shapes_iter.at_end():
        shapes_iter.shape().delete()
        shapes_iter.next()
```

and/or

```python
while not shapes_iter.at_end():
        _shape = shapes_iter.shape()
        _shape.layer = layout.layer(layer_idx)
        shapes_iter.next()
```

Sometimes I get `Segmentation fault: 11` as a result. Maybe this is not the standard use case, but any ideas about what could be generating this segfault?

thomaslima, 2020-11-05:
Update: When I aggregate all shapes to a list first, and then iterate the list through python, in the second example (changing layers), I am getting the following error:

```
ERROR: src/tl/tl/tlReuseVector.h,277,mp_v->is_used (m_n)
Traceback (most recent call last):
    shape.layer = layout.layer(another_layer_idx)
RuntimeError: Internal error: src/tl/tl/tlReuseVector.h:277 mp_v->is_used (m_n) was not true in Shape.layer
```

Hopefully that helps.

thomaslima, 2020-11-05:
I fixed it by using `layout.move_layer(from_idx, to_idx)` method.

klayoutmatthias, 2020-11-11:
Hi Thomas,

First of all, deleting a shape while iterating isn't supported by the RecursiveShapeIterator. Essentially, manipulating the layout while iterating isn't supported by most of the iterators (also: the "each" iterator methods). Deleting a shape will invalidate pointers and such and make the application crash in unpredictable ways. It's always better to collect the objects you want to delete or modify in a first pass into a plain Python list or hash and then change it in a second pass.

And a shape can't just be moved to a different layer by assigning the layer property. Frankly I don't know why it's possible to assign the layer attribute at all. Instead, you have to insert a shape into the new layer and delete it in the previous place. But the idea of layer index assignment is interesting ... 

Anyway, if moving the whole layer solves your problem that's even much more efficient.

Matthias

thomaslima, 2020-11-12:
Thanks for replying @klayoutmatthias . 
1. I figured that changing objects while iterating is always a dangerous proposition. I just didn't expect a segfault. I did what you said before I produced my second example: I collected the objects into a list first.
2. Regarding moving shapes between layers, I was just following the [documentation](https://www.klayout.de/doc-qt5/code/class_Shape.html#method121). Contrary to what you stated above, the documentation states "void layer= (unsigned int layer_index) : Moves the shape to a layer given by the layer index object". Not only that, but this function worked for me for a long time, until I encountered a probable edge-case layout structure.

klayoutmatthias, 2020-11-13:
Hi Thomas,

I stand corrected regarding 2.) ... looks like I start forgetting the things I put into the code myself. It's been some years already and I'm not getting younger ...

Thanks,

Matthias

closed 2022-03-31 by thomaslima