KLayout/klayout #670: String properties not working in Python3's klayout package

thomaslima, 2020-11-06:
Hello Matthias. I think I might have found a bug related to #109 . I am trying to set a cell/layout property with a string name and a string value. But I noticed that they were not saving correctly. Then, I tried to save a property with name "123", but upon reading the gds file, it read as 123 (integer). The values were translated correctly.

Here's a sample of what I am talking about.
```
___________________________________________ test_metadata ____________________________________________

    def test_metadata():
        save_options = kdb.SaveLayoutOptions()
        save_options.gds2_write_file_properties = True
        save_options.gds2_write_cell_properties = True
        load_options = kdb.LoadLayoutOptions()
        load_options.properties_enabled = True
        layout = kdb.Layout()
        TOP = layout.create_cell("TOP")
        TOP.set_property("123", "string")
        layout.write("tests/tmp/test_metadata.gds", save_options)
        print(layout.has_prop_id(), TOP.has_prop_id())
        print([x for x in layout.properties(TOP.prop_id)])
        layout2 = kdb.Layout()
        layout2.read("tests/tmp/test_metadata.gds", load_options)
        TOP = layout2.top_cell()
        print(layout2.has_prop_id(), TOP.has_prop_id())
        print([x for x in layout2.properties(TOP.prop_id)])
>       assert TOP.property("123") == "test1"
E       AssertionError: assert None == 'test1'
E        +  where None = <built-in method property of Cell object at 0x1274728c0>('123')
E        +    where <built-in method property of Cell object at 0x1274728c0> = <klayout.dbcore.Cell object at 0x1274728c0>.property

tests/layout/test_metadata.py:24: AssertionError
---------------------------------------- Captured stdout call ----------------------------------------
False True
[['123', 'string']]
False True
[[123, 'string']]
====================================== short test summary info =======================================
FAILED tests/layout/test_metadata.py::test_metadata - AssertionError: assert None == 'test1'
```

klayoutmatthias, 2020-11-11:
Hi Thomas,

GDS does not have string names for properties. GDS only has numeric property keys. OASIS can handle string and numeric property "names". 

KLayout tries to convert the names into numbers if possible (i.e. if it sees that the string is a number) when saving GDS. 

This is a limitation of the GDS format, not of KLayout.

Matthias

thomaslima, 2020-11-12:
Thank you for this clarification. I edited the documentation as a tentative solution to this issue. :)

The pull request is not going to close this issue automatically because it is not targeting the master branch. I am closing this manually.

PR #674 (2020-11-13): Documentation clarification regarding properties in gds/oasis formats
- Fixes #670 

I figured I could help update the documentation which led me to error and to file the bug report in the first place. I hope this helps.
closed 2020-11-12 by thomaslima