KLayout/klayout #695: How to select polygon by the number of box enclosed in DRC check?

Zwlzzm, 2020-12-24:
Dear @klayoutmatthias,

Thank you for  providing so useful function for DRC.

I have  2 questions when doing DRC check and seek for your help and suggestions.

1.  How to select polygons if they has only one box of another layer?
 I could find way to achieve it.
2. The function of klayout DRC is  strong, and it achieves almost all the requirement,  however, for larger gds file (about 500M),
it is always very slow, and sometimes it cannot came out results in 1 way.  Is there any way to make it faster?

Looking forward to hearing from you!

Best regards!

klayoutmatthias, 2020-12-27:
Hi,

I'd appreciate if you could take such requests to the Forum: https://www.klayout.de/forum/

This is a bug tracker and not intended for support requests. 

Regarding the DRC performance:
* there is the tiling option (see https://www.klayout.de/forum/discussion/comment/5278#Comment_5278) and https://www.klayout.de/doc-qt5/manual/drc_runsets.html#h2-639
* you can try (somewhat experimental) hierarchical mode: https://www.klayout.de/doc-qt5/manual/drc_runsets.html#h2-718

For the polygon selection issue there is no simple solution for 0.26. With the next major release (0.27) you will be able to specify a count on "interact" which triggers interaction for a certain range of shape count the subject interacts with.

Matthias

Zwlzzm, 2020-12-28:
Dear  Matthias,

Thank you for respond and kindness.
When I sign up for klayout for the first time, there is not a button "New discussion", so I think I cannot put questions in klayuout website.  I am sorry for it and  I will put question on the Forum"https://www.klayout.de/forum/" next time

For the first question, I have add those lines to the DRC script as what you suggested.
But it is stuck  (the results cannot come out in 30 minutes) when it comes to 99%，no matter I change thread to 24 or 35, which is the maximum thread which the computer can support. 
And it is still stuck (the results cannot come out in 30 minutes) when I  change titles(titles=500,1000,2000,5000)
Do you have any other suggestions for this situation？

Generally, how long does it take do run verification for a large gds (about 600M) for you? Or in other words, if the running time is greater than how many minutes, we think it is stuck and the results cannot come out again?

For the second question，I will look forward to the 0.27 version released by you.

Wish you a happy new year！

Best regards!
Weiling

klayoutmatthias, 2020-12-28:
It's hard to say how long 600MB will take because the actual number of shapes can be huge (with a lot of hierarchy) or small (when flat). If the progress gets stuck at oo% in tiled mode with a decend tile size (i.e. 10% of the chip's width or height), then I assume it's because the DRC collected a lot of errors or huge intermediate layers.

I assume you can't send your test case, so I have no chance (as often) to debug the issue.

Matthias

Zwlzzm, 2020-12-31:
Dear Matthias,

Sorry for the late reply.  Does  the“ test case” mean both the gds file and the DRC file?
The gds file will be  a little trouble. I will try to prepare a test gds file and send it to your email together with the  DRC file. Can you help us test if there are methods to make the results come out then?

Best regards!
Weiling

closed 2022-12-06 by klayoutmatthias