KLayout/klayout #697: Core dump showing 2.5D view when geometry shader compilation fails

Partmedia, 2020-12-26:
I have a graphics driver that does not support ES 3.2.0. When I try to open the 2.5D view, I get an error message: "Shapes geometry shader compilation failed." When I press OK, the exception handler window starts to show up, but before they are fully renderered, KLayout dumps core with this backtrace:

```
#0  0x000000080f417af7 in QOpenGLShaderProgram::bind() () from /usr/local/lib/qt5/libQt5Gui.so.5
#1  0x000000081ae86f14 in lay::D25ViewWidget::paintGL (this=0x824df2300) at /home/kevinz/workspace/eda/klayout/src/plugins/tools/view_25d/lay_plugin/layD25ViewWidget.cc:881
#2  0x000000080eafb26e in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#3  0x000000080ead69a8 in QWidget::event(QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#4  0x000000080ea9bb00 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#5  0x000000080ea9d059 in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#6  0x0000000804a892c9 in lay::GuiApplication::notify (this=0x8184ba880, receiver=0x824df2300, e=0x7fffffff7c80) at ../../../src/lay/lay/layApplication.cc:1325
#7  0x000000080fe511a2 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Core.so.5
#8  0x000000080eac6ecc in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*)
    () from /usr/local/lib/qt5/libQt5Widgets.so.5
#9  0x000000080ead05a1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#10 0x000000080ead044a in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#11 0x000000080eac7213 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*)
    () from /usr/local/lib/qt5/libQt5Widgets.so.5
#12 0x000000080eaa8367 in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#13 0x000000080eaa6645 in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#14 0x000000080eaf76d4 in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#15 0x000000080eaf44a6 in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#16 0x000000080ea9bb00 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#17 0x000000080ea9d059 in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#18 0x0000000804a892c9 in lay::GuiApplication::notify (this=0x8184ba880, receiver=0x8231f9080, e=0x7fffffff8668) at ../../../src/lay/lay/layApplication.cc:1325
#19 0x000000080fe511a2 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Core.so.5
#20 0x000000080f0c4c79 in QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) () from /usr/local/lib/qt5/libQt5Gui.so.5
#21 0x000000080f0a41bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/qt5/libQt5Gui.so.5
#22 0x00000008186f1e6f in ?? () from /usr/local/lib/qt5/libQt5XcbQpa.so.5
#23 0x0000000810bff487 in g_main_context_dispatch () from /usr/local/lib/libglib-2.0.so.0
#24 0x0000000810bff84a in ?? () from /usr/local/lib/libglib-2.0.so.0
#25 0x0000000810bff906 in g_main_context_iteration () from /usr/local/lib/libglib-2.0.so.0
#26 0x000000080fead290 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/qt5/libQt5Core.so.5
#27 0x0000000804d277d8 in lay::signal_handler (signo=11, si=0x7fffffffa2b0) at ../../../src/lay/lay/laySignalHandler.cc:336
#28 0x000000080e1ddb7e in ?? () from /lib/libthr.so.3
#29 0x000000080e1dd13f in ?? () from /lib/libthr.so.3
#30 <signal handler called>
#31 0x000000080f417af7 in QOpenGLShaderProgram::bind() () from /usr/local/lib/qt5/libQt5Gui.so.5
#32 0x000000081ae86f14 in lay::D25ViewWidget::paintGL (this=0x824df2300) at /home/kevinz/workspace/eda/klayout/src/plugins/tools/view_25d/lay_plugin/layD25ViewWidget.cc:881
#33 0x000000080eafb26e in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#34 0x000000080ead69a8 in QWidget::event(QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#35 0x000000080ea9bb00 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#36 0x000000080ea9d059 in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#37 0x0000000804a892c9 in lay::GuiApplication::notify (this=0x8184ba880, receiver=0x824df2300, e=0x7fffffffb4d0) at ../../../src/lay/lay/layApplication.cc:1325
#38 0x000000080fe511a2 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Core.so.5
#39 0x000000080ead00ac in QWidgetPrivate::sendPaintEvent(QRegion const&) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#40 0x000000080eafbd25 in QOpenGLWidget::resizeEvent(QResizeEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#41 0x000000080ead69cf in QWidget::event(QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#42 0x000000080ea9bb00 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#43 0x000000080ea9d059 in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#44 0x0000000804a892c9 in lay::GuiApplication::notify (this=0x8184ba880, receiver=0x824df2300, e=0x7fffffffb908) at ../../../src/lay/lay/layApplication.cc:1325
#45 0x000000080fe511a2 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Core.so.5
#46 0x000000080ead3871 in QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#47 0x000000080ead3b2c in QWidget::setGeometry(QRect const&) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#48 0x000000080eabbbd4 in QWidgetItem::setGeometry(QRect const&) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#49 0x000000080eaaf2ad in QBoxLayout::setGeometry(QRect const&) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#50 0x000000080eab8633 in QLayoutPrivate::doResize() () from /usr/local/lib/qt5/libQt5Widgets.so.5
#51 0x000000080eab8b10 in QLayout::activate() () from /usr/local/lib/qt5/libQt5Widgets.so.5
#52 0x000000080ea9bae0 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#53 0x000000080ea9d059 in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#54 0x0000000804a892c9 in lay::GuiApplication::notify (this=0x8184ba880, receiver=0x82567cba0, e=0x7fffffffc188) at ../../../src/lay/lay/layApplication.cc:1325
#55 0x000000080fe511a2 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Core.so.5
#56 0x000000080eaf65fe in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#57 0x000000080eaf42ab in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#58 0x000000080ea9bb00 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#59 0x000000080ea9d059 in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#60 0x0000000804a892c9 in lay::GuiApplication::notify (this=0x8184ba880, receiver=0x8231f9080, e=0x7fffffffc188) at ../../../src/lay/lay/layApplication.cc:1325
#61 0x000000080fe511a2 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Core.so.5
#62 0x000000080f0c3aa9 in QGuiApplicationPrivate::processGeometryChangeEvent(QWindowSystemInterfacePrivate::GeometryChangeEvent*) () from /usr/local/lib/qt5/libQt5Gui.so.5
#63 0x000000080f0a41bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/qt5/libQt5Gui.so.5
#64 0x00000008186f1e6f in ?? () from /usr/local/lib/qt5/libQt5XcbQpa.so.5
[... 26 lines cut ...]
#91 0x000000080eaf4140 in ?? () from /usr/local/lib/qt5/libQt5Widgets.so.5
#92 0x000000080ea9bb00 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
#93 0x000000080ea9d059 in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/qt5/libQt5Widgets.so.5
[... cut ...]
#111 0x00000008008c33ba in rb_vm_exec () from /usr/local/lib/libruby27.so.27
#112 0x000000080073c1d0 in ruby_run_node () from /usr/local/lib/libruby27.so.27
#113 0x0000000802f71d23 in rba::RubyInterpreter::initialize (main_argc=@0x7fffffffe548: 1, main_argv=0x812dcb440, main_func=0x205390 <klayout_main_cont(int&, char**)>) at ../../../src/rba/rba/rba.cc:1913
#114 0x00000000002050a4 in klayout_main (argc=@0x7fffffffe548: 1, argv=0x812dcb440) at ../../../src/klayout_main/klayout_main/klayout.cc:195
```

klayoutmatthias, 2020-12-27:
Thanks for reporting this.

Do you mean your driver doesn't support ES in general, or only version 3.2.0? Which is the version your driver supports? 

Sorry, for asking, but I'm trying to figure out the minimum viable requirements.

Thanks,

Matthias

Partmedia, 2020-12-27:
I appreciate you investigating. I would consider this issue fixed as long as it stops dumping core, but I certainly do also like to be able to try the 2.5D view out!

My `glxinfo` reports:
```
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 20.2.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
```

My graphics card is the Intel(R) HD Graphics 4000 mobile integrated graphics chipset. This is integrated on an Ivy Bridge class Intel CPU from circa ~2012.

klayoutmatthias, 2020-12-28:
@Partmedia thanks for these inputs.

I actually tried porting the view down to OpenGL ES 3.0, but lacking the geometry shader I'd need to precompute the face normals for the coloring. That comes with some memory penalty and I'm not firm enough in OpenGL to know how to overcome this. So for now, it's going to be OpenGL ES 3.2 :(

But I was able to fix the segfault :)

Partmedia, 2020-12-30:
Thanks for taking a look. I can confirm that the branch fixes the segfault. Thanks for also taking a look at ES 3.0; even though it's been inconclusive now.

(P.S. switching branches and recompiling took a little over an hour for me -- is this similar for you?)

klayoutmatthias, 2020-12-30:
@Partmedia Yes, full recompilation takes some time. You can speed it up considerably by skipping the Qt binding if you need ("-without-qtbinding"). 

I also provide continuous builds for various platforms on the master branch here: http://www.klayout.org/downloads/master/. As the PR is now merged, you should see a new package there soon.

closed 2020-12-30 by klayoutmatthias