KLayout/klayout #700: LEF/DEF import mixes up layers

antonblanchard, 2021-01-01:
The following test case shows the `met4` layer as `cldntm.mask`: [klayout-def-layer-issue.tar.gz](https://github.com/KLayout/klayout/files/5759319/klayout-def-layer-issue.tar.gz)

It looks like klayout is truncating the layer numbers and `met4` 71 is getting mapped to `cldntm.mask` 11. If I remove `cldntm.mask` from the layer properties file, `met4` is shown correctly.

klayoutmatthias, 2021-01-20:
@antonblanchard Thanks for the test case.

I actually don't see any layer mapping from LEF/DEF to GDS layers inside the tech file. The layer numbers are auto-assigned and 11 is simple the next available layer.

To fix this, the tech file needs a LEF/DEF reader layer mapping. This will tell the reader how to translate a layer name/purpose combination from LEF/DEF into GDS layer/datatype.

I have attached an enhanced version of the .lyt file, but that's only halfway done. I just included the metal and via layers as far as I could identify them. I assume you took that from somewhere so maybe you can feedback this enhancement request there.

The LEF/DEF input layer map is found in the technology manager under "Reader Options", "Layer Mapping" at the bottom of the LEF/DEF tab. 

[sky130A.zip](https://github.com/KLayout/klayout/files/5845908/sky130A.zip)

Matthias

antonblanchard, 2021-01-21:
@klayoutmatthias thank you for the detailed response, the issue makes sense to me now.

The file came from `open_pdks`, which has since been updated to use https://github.com/laurentc2/SKY130_for_KLayout/ . There are some LEF/DEF to GDS layer mappings, but it is incomplete. I'll raise an issue there about it.

klayoutmatthias, 2021-09-24:
I think this is not a KLayout bug, so I'll close this topic for now.

Thanks,

Matthias

closed 2021-09-24 by klayoutmatthias