KLayout/klayout #701: Ruby 3.0?

wshanks, 2021-01-09:
Should KLayout work with Ruby 3.0? After building it with Ruby 3.0, I get an "Application crashed" message with this "Problem details" backtrace:

```
Signal number: 11
Address: 0x8
Program Version: KLayout 0.26.9 (2021-01-08 rLatestSourcePackage)

Backtrace:
/app/bin/libklayout_lay.so.0 +0x2968b3
/usr/lib/x86_64-linux-gnu/libpthread.so.0 +0x13a70
/app/lib/libruby.so.3.0 +0xcdcad
/app/lib/libruby.so.3.0 +0xcde41
/app/bin/libklayout_rba.so.0 +0x3e311
/app/bin/libklayout_rba.so.0 +0x3e56e
/app/bin/libklayout_rba.so.0 +0x55b4c
/app/bin/libklayout_rba.so.0 +0x4fe14
/app/bin/libklayout_rba.so.0 +0x2662b
/app/lib/libruby.so.3.0 +0x234b83
/app/lib/libruby.so.3.0 +0x23db76
/app/lib/libruby.so.3.0 +0x2407a2
/app/lib/libruby.so.3.0 +0x245895 rb_vm_exec
/app/lib/libruby.so.3.0 +0x24aa84 rb_vm_call0
/app/lib/libruby.so.3.0 +0x25141a rb_funcallv_kw
/app/lib/libruby.so.3.0 +0x14bc92 rb_class_new_instance_pass_kw
/app/lib/libruby.so.3.0 +0x234b83
/app/lib/libruby.so.3.0 +0x23db76
/app/lib/libruby.so.3.0 +0x2407a2
/app/lib/libruby.so.3.0 +0x245895 rb_vm_exec
/app/lib/libruby.so.3.0 +0x246522
/app/lib/libruby.so.3.0 +0x24a848 rb_vm_call0
/app/lib/libruby.so.3.0 +0x24cbfb rb_funcallv
/app/lib/libruby.so.3.0 +0xb508a rb_protect
/app/bin/libklayout_rba.so.0 +0x45109
/app/bin/libklayout_rba.so.0 +0x45225
/app/bin/libklayout_lym.so.0 +0x33da1 _ZNK3lym5Macro3runEv
/app/bin/libklayout_lym.so.0 +0x362e5
/app/bin/libklayout_lym.so.0 +0x36291
/app/bin/libklayout_lay.so.0 +0x11c22d _ZN3lay15ApplicationBase8init_appEv
klayout +0x40f3
/app/bin/libklayout_rba.so.0 +0x22e20
/app/lib/libruby.so.3.0 +0x234b83
/app/lib/libruby.so.3.0 +0x23db76
/app/lib/libruby.so.3.0 +0x2407a2
/app/lib/libruby.so.3.0 +0x245895 rb_vm_exec
/app/lib/libruby.so.3.0 +0xb3486
/app/lib/libruby.so.3.0 +0xb725a ruby_run_node
/app/bin/libklayout_rba.so.0 +0x23ece _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
klayout +0x3d3e
klayout +0x38d5
/usr/lib/x86_64-linux-gnu/libc.so.6 +0x24022 __libc_start_main
klayout +0x3a3e
```

I get the error three times and in the console I see:

```
sh: addr2line: command not found
ERROR: :/built-in-macros/lvs_interpreters.lym:62: Operation cancelled in Application::instance
  :/built-in-macros/lvs_interpreters.lym:62:in `instance'
  :/built-in-macros/lvs_interpreters.lym:62:in `initialize'
  :/built-in-macros/lvs_interpreters.lym:130:in `new'
  :/built-in-macros/lvs_interpreters.lym:130:in `<module:LVS>'
  :/built-in-macros/lvs_interpreters.lym:2:in `<main>' (class RuntimeError)
sh: addr2line: command not found
ERROR: :/built-in-macros/drc_install.lym:6: Operation cancelled in Application::instance
  :/built-in-macros/drc_install.lym:6:in `instance'
  :/built-in-macros/drc_install.lym:6:in `<module:DRC>'
  :/built-in-macros/drc_install.lym:2:in `<main>' (class RuntimeError)
sh: addr2line: command not found
ERROR: :/built-in-macros/lvs_install.lym:6: Operation cancelled in Application::instance
  :/built-in-macros/lvs_install.lym:6:in `instance'
  :/built-in-macros/lvs_install.lym:6:in `<module:LVS>'
  :/built-in-macros/lvs_install.lym:2:in `<main>' (class RuntimeError)
```

After the three crash windows, KLayout starts and seems to operate fine. I tried running some simple commands in the Ruby interpreter and they seemed to work. For context, I maintain [the flatpak package](https://github.com/flathub/de.klayout.KLayout) for KLayout on flathub.org. I bundle Ruby in the flatpak to provide as much KLayout functionality as possible, but I don't use the Ruby bindings myself. I can keep Ruby pinned to version 2 until KLayout supports version 3 if that is the problem. I just like to keep the components in the flatpak as up to date as possible.

klayoutmatthias, 2021-01-15:
@wshanks Thanks for trying that and for sharing your experiences. 

I have not tried building with Ruby 3.0 and I'm a bit surprised it works even a little. So far, Ruby 2 still seems to be mainstream and I hope it will stay like this for a while. But I'll try to fix Ruby 3 compatibility occasionally. It's always good to be prepared.

Thanks,

Matthias

kaugummiheld, 2021-04-19:
Hey, 
maybe just as a confirmation/clarification. I just installed KLayout from the official Manjaro Repo (not as an AUR package) and I am facing the same errors/warnings as @wshanks. I'm not using Ruby actively, but of course maybe some other program on my PC. 
Am I save to just ignoring this or should I do something about it? (Tbh I'm not yet attached to KLayout, but a college was recommending it.)

wshanks, 2021-04-19:
I don't have any update over what I said above (that I tried some simple uses of the Ruby bindings and they seemed to work despite the error messages). I pinned Ruby in the flatpak to version 2 though, so I haven't tested beyond that initial test.

Perhaps [the bump to Ruby 3.0 in the Arch package](https://github.com/archlinux/svntogit-community/commit/6201eddd2abdc0fea7ae0431e0975dfa17846c23#diff-3e341d2d9c67be01819b25b25d5e53ea3cdf3a38d28846cda85a195eb9b7203a) was an automated rebuild that no one tested (other than checking that there were no build errors). The Arch package should be updated to build against the ruby2.7 package instead of the ruby package, since from this issue we know that klayout does not currently support Ruby 3.0.

cfriesicke, 2021-04-26:
I also have this problem with the latest Arch/Manjaro package for KLayout 0.26.11. This should probably brought to the attention of the maintainer by filing a bug in the Arch package repository.
Is there any fix that can meanwhile be applied to get this running locally? Maybe @wshanks or @kaugummiheld you have a temporary solution or workaround?

flaport, 2021-04-26:
Clicking the windows away does indeed solve most issues for the GUI but the CLI does unfortunately not work...

klayoutmatthias, 2021-04-26:
Frankly, the crash is a serious issue and if it's dismissed there is a good chance something is corrupted.

Is that all related to Ruby 3.0?

Usually the distributions will still provide Ruby 2.x. So it should be possible to build against that. Ruby 3.0 is without warranty.

Matthias

flaport, 2021-04-27:
FYI, I opened a bug report here: https://bugs.archlinux.org/task/70601

cfriesicke, 2021-04-27:
Thanks, @flaport -- very much appreciated!

cfriesicke, 2021-05-19:
The Ruby dependency is now fixed in Arch/Manjaro. I just received an update of the community/klayout package (version 0.27-2) that now again depends on Ruby 2.7.

klayoutmatthias, 2021-08-24:
I'm building regularly on Windows now with Ruby 3.0.

Can I close this ticket?

Thanks,

Matthias

jlemonde, 2022-04-19:
I get almost the same error when launching `klayout 0.26.2-3build3 amd64`, installed as a deb on Ubuntu 22.04 LTS (pre-release, the version I'm using will get released next week). This same installed deb worked flawlessly in Ubuntu 21.10. Note that earlier versions are not yet available in Ubuntu, and trying to manually install klayout 0.27  from the deb executables provided on the Klayout download page fails due to missing dependencies (libruby2.7 and libpython3.8).

```
Signal number: 11
Address: 0x8
Program Version: KLayout 0.26.2 (2022-01-13 rLatestSourcePackage)

Backtrace:
/usr/lib/klayout/libklayout_lay.so.0 +0x24b786 lay::SaltDownloadManager::execute(lay::SaltManagerDialog*, lay::Salt&) [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x42520
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0xbf649
/usr/lib/klayout/libklayout_rba.so.0 +0x43782
/usr/lib/klayout/libklayout_rba.so.0 +0x47ce2
/usr/lib/klayout/libklayout_rba.so.0 +0x2a78d
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x22fc15
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x231fa6
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x234c12
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x23a2ed rb_vm_exec
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x241275
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x241d86 rb_funcallv_kw
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x13e1bb rb_class_new_instance_pass_kw
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x22fc15
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x231fa6
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x234c12
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x23a2ed rb_vm_exec
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x23b052
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x23f2d7
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x24291c rb_funcallv
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0xa69bd rb_protect
/usr/lib/klayout/libklayout_rba.so.0 +0x40d9e
/usr/lib/klayout/libklayout_lym.so.0 +0x2f2e2 _ZNK3lym5Macro3runEv
/usr/lib/klayout/libklayout_lym.so.0 +0x2f455
/usr/lib/klayout/libklayout_lym.so.0 +0x2f401
/usr/lib/klayout/libklayout_lay.so.0 +0x10e4ed _ZN3lay15ApplicationBase8init_appEv
/usr/lib/klayout/klayout +0x4595
/usr/lib/klayout/libklayout_rba.so.0 +0x24944
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x22fc15
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x231fa6
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x234c12
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x23a2ed rb_vm_exec
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0xa5328
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0xa8e2a ruby_run_node
/usr/lib/klayout/libklayout_rba.so.0 +0x273e2 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/usr/lib/klayout/klayout +0x3d1e
/lib/x86_64-linux-gnu/libc.so.6 +0x29d90
/lib/x86_64-linux-gnu/libc.so.6 +0x29e40 __libc_start_main
/usr/lib/klayout/klayout +0x3e95

```

klayoutmatthias, 2022-04-19:
Please build from source rather than installing something for an OS it was not made for. No warranties given in that case.

I will start preparing deb packages for Ubuntu 22 as soon it is released.

If you like to build for Ubuntu 22 yourself - I had to install the following packages: gcc g++ make libz-dev qtbase5-dev qttools5-dev libqt5xmlpatterns5-dev qtmultimedia5-dev libqt5multimediawidgets5 libqt5svg5-dev ruby ruby-dev python3 python3-dev

Matthias

wshanks, 2022-04-19:
@jlemonde you could also try the Flatpak on 22.04 if you like 🙂

jlemonde, 2022-04-20:
Dear @wshanks, I tried the flatpak. It installs flawlessly, but then requires 40+ seconds to launch and finally I'm unable to open my files (File>Open>Select a file>OK) just does nothing at all. Then I tried to close Klayout (First Alt+f4, then by clicking the x button) and it froze completely.

wshanks, 2022-04-20:
Sorry to hear that. I tested just now on 22.04 and it worked fine for me. I don't want to clutter things here. You can open an issue on https://github.com/flathub/de.klayout.KLayout if you want to look into the flatpak package more.

jlemonde, 2022-04-21:
Dear Matthias,
I compiled successfully klayout 0.27.8 on (K)ubuntu 22.04 LTS (using the -qt5 flag) after installing the dependencies you suggested. Thank you!
Now how should I go ahead and install it? I can't find this in the README. 
Kind regards,
Johannes

klayoutmatthias, 2022-04-23:
The easiest way to "install" klayout is to use "/usr/bin" for the binary path ("build.sh -bin /usr/bin"). But you need to be root be able to install it. You can also execute directly from the binary path where the build script has installed the binary. You may need to set LD_LIBRARY_PATH to this directory before you run klayout.