KLayout/klayout #713: Mac DMGs for 0.26.10

Kazzz-S, 2021-01-27:
Dear @klayoutmatthias,

Thank you for merging PR #711.
I've test built 12439624  and got an error in the QA test as below.

```
Running tlWebDAV:1
    Signal number: 11
    Address: 0x18
    Program Version: KLayout 0.26.10 (2021-01-27 r124396247)

    Backtrace:
    /Users/sekigawa/GitWork/klayout/qt5MP.build.macos-Catalina-release-RsysPsys.macQAT/libklayout_lay.0.26.10.dylib +0x1fa5ab _ZN3lay14signal_handlerEiP9__siginfoPv
    /usr/lib/system/libsystem_platform.dylib +0x35fd _sigtramp
 :
 :
```

* The full log is [here](https://www.dropbox.com/s/ft1cjjbfktfu11o/QATest_124396247_2021_0127_2217__qt5MP.build.macos-Catalina-release-RsysPsys.macQAT.log?dl=0).
* The standard DMG for **Catalina** is [in this directory](https://www.dropbox.com/sh/gue33snflslu0ys/AACrrm8nG7K4afVE6mgRnAL3a?dl=0).

Warm regards,
Kazzz-S

klayoutmatthias, 2021-01-31:
Dear @Kazzz-S,

Thanks for reporting this issue.

There are no changes in the area of this feature and the log does not reveal any details which could help me debug that issue.

I'm actually considering to switch the network functionality to libcurl. I have some weird issues with proxies and redirects on Windows too. I'll try if I can reproduce the issue and if using curl will make a difference.

Thanks,

Matthias

klayoutmatthias, 2021-01-31:
Update: I just built 0.26.10 on a VirtualBox with Catalina. My configuration was Anaconda with Python 3.8. 

So far the tests pass. I could not reproduce the tlWebDAV issue :(

Kazzz-S, 2021-01-31:
Update from my side: I could not reproduce this issue either in the Anaconda with Python 3.8 environment.

I will test in all possible environments while building 0.26.10.

Kazzz-S, 2021-02-01:
## 3. Homebrew with Qt 5.15.2 ##

```
$ ./macQAT.py -x 'pymod,pya' -r
```
```
Running tlWebDAV:1
    Signal number: 11
    Address: 0x18
    Program Version: KLayout 0.26.10 (2021-02-01 rc3c8f20e3)

    Backtrace:
    /Users/sekigawa/GitWork/klayout/qt5Brew.build.macos-Catalina-release-Rhb27Phb38.macQAT/libklayout_lay.0.26.10.dylib +0x1fa5ab _ZN3lay14signal_handlerEiP9__siginfoPv
    /usr/lib/system/libsystem_platform.dylib +0x35fd _sigtramp
    /usr/lib/system/libsystem_platform.dylib +0xffff80008f860000 _sigtramp
   :
```
* [Full Log](https://www.dropbox.com/s/8rguvopa0c2hz5p/QATest_c3c8f20e3_2021_0201_2037__qt5Brew.build.macos-Catalina-release-Rhb27Phb38.macQAT.log?dl=0)
* The issue reproduced.
----
```
$ ./macQAT.py -x 'pymod,pya,tlWebDAV' -r
```
```
Running tlHttpStream:1
    Signal number: 11
    Address: 0x18
    Program Version: KLayout 0.26.10 (2021-02-01 rc3c8f20e3)

    Backtrace:
    /Users/sekigawa/GitWork/klayout/qt5Brew.build.macos-Catalina-release-Rhb27Phb38.macQAT/libklayout_lay.0.26.10.dylib +0x1fa5ab _ZN3lay14signal_handlerEiP9__siginfoPv
    /usr/lib/system/libsystem_platform.dylib +0x35fd _sigtramp
    /usr/lib/system/libsystem_platform.dylib +0xffff80008f860000 _sigtramp
  :
```
* [Full Log](https://www.dropbox.com/s/1p2yc6xylldnxou/QATest_c3c8f20e3_2021_0201_2042__qt5Brew.build.macos-Catalina-release-Rhb27Phb38.macQAT.log?dl=0)
* Another issue...
----
```
$ ./macQAT.py -x 'pymod,pya,tlWebDAV,tlHttpStream' -r
```
```
:
        rdbRVEReaderTests:2
    276 test(s) skipped
    All tests passed.
Grand total time: 352.445s (wall) 301.51s (user) 71.08s (sys)
```
[Full Log](https://www.dropbox.com/s/18f1js3kex5fd0t/QATest_c3c8f20e3_2021_0201_2044__qt5Brew.build.macos-Catalina-release-Rhb27Phb38.macQAT.log?dl=0)
* I needed to exclude **tlWebDAV** and **tlHttpStream** to run through the tests.

klayoutmatthias, 2021-02-02:
Hi @kazzz,

I really appreciate you insisted on this! :)

I was finally able to debug and fix the issue and I think this will also fix a problem that was haunting me for years on Windows! 

Technically the cause wasn't the 0.26 version but the fact that I switched the www.klayout.org server to always use https and enabled automatic redirect from http to https. It turned out that in the HTTP implementation of KLayout I deleted the reply object too early when following this redirection. While this did not create issues on Linux, it caused the crash you observed.

I also believe that this is responsible for mysterious "Error 99" fails I sporadically see on Windows. I need to confirm this, but if that is fixed, it will really be a big step forward.

I have added the patch to 0.26 and master.

Thanks and best regards,

Matthias

Kazzz-S, 2021-02-02:
## 4. Homebrew with Qt 5.15.2 again ##

Hi @klayoutmatthias,

Bingo! Thank you for your effort.
* I have tested ab7db78e.
```
$ ./macQAT.py -x 'pymod,pya' -r
```

```
 : 
        dbGerberImport:X2_5b
        rdbRVEReaderTests:1
        rdbRVEReaderTests:2
    276 test(s) skipped
    All tests passed.
Grand total time: 358.764s (wall) 319.73s (user) 55.59s (sys)
```

* [Full Log](https://www.dropbox.com/s/vlujlkbpnv9au48/QATest_ab7db78ee_2021_0203_0705__qt5Brew.build.macos-Catalina-release-Rhb27Phb38.macQAT.log?dl=0)

Regards,
Kazzz-S

Kazzz-S, 2021-02-03:
Hi @klayoutmatthias,

Thank you for making the maintenance release of 0.26.10.
I was supposed to build different DMG packages for Mac, as usual; however, I got hardware trouble while building!

So far, no idea how harmful the trouble is and how long it will take to repair. I'm going to contact Apple support this evening.

Kazzz-S

Kazzz-S, 2021-02-06:
Dear @klayoutmatthias,

Thank you very much for making the maintenance release.

As usual, different DMG packages of **KLayou 0.26.10 for Mac** are available as below [0.26.10 Dropbox Link](https://www.dropbox.com/sh/a9gito9vzxcrfam/AAB2OZWU5HgJPslPq6fwD2s8a?dl=0).
**Please note that _pymod_ is not included in those packages.**

## DMG file naming convention ##
**0.26.10 for Catalina** is taken as an example here.
[image]

where
(a) Type of package:  ST-: standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt5 type:  qt5MP: MacPorts; qt5Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:   sys: OS-bundled; mp27: Ruby2.7 from MacPorts; hb27: Ruby2.7 from Homebrew; ana3: Ruby2.5 from Anaconda3
(f) **P**ython type: sys: OS-bundled; mp38: Python 3.8 from MacPorts; hb38: Python 3.8 from Homebrew; ana3: Python3.8 from Anaconda3

## For the primary platform, macOS Catalina (10.15) ##
**LW- and HW- packages use Python 3.8 in MacPorts, Homebrew and Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
1|Catalina (10.15)|ST-klayout-0.26.10-macOS-Catalina-1-qt5MP-RsysPsys.dmg|Homebrew|OS-bundled|OS-bundled|(1)|
2|Catalina (10.15) with MacPorts dev. env.|LW-klayout-0.26.10-macOS-Catalina-1-qt5MP-Rmp27Pmp38.dmg|MacPorts|MacPorts|MacPorts|(2)|
3|Catalina (10.15) with Homebrew dev. env.|LW-klayout-0.26.10-macOS-Catalina-1-qt5Brew-Rhb27Phb38.dmg|Homebrew|Homebrew|Homebrew|(3)|
4|Catalina (10.15) with Anaconda3 dev. env.|LW-klayout-0.26.10-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(4)|
5|Catalina (10.15)|HW-klayout-0.26.10-macOS-Catalina-1-qt5Brew-RsysPhb38.dmg|Homebrew|OS-bundled|Homebrew|(5)|

(1) shares OS-bundled Ruby2.6 and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).
(2) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(3) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(4) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(5) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.8 from Homebrew are embedded, hence, heavy.

## For the latest macOS Big Sur (11.2) ##
Five macOS DMGs for **Big Sur** are ready for experimental purposes.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 12.1.0.

**LW- and HW- packages use Python 3.8 in MacPorts, Homebrew and Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
21|BigSur (11.2)|ST-klayout-0.26.10-macOS-BigSur-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(21)|
22|BigSur (11.2) with MacPorts dev. env.|LW-klayout-0.26.10-macOS-BigSur-1-qt5MP-Rmp27Pmp38.dmg|MacPorts|MacPorts|MacPorts|(22)|
23|BigSur (11.2) with Homebrew dev. env.|LW-klayout-0.26.10-macOS-BigSur-1-qt5Brew-Rhb27Phb38.dmg|Homebrew|Homebrew|Homebrew|(23)|
24|BigSur (11.2) with Anaconda3 dev. env.|LW-klayout-0.26.10-macOS-BigSur-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(24)|
25|BigSur (11.2)|HW-klayout-0.26.10-macOS-BigSur-1-qt5Brew-RsysPhb38.dmg|Homebrew|OS-bundled|Homebrew|(25)|

(21) **experimental**; shares OS-bundled Ruby2.6 and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).
(22) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(23) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(24) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(25) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.8 from Homebrew are embedded, hence, heavy.

## For previous OS ##
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
6|Mojave (10.14)|ST-klayout-0.26.10-macOS-Mojave-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1p)|
7|HighSierra ( 10.13)|ST-klayout-0.26.10-macOS-HighSierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1p)|
8|Sierra (10.12)|ST-klayout-0.26.10-macOS-Sierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1p)|
9|ElCapitan (10.11)|ST-klayout-0.26.10-MacOSX-ElCapitan-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled| (1p) |
x|Yosemite (10.10)|has been dropped due to the compiler issue reported in #360.|

(1p) shares OS-bundled Ruby[2.0 ~ 2.3] and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).

Please close this ticket after publishing the packages.
With warm regards,
Kazzz-S

klayoutmatthias, 2021-02-23:
Closing this issue as the DMG are published.

closed 2021-02-23 by klayoutmatthias