KLayout/klayout #715: Object Selection larger than drawn

demisjohn, 2021-01-28:
When dragging to select multiple objects, the actual selection rectangle is invisibly lrger than what is drawn on screen.
Here is a video showing this.

https://user-images.githubusercontent.com/5370181/106155261-527cbe80-6135-11eb-84d3-93105d63a884.mov

This is an array of circle-like polygons, you can see that where I end the selection box, but objects are selected in an area larger than I expected.

(Klayout v. 0.26.9, Mac OS 10.14.6, MacBook Pro "MacBookPro14,2")

klayoutmatthias, 2021-01-29:
Cool .. I did not know you can add videos too.

Maybe the circles are within an instance which is rotated?

If so, the explanation is that the selection rectangle gets transformed "into" the instance. But as it stays a rectangle it will get larger - the bounding box of a rotated rectangle is bigger than the original.

If so, I'm afraid to fix that the concept of the "selection rectangle" had to be extended to a "selection polygon". That's a more difficult topic then. Not impossible, but requires some effort.

demisjohn, 2021-03-02:
Actually no - this is a single-cell (flattened) file.  No instances as far as I can tell.
It is a 250MB file - here is a link to the GDS in question to test the issue yourself.
https://drive.google.com/file/d/1-Hb8RgUaIyyx1mAqIFlFc7IFNe59fj8b/view?usp=sharing

klayoutmatthias, 2021-04-14:
@demisjohn Sorry, I missed this ticket and it deserves a response.

There is a parameter called "capture range": 

[image]

It expands the selection box by the specified number of pixels. If you set it 0, the captured shapes should all be inside the box selected.

Matthias

demisjohn, 2021-04-14:
Yes that was indeed the issue - thanks!  I assume it's set greater than 0 to start because there is some case in which this is important.  This explains some confusion I have previously had over the selection-box, where I hadn't been able to always predict exactly what would be selected, requiring me to zoom in in order to do more careful selections between adjacent objects.

demisjohn, 2021-04-16:
@klayoutmatthias 
I see the issue with the "Capture range to mouse cursor" setting:

- If it is set to 0, then clicking with the mouse is **unable to select an object**, 
- however, setting it to 0 makes Dragging a selection box **invisibly larger than drawn** on the screen.

Possibly remedies:
1) Split "capture range" into 2 separate settings: one for clicking, and one for dragging.  I assume this would be a lot of work and bug fixing.
2) Make the Selection Box, that is drawn dynamically during a Drag, draw the *real* selection window (include the capture range in what is drawn).  It should draw a larger selection box than just where you clicked to begin with. The selection box is not invisible this way, but actual behavior is the same.

klayoutmatthias, 2021-04-18:
Solution 1 was actually easier. This is how the options look like now:

[image]

The first range applies to single clicks. The second one to boxes. Default for boxes is 0 so only things really inside the box get selected.

demisjohn, 2021-04-19:
Very nice, looks good.  Impressively fast fix!

commit 7ef56c28 (2021-04-19): Merge pull request #774 from KLayout/issue-715

Fixes #715 (selection box larger than drawn)
commit af6f11c2 (2021-04-19): Fixes #715 (selection box larger than drawn)
closed 2021-04-19 by demisjohn