KLayout/klayout #716: Parallel build options?

mguthaus, 2021-01-28:
Hi Matthias,

Is there an option for multi-threaded building? Klayout takes... a long time to build. I don't see anything in the ./build.sh -h information.

Matt

thomaslima, 2021-01-28:
Not sure it is relevant to you but I use 4 threads for building with `-option -j4`.

klayoutmatthias, 2021-01-29:
@thomaslima Thanks for this comment.

build.sh will pass every unknown option to "make". So all "make" options are implicit "build.sh" options. "-j x" is the "make" option to perform multi-threaded builds.

Matthias

klayoutmatthias, 2021-03-22:
I think this issue is solved.

closed 2021-03-22 by klayoutmatthias

apteryks, 2026-10-18:
The -jN doesn't seem to be passed through for me, build.sh ends up calling `make install` gets invoked without any -j option, at least as arguments.