KLayout/klayout #717: Klayout error for missing python version 3

staghavi3, 2021-01-29:
Hi all,

Some MAC users have reported that after installing Klayout and then install SiEPIC packages, they see this error which cause "SiEPIC  ... py v..." option to be vanished from the top menu. 

[image]

Any idea could help us a lot.
Thanks,

klayoutmatthias, 2021-01-29:
Are you saying that even though Python 3 is present, this error message will pop up?

In the downloads section there are some MacOS versions available which are explicitly using Python 3. Does it work with them?

Thanks,

Matthias

Kazzz-S, 2021-01-30:
The DMG packages carrying names of `ST-*.dmg` use the system Ruby and Python.
Even the latest macOS **Big Sur** bundles Python version 2.7, not 3.x!

As Matthias commented above, you can find some DMGs built with Python 3.x on KLayout's site.
Those `LW-*.dmg` are available for only **Catalina** (primary OS), and you need a matching development environment such as **MacPorts**, **Homebrew**, or **Anaconda3**.

Please refer to [this ticket](https://github.com/KLayout/klayout/issues/689) for the DMG naming convention details.

I have recently uploaded similar DMGs for [**Big Sure**](https://github.com/KLayout/klayout/issues/689#issuecomment-767806927) for testing purposes. They are not yet published on KLayout's site.

Regrettably, those variations are not provided for older operating systems, including Mojave (10.14), High Sierra (10.13), Sierra (10.12), and El Capitan (10.11).
Keeping such complicated backward compatibility is very hard. So, only `ST-*.dmg` are built for them.

However, there is one exception I have tested so far.
If you are using **Mojave (10.14)** and able to install MacPorts with Ruby 2.7 and Python 3.8,
```
LW-klayout-0.26.9-macOS-Catalina-2-qt5MP-Rmp27Pmp38.dmg (Catalina (10.15) MacPorts based - Ruby 2.7 and Python 3.8 from MacPorts) 
```
can run on Mojave, too.

The final option is building KLayout from the source.

Thanks,
Kazzz-S

closed 2022-12-06 by klayoutmatthias