KLayout/klayout #730: Support of class 'QUiLoader' in PYA and RBA

Kazzz-S, 2021-02-16:
Dear @klayoutmatthias,

This is related to https://www.klayout.de/forum/discussion/comment/3035

In my PYA, I want to use a custom widget (promoted from QTableView) with various features, including keyboard Cut & Paste.
The user interface (*.ui) designed by using Qt Designer is complicated.
To enable these features, I've defined a subclass (promoted in Qt Designer) that is defined as:

```
class MyTableView( py.QTableView ):
    def __init_( self, parent=None ):
      super( MyTableView, self ).__init__(parent)
        :

    def keyPressEvent( self, event ):
        :
```

However, when starting the PYA script that reads the UI file by `pya.QFormBuilder().load()`, a warning below is delivered.

```
Warning: "QFormBuilder was unable to create a custom widget of the class 'MyTableView'; defaulting to base class 'QTableView'."
```

Referring to the idea in:
  https://stackoverflow.com/questions/37775472/quiloader-requirements-for-loading-ui-file-with-custom-widgets
Do you think it's possible to newly provide class 'QUiLoader' in PYA and RBA that can be used as follows?

```
class CustomQUiLoader( pya.QUiLoader ):
    def createWidget( self, className, parent=None, name='' ):
        if className == 'MyTableView':
            ret = MyTableView(parent)
            ret.setObjectName(name)
            return ret
        return super().createWidget(className, parent, name)

dialog = CustomQUiLoader().load( 'myComplicatedTable.ui', parent )
:
:
:
```

Thanks and best regards,
Kazzz-S

klayoutmatthias, 2021-02-23:
I have implemented the new Qt module now. You can use it from Ruby or Python. 

Here is my test code which works nicely with this version:

```

# Password dialog
FORM_TEXT = <<"END"
<?xml version="1.0" encoding="UTF-8"?>
<ui version="4.0">
 <class>PasswordDialog</class>
 <widget class="QDialog" name="PasswordDialog">
 
  <property name="geometry">
   <rect>
    <x>0</x>
    <y>0</y>
    <width>338</width>
    <height>229</height>
   </rect>
  </property>
  
  <property name="windowTitle">
   <string>Authentication Required</string>
  </property>
  
  <layout class="QGridLayout">

   <property name="leftMargin">
    <number>9</number>
   </property>
   <property name="topMargin">
    <number>9</number>
   </property>
   <property name="rightMargin">
    <number>9</number>
   </property>
   <property name="bottomMargin">
    <number>9</number>
   </property>
   <property name="spacing">
    <number>6</number>
   </property>

   <item row="2" column="1" colspan="2">
    <spacer>
     <property name="orientation">
      <enum>Qt::Vertical</enum>
     </property>
     <property name="sizeHint" stdset="0">
      <size>
       <width>20</width>
       <height>0</height>
      </size>
     </property>
    </spacer>
   </item>

   <item row="0" column="1">
    <widget class="QLabel" name="label">
     <property name="text">
      <string>User</string>
     </property>
    </widget>
   </item>

   <item row="0" column="2">
    <widget class="QLineEdit" name="user_le">
     <property name="sizePolicy">
      <sizepolicy hsizetype="Preferred" vsizetype="Fixed">
[... 176 lines cut ...]
    w = super
    if ! w
      if className == "PasswordEditBox"
        w = PasswordEditBox::new(parent)
        w.setObjectName(name)
      end
    end
    w
  end

end

# Create the form and show it

file = RBA::QBuffer::new
file.setData(FORM_TEXT)
file.open(RBA::QIODevice::ReadOnly)

loader = MyLoader::new
w = loader.load(file)

file.close

w.exec
```

Kazzz-S, 2021-02-23:
Dear @klayoutmatthias,

Thanks a lot for adding the new feature, which I have confirmed with Linux (Mint 19), as shown below.

[image] [image]
[image]
----
However, I got some build errors on Mac even after modifying a source file (a separate PR will be issued).
Please refer to the log file [in this directory.](https://www.dropbox.com/sh/uc6tt3ah2co73w5/AACzb4QeMc-8ikIH6-y6XUGoa?dl=0)

Warm regards,
Kazzz-S

klayoutmatthias, 2021-02-23:
I took the liberty of extending the scope of this ticket somewhat: I tried to test the QUiLoader on a standalone Python environment using the Python modules which come with KLayout (not the ones from PyPI). I found some severe issues with QApplication binding and finally had to remove "QApplication#notify" from the list of supported methods in Python. 

With this, the following code works (provided you have KLayout's Python modules built and in the path):

```

import klayout.QtCore as qtcore
import klayout.QtGui as qtgui
import klayout.QtWidgets as qtwidgets
import klayout.QtUiTools as qtuitools

# Password dialog
form_text = """<?xml version="1.0" encoding="UTF-8"?>
... (shortened) ...
"""

# Test code for QUiLoader customization
# 
# It supplies a custom widget called "PasswordEditBox" which changes
# the background color depending on the password "strength".

class PasswordEditBox(qtwidgets.QLineEdit):

  def __init__(self, parent):
    super(PasswordEditBox, self).__init__()
    self.textEdited += lambda s: self.update_background(s)
    self.update_background("")
  
  def update_background(self, s):
    # sets the background depending on the password "strength" 
    # TODO: better assessment of "strength" ...
    if len(s) > 12:
      color = qtgui.QColor(192, 255, 192)  # green -> good
    elif len(s) > 7:
      color = qtgui.QColor(255, 255, 192)  # yellow -> medium
    else:
      color = qtgui.QColor(255, 192, 192)  # red -> bad

    p = self.palette
    p.setColor(qtgui.QPalette.Base, color)
    self.palette = p
  
# Customization of the loader - supports the "PasswordEditBox" custom widget

class MyLoader(qtuitools.QUiLoader):

  def __init__(self):
    super(MyLoader, self).__init__()

  def createWidget(self, className, parent, name):
    if className == "PasswordEditBox":
      w = PasswordEditBox(parent)
      w.setObjectName(name)
    else:
      w = super(MyLoader, self).createWidget(className, parent, name)
    return w

# Create the form and show it

app = qtwidgets.QApplication(["test"])

file = qtcore.QBuffer()
file.setData(form_text)
file.open(qtcore.QIODevice.ReadOnly)

loader = MyLoader()
w = loader.load(file)

file.close()

w.exec_()
```

Kazzz-S, 2021-02-24:
Thanks for updating the code with commit 7352756f.

I've rebuilt it on Mac with the  "-j1" make option to stop at the first occurrence of error.
The last part of the messages is as below.
```
:
:
Undefined symbols for architecture x86_64:
  "gsi::qtdecl_QObject()", referenced from:
      __GLOBAL__sub_I_gsiDeclQUiLoader.cc in gsiDeclQUiLoader.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [../../../../qt5MP.build.macos-Catalina-release-RsysPsys/libklayout_QtUiTools.0.27.0.dylib] Error 1
make[2]: *** [sub-QtUiTools-make_first] Error 2
make[1]: *** [sub-qt5-make_first] Error 2
make: *** [sub-gsiqt-make_first] Error 2
:
:
```

Kazzz-S

closed 2021-02-25 by klayoutmatthias

Kazzz-S, 2021-02-25:
Dear @klayoutmatthias,

I've built the commit f299b3aa on Mac, and I still have a linker error.
```
:
Undefined symbols for architecture x86_64:
  "gsi::qtdecl_QObject()", referenced from:
      __GLOBAL__sub_I_gsiDeclQUiLoader.cc in gsiDeclQUiLoader.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [../../../../qt5MP.build.macos-Catalina-release-RsysPsys/libklayout_QtUiTools.0.26.11.dylib] Error 1
make[2]: *** [sub-QtUiTools-make_first] Error 2
:
```

I guess this is a similar issue as you explained in the commit log of 4ffcaba5.
I have gone through the relevant source files. However, I could not find the cause.  
As far as I can see, it looks the care is already taken :-(

Kazzz-S

Kazzz-S, 2021-02-26:
## PYA-Linux ##

* KLayout commit = 25d5eae9
* [PYA version](https://github.com/KLayout/klayout/files/6047720/PasswordDialog.pya.zip) of https://github.com/KLayout/klayout/issues/730#issuecomment-783769990

[image][image]
[image]

Kazzz-S

klayoutmatthias, 2021-02-26:
Hi @kazzz,

the branch is already merged to master (otherwise I cannot port it to 0.26).

There is one more commit on master: https://github.com/KLayout/klayout/commit/f993c0397d775a978024f14ae071a43d243d7330

I think this will fix the linker issue.

With this, all Linux and Windows builds are passing. I'll try Catalina too.

Best regards,

Matthias

Kazzz-S, 2021-02-27:
Hi @klayoutmatthias,

Thank you for fixing the link error. 💯 
 I could successfully build branch 889f3183 and run the sample scripts (both RBA and PYA) as shown in the images below.

[image]

## RBA-Mac ##
[image][image]
[image]

## PYA-Mac ##
[image][image]
[image]

Best regards,
Kazzz-S